Sevenstax/FreeV2G #3: Doku: WhiteBeet User Manual rev. 2.2

klth, 2021-07-13:
**In 9.1.3**
Hier wird angegeben: 
> „When sending messages not related to any particular request by the host controller, the WHITE beet –module will set the Request ID to 0xFF.“

Stimmt allerdings nicht für das V2G Module dort ist der Broadcast 0x00.

**In 10.3.1**
[image]
Aufbau undeutlich, es müsste sein:

> Code uint8
> String Len uint8
> Version string

Dies betrifft auch 10.3.2, 10.3.5 etc. alle wo eine String Variable definiert ist.

**In 15.10.1 und 15.10.2**
Example: Hex Wert für „2 protocols“ falsch. 0 -> 2

**In 15.1.**
Eventuell Type boolean hinzufügen und dass es sich um uint8 mit 1 = true und 0 = false handelt.

Bei Type exponential hinzufügen das der Exponent nicht kleiner als -3 sein darf. Zum Beispiel bei DiscoverChargeParameters->PeakCurrentRipple von 20000 * 10^-4, was laut Definition gültig sein müsste, bricht der Ladevorgang ab. Hoch -3 geht

**In 15.12.10**
Die Boolean Variable ob die nachfolgende Info existiert fehlt.
[image]
[image]

jpo-stx, 2021-07-15:
Hi @klth,

thank you for detailed input. Here my comments:

> Stimmt allerdings nicht für das V2G Module dort ist der Broadcast 0x00.

The broadcast should always be 0xFF. If it's off in the V2G module we need to fix this.

> Dies betrifft auch 10.3.2, 10.3.5 etc. alle wo eine String Variable definiert ist.

Generally we need to define the type string. Strings with a variable length are always prefixed with the length.

> In 15.10.1 und 15.10.2
> Example: Hex Wert für „2 protocols“ falsch. 0 -> 2

Thanks for the hint!

> In 15.1.
> Eventuell Type boolean hinzufügen und dass es sich um uint8 mit 1 = true und 0 = false handelt.

As with the string type we need a chapter describing the "standard" types.

> Bei Type exponential hinzufügen das der Exponent nicht kleiner als -3 sein darf. Zum Beispiel bei DiscoverChargeParameters->PeakCurrentRipple von 20000 * 10^-4, was laut Definition gültig sein müsste, bricht der Ladevorgang ab. Hoch -3 geht

You are correct, the range of the exponent needs to be specified. It is -3 to 3.

> In 15.12.10
> Die Boolean Variable ob die nachfolgende Info existiert fehlt.

Correct

BR
Jan

jpo-stx, 2021-11-11:
The issues should be solved with the latest Whitebeet firmware version V01_01_03 and the latest user manual v2.8.

closed 2021-11-11 by jpo-stx