Sevenstax/FreeV2G #4: Problem bei ChargeLoopParameters

klth, 2021-07-13:
Das Problem bei ChargeLoopParameters ist folgendes, anbei ein Wireshark Mitschnitt (Aufgrund von Vmware tauchen meine erzeugten Pakte immer dreimal im Trace auf, so dass Paket 2-4, 5-7, 8-9 und 11-13 jeweils als ein Paket zusehen sind):
-	Der Request ChargeLoopParameters kommt rein, der Timeout im Request wird mit 68ms angegeben (Paket Nr. 0)
-	Ich antworte mit SetChargeLoopParameters in diesem Trace nach 1,6ms  (Paket Nr. 2)
-	Die Antwort erhalte allerdings erst nach 106 ms mit dem Fehlercode Module Busy Try Later

Verzögere ich die Antwort um 30ms kommt „busy try later“ direkt. Um das Try Later zu berücksichtigen sende ich anschließend 10x im 3ms Abstand die Nachricht erneut. Hilft jedoch auch nicht. 

Woran habe ich nicht gedacht bzw. was mache ich falsch?
Gibt es eventuell ein Software Update? GetFirmwareVersion sagt V01_00_01

Dieses Verhalten tritt bei mir in der Simulation am Schreibtisch sowie an realer Hardware (Ladegerät + Auto) auf.

klth, 2021-07-14:
firmware update to V01_01_01 seems to solve this issue.

jpo-stx, 2021-07-15:
Hi @klth,

yes we had an issue in this firmware version, the new one fixes this.

BR
Jan

closed 2021-07-15 by jpo-stx