Sevenstax/FreeV2G #5: Gelegentlich erfolgt die Antwort nicht direkt

klth, 2021-07-13:
Es kommt gelegentlich vor das eine Antwort auf eine Anfrage ausbleibt, diese aber mit der nächsten Anfrage mit beantwortet wird. Zum verdeutlich was ich meine (als zeitliche Abfolge zubetrachten): 
- Request 1
-	Response 1
-	Request 2
-	Request 3
- Response 2
-	Response 3
-	Request 4
-	Response 4

Mir ist das beim synchronen Abfragen von Informationen aufgefallen, hier kam dann leider kein neuer Request so dass ich immer in den internen Timeout in meinem Programm gelaufen bin.

klth, 2021-07-15:
after a few hours of testing... firmware update to V01_01_01 seems to solve this issue.

jpo-stx, 2021-07-15:
Hi @klth,

nice to hear that the issue is solved. I was just about to ask you if you can retest with the V01_01_01 firmware, because it should solve this issue.

BR
Jan

closed 2021-07-15 by jpo-stx