Sevenstax/FreeV2G #7: Normaler Shutdown möglich?

klth, 2021-07-15:
Wie kann ich den Ladevorgang als Ladegerät regulär beenden? Also EVSE_Shutdown und nicht EVSE_EmergencyShutdown

Versucht habe ich bisher in Set Charge Loop Parameters den Code auf "EV parameters invalid" oder "Unable to deliver EVSE parameters" zu setzen bzw. die Anfrage nicht zu beantworten (Timeout). Jeder dieser drei Abbrüche sieht aber nach EmergencyShutdown aus. Was meiner Meinung nach bei der Namensgebung der Status Codes auch richtig ist. 

Ich komme darauf das es EmergencyShutdown sein müsste weil ich kein Shutdown Request vom Auto im Trace sehe. Von Charge Loop Parameter geht es direkt zum CP State B.

Die Fahrzeuge VW eUp und Hyundai Kona melden diesen Abbruch nicht als Fehler zum Anwender. Der BMW i3 regagiert hierauf  Rot blinkend und gibt mehrere Warntöne von sich.

Gibt es noch eine Möglichkeit? Ein Stop Knopf an einer Ladesäule ist ja vermutlich nicht unüblich und sollte ja nicht zu einer Fehlersignalisierung beim Anwender führen.

jpo-stx, 2021-07-15:
Hi @klth,

setting the Code parameter to "EV parameters invalid" or "Unable to deliver EVSE parameters" is the right way to stop a charging process. Currently the status information is not settable by the host controller since it has informational purpose only. But it would be a helpful information to have on EV side. Therefore I will check if we can implement this in a future version of the Whitebeet firmware.

Can you describe the usecase where you need to stop the charging process?

BR
Jan

klth, 2021-07-15:
Hi @jpo-stx,

the usecase to stop the charging process from the charger is that some car's don't have the possibility to abort the process by the user (e.g. they want go). In our case, this concerns the Hyundai Kona and the i3. This may only affect older cars but they exist.

From the power electronics side emergency stop or normal stop makes a hugh differnence. For example we are charging at full power (e.g. 350kW) and a emergency stop is triggered. The cp state switches from C to B. The only vaild state for a closed power relais is C. The reaction is that we have to open the relays under power. Power relais only allowed to be opened under full power a few times and then they must be replaced. I think the spec defines that the charger must shutoff in this case in a few milliseconds. This means that in most cases we are safe and the chargers shutoff faster than the relays and the relays open without power.

For this usecase (user abort) the better way is a normal stop. The car sends a stop request (Request Stop Charging) and the charger is shutting down. Waiting that is no voltage in the cable (Post Charge Parameters). Shutoff the relais (C->B). 
I think to achieve this we have to send the status "EVSE Shutdown" to the car.

At the moment, as you wrote, i can only quit the charging process over "EV parameters invalid" or "Unable to deliver EVSE parameters" and the result of this is the emergency stop. After i set one of this codes in "Set Charge Loop Parameters" the next event on my trace is "state changed C -> B" and not "request stop charging"

Maybe you can add a value to the parameters code, for example "3: request shutdown",  so that it is possible to make a normal stop.

best regards 
Klaus

ttu-stx, 2021-07-20:
As agreed we will implement normal shutdown information functionality and provide you with a test firmware for verfication. If this firmware will solve the problem we will foresee this feature in the next White Beet Firmware Release.
Best Regards
Tolga