Sevenstax/FreeV2G #8: SLAC Service: Join HPGP Network

zacharyalutz, 2021-07-15:
In user manual r2.2, a command was added for joining a HPGP logical network using network id (NID) and network management key (NMK).

1. **How does the host obtain _NID_ and _NMK_?**
    - Does Whitebeet manage NID/NMK generation or is this function to be implemented in host?
    - There is no other mention of this command nor of these parameters throughout the rest of the document.

2. **What is the intended use case for such a command?**
    - Is the host responsible for telling whitebeet to join HPGP network explicitly during each charging session?
    - Based on the application example for successful SLAC matching process, it appears that the Whitebeet takes care of joining the homeplug network through the _Start SLAC Matching_  command.
        - see  _CM_SLAC_MATCH.CNF_ with no mention of _Join HPGP Network_ command

    [image]

jpo-stx, 2021-07-20:
Hello @zacharyalutz ,

the command for joining a HPGP logical network was added so you can just join a network without doing the SLAC process. You would need to do this joining on both sides for EVSE and EV. It is mainly intended for development purposes.

> - Does Whitebeet manage NID/NMK generation or is this function to be implemented in host?
> - There is no other mention of this command nor of these parameters throughout the rest of the document.

Yes the Whitebeet is handling this you do not need to generate these parameters.

> - Is the host responsible for telling whitebeet to join HPGP network explicitly during each charging session?

No. The host just needs to initialize the SLAC module and start the matching when an EV connects. The EV will start the matching on it's side and the SLAC process is performed. You will receive a notification of the result. No join command needs to be sent to the Whitebeet.

> - Based on the application example for successful SLAC matching process, it appears that the Whitebeet takes care of joining the homeplug network through the Start SLAC Matching command.

Yes, the Whitebeet takes care of the joining.

varshilgandhi, 2024-03-27:
Hey @zacharyalutz ,
Did you implement the HPGP Command into your development?
I need some guidance of it. Because I need to implement HPGP into my development.

Thank you,
Varshil