Sevenstax/FreeV2G #9: Vehicle-to-Grid Service: Set Schedules

zacharyalutz, 2021-07-18:
If one inspects the example for the `Set Schedules` command, it appears that it does not include the `schedule ID` as part of the payload (see  **image 1** below). However,  in the `Request Start Charging` command, we see the parameter `schedule ID` appear again, only this time the data type is different: we have `uint8` instead of `uint16` (see **image 2** below). What should I expect from each command?

## Image 1
   [image]

## Image 2
   [image]

jpo-stx, 2021-07-19:
Hi @zacharyalutz,

the example lacks the schedule ID. It needs to be set by the host controller. It will be referenced by the EV when the charging is started.

Internally we use a uint8 for the schedule ID. I am not sure why a uint16 is used in the "Set Schedules" message. Only the least significant byte is used for the ID. Maybe we need a fix here. Thanks for pointing it out.

zacharyalutz, 2021-07-19:
Hi @jpo-stx , thanks for clarifying. I will make a note in my user manual. I'll leave this open for tracking purposes unless you say otherwise.

jpo-stx, 2021-07-20:
Hi @zacharyalutz , please leave this open. I will close it with the next updated manual, which will be uploaded in the next days.