Sevenstax/FreeV2G #10: Charging session stopping at precharging stage

semalinear, 2021-10-13:
Hi,

I am trying Codico WB-CARRIER-BOARD-EI board for ISO15118 charging. RIght now I'm using FreeV2G python demo app from sevenstax.
I observed that the application stops at the pre-charge stage and does not send some required information to start the charging.
Could you please have a look on to the logs? (Bothe command line and serial port logs are attached.)
Firmware used in the board is Sevenstax_WHITEBEET_ISO15118_EVSE_SW_V01_01_02 (as updated in codico documentation area ).
[codico_commandline_log_003.txt](https://github.com/Sevenstax/FreeV2G/files/7334787/codico_commandline_log_003.txt)
[codico_serial_debug_log_003.txt](https://github.com/Sevenstax/FreeV2G/files/7334788/codico_serial_debug_log_003.txt)

-Ganesh LD

ganeshldsema, 2021-10-18:
> Sure will upload the captured log..
> Hi Ganesh,
> Would it be possible to get a Wireshark capture at the ethernet interface
> of the Demo application?
> Thanks!
> Vasily Budko
> CODICO GmbH

Here is the Wireshark log

ganeshldsema, 2021-10-19:
Hi Vasily,

Here is the wireshark capture log.
Please rename the file extension to ".pcapng" before you open it with wireshark.

[codico_board_wireshark_log_003_pcapng_.txt](https://github.com/Sevenstax/FreeV2G/files/7370237/codico_board_wireshark_log_003_pcapng_.txt)

ganeshldsema, 2021-10-20:
Hi..

Were you able to look into the issue?

ttu-stx, 2021-10-20:
We will look into it within this week and come back to you.
BR

ganeshldsema, 2021-11-10:
Hello Jan,

Any updates on this issue?

Thanks!

vbudko, 2021-11-10:
Hi Ganesh,
We will place an update of the SW v01.01.03 today on the [](https://downloads.codico.com/misc/plc)
I will inform you with a separate email once it is done. 
BR, Vasily

jpo-stx, 2021-11-10:
> 
> 
> Hello Jan,
> 
> Looks like that Codico board (EVSE) is not populating the charging schedules properly, (as per logs by simulator). All the 3 charging schedules are coming as "0" in the simulator side. Please refer to the logs.
> 
> -Ganesh
> 
> [codico_commandline_dc_003.txt](https://github.com/Sevenstax/FreeV2G/files/7431785/codico_commandline_dc_003.txt) [codico_serial_dc_003.txt](https://github.com/Sevenstax/FreeV2G/files/7431787/codico_serial_dc_003.txt) [trialog_dc__ev-combocs-log_001.txt](https://github.com/Sevenstax/FreeV2G/files/7431792/trialog_dc__ev-combocs-log_001.txt)

Hi @ganeshldsema ,
we have reworked the schedule handling, please check the updated user manual, which will also be available this week.

jpo-stx, 2021-11-11:
@ganeshldsema , please check out the latest firmware version and user manual available for the Whitebeet.
With this I checked the schedules received on the EV:

```
ChargeParameterDiscoveryRes:
  Session ID: 07E0B6D14103DBAF
  ResponseCode: OK
  EVSEProcessing: Finished
  DC_EVSEChargeParameter:
    DC_EVSEStatus:
      EVSEIsolationStatus: Valid
      EVSEStatusCode: EVSE_Ready
      NotificationMaxDelay: 0
      EVSENotification: None
    EVSEMaximumCurrentLimit: 100e+0 A
    EVSEMaximumPowerLimit: 25000e+0 W
    EVSEMaximumVoltageLimit: 400e+0 V
    EVSEMinimumCurrentLimit: 0e+0 A
    EVSEMinimumVoltageLimit: 0e+0 V
    EVSEPeakCurrentRipple: 2e+0 A
  SAScheduleList:
    SAScheduleTuple #0:
      SAScheduleTupleID: 1
      PMaxSchedule:
        PMaxScheduleEntry #0:
          TimeInterval:
            start: 0
            duration: 1800
          EVMaximumPowerLimit: 25000e+0 W
        PMaxScheduleEntry #1:
          TimeInterval:
            start: 1800
            duration: 1800
          EVMaximumPowerLimit: 18750e+0 W
        PMaxScheduleEntry #2:
          TimeInterval:
            start: 3600
            duration: 82800
          EVMaximumPowerLimit: 12500e+0 W
```

sabarish-sema, 2021-11-12:
Hi,
This is  Sabarish, an employee of semaconnect.

I am trying Codico WB-CARRIER-BOARD-EI board for ISO15118 charging with the latest updated firmware SW v01.01.03.
RIght now I'm using FreeV2G python demo application from sevenstax.
I observed that the application stops at Set Schedules state.I think the Free V2G python demo code is not updated w.r.t the latest changes in updated firmware SW v01.01.03.

Thanks!

jpo-stx, 2021-11-12:
Hi @sabarish-sema , the demo code was updated yesterday. Can you provide a wireshark log of the ethernet interface?

sabarish-sema, 2021-11-12:
Hello Jan,

we tried with the latest Free V2G python demo code with updated firmware SW v01.01.03.
The demo application stops in the  Charge Loop Parameters state after charging to 30-40 seconds.please suggest where we are doing wrong.
Please refer to logs.

[ver_3_codico_commandline_log_005.txt](https://github.com/Sevenstax/FreeV2G/files/7527138/ver_3_codico_commandline_log_005.txt)
[ver_3_trialog_simulator_EV_log_004.txt](https://github.com/Sevenstax/FreeV2G/files/7527141/ver_3_trialog_simulator_EV_log_004.txt)
[ver_3_codico_serial_log_006.txt](https://github.com/Sevenstax/FreeV2G/files/7527146/ver_3_codico_serial_log_006.txt)

Thanks!

jpo-stx, 2021-11-12:
From the log file [ver_3_codico_serial_log_006.txt](https://github.com/Sevenstax/FreeV2G/files/7527146/ver_3_codico_serial_log_006.txt) I can see that you do not meet the timing requirements for setting the charge loop parameters and therefore the Whitebeet is sending a the "Failed" response code to the EV:

`[   74.195] [NOTICE] V2GISO13: Message performance timeout of 75ms reached.`

Are you running the host application in a virtual machine?

sabarish-sema, 2021-11-12:
Hello Jan,
We are not using virtual machine.For running the application we are ubuntu machine with super user permission.

Thanks

jpo-stx, 2021-11-12:
I will analyse the issue and come back to you at the beginning of next week.

jpo-stx, 2021-11-22:
@sabarish-sema can you please use the newest version V01_01_05 and check if the issue still persists?

sabarish-sema, 2021-11-22:
hello Jan

we tried with latest updated firmware WHITEBEET_ISO15118_EVSE_V01_01_05.
we are still facing same issue the demo application stops in the Charge Loop Parameters state after charging to few seconds.

[ver_5_codico_commandline_log_010.txt](https://github.com/Sevenstax/FreeV2G/files/7579356/ver_5_codico_commandline_log_010.txt)
[ver_5_codico_serialline_log_011.txt](https://github.com/Sevenstax/FreeV2G/files/7579357/ver_5_codico_serialline_log_011.txt)

Thanks!

jpo-stx, 2021-11-22:
@sabarish-sema it seems that a timeout is reached on the Whitebeet because the Charge Loop Parameters message is sent too late by the host. This seems to be a problem of scapy. We have tried with pylibpcap for python wich seems to be faster most of the time, but it is only easily available on Linux systems. Let me see if I can push a branch with the pylibpcap

jpo-stx, 2021-11-22:
@sabarish-sema please try with branch https://github.com/Sevenstax/FreeV2G/tree/eth_pylibpcap

closed 2022-02-24 by jpo-stx

lherreraalloza, 2022-04-22:
@jpo-stx I have the same problem (Set Charge Loop parameters is sent over the 75ms timeout) with the suggested branch (eth_pylibpcap). Testbed used was:
EVSE firmware image: V01_01_07
EV firmware image: V01_00_04
Host processor: Windows 10 Pro running on Intel i7

I have also tested the EVSE_v1.1.7_0 tag and the master branch. Master branch takes more time in triggering the timeout error, but at the end the issue is the same in all tested FreeV2G commits.

Could I relax the EVSE timeout by sending some command?

[HOST EVSE.txt](https://github.com/Sevenstax/FreeV2G/files/8539465/HOST.EVSE.txt)
[UART WB EVSE.txt](https://github.com/Sevenstax/FreeV2G/files/8539466/UART.WB.EVSE.txt)
[HOST EV.txt](https://github.com/Sevenstax/FreeV2G/files/8539467/HOST.EV.txt)