Sevenstax/FreeV2G #16: "Session stopped" received

buktemirlnk, 2021-12-06:
Hello to everyone,
I have been working on the WB-CARRIER-BOARD-EI ISO 15118(EVSE) module for a while. The vehicle I tested on is the Renault Zoe. I am running the host software on Eclipse.

Looking at the debug outputs and system steps, I reach a certain stage and then I get a "Session Stopped Received" message from the module. Every time I see the message “No EVSE ID available” on the terminal screen. In this case, what is the factor that causes the session to be terminated?

I look forward to your feedback on the matter.

The software version I use in Whitebeet: Sevenstax_WHITEBEET_ISO15118_EVSE_SW_V01_01_05
I am using VM (Ubuntu 20.04) to run FreeV2G software.

I am sharing the outputs of the study in the attachment.

Appendix 1: Wireshark capture on running test

Appendix 2: Module debug interface outputs

Appendix 3: FreeV2G terminal outputs

[Attachments.zip](https://github.com/Sevenstax/FreeV2G/files/7660689/Attachments.zip)

jpo-stx, 2021-12-06:
@buktemirlnk have you tried setting an EVSE ID for the DIN format?

Instead of:
https://github.com/Sevenstax/FreeV2G/blob/74226facf3acac897c1aca0b803271b577df4d01/Evse.py#L162-L164

I.e. set the ID to:
```
evseid = "+49*123*456*789"
print("Set EVSE ID: {}".format(evseid))
try:
    self.whitebeet.v2gSetEvseId(evseid)
```

Normally the EV should accept it when the EVSE ID is set to zero, but you can give it a try.

buktemirlnk, 2021-12-07:
First of all, thank you for your feedback.

I tried your solution, now I don't have a problem with EVSE ID, but after the EVSE ID is accepted, I get "Session stopped received" again. The system finishes the process directly. Do you have any solution for this issue?

jpo-stx, 2021-12-07:
@buktemirlnk can you please activate the port mirror (Whitebeet Manual V2.9 Chapter 12.3.10 and 12.3.12) and capture with wireshark again? The port mirror will mirror the frames that are sent/received on the the PLC interface to the ethernet interface.

jpo-stx, 2021-12-08:
Hello @buktemirlnk ,

the problem seems to be the selection of the energy transfer modes. When DIN70121 is set to supported you can only set the energy transfer modes to the following combinations:

    only 0 (DC Core)
    only 1 (DC extended)
    only 2 (DC combo core)
    1 and 2 (DC extended and DC combo core)

I will add this as a note to the user manual. Please check if this fixes your issue

buktemirlnk, 2021-12-08:
Hello @jpo-stx ,
Thank you for your answers. We are currently creating a setup to try your answers. I will share it with you again when our experiments yield results.

jpo-stx, 2021-12-08:
Thank you @buktemirlnk . Looking forward to your results.

anr32, 2021-12-10:
Hello all,
I just wanted to thank you for your insight and very timely suggestions!
I have been following a few days behind this thread with my own attempts to communicate with a BMW i3 using a CODICO white-beet development board. I was having the same issue and can confirm that setting the supported energy transfer modes to "only 1 (DC extended)" was the solution in my case. With this option (and a further small tweak so that the advertised EVSE max voltage was greater than the EV's max voltage) I have been able to get as far as (and into) the pre-charging stage.
Thanks again for the insights - I thought I would share my mileage with the BMW i3.
Alex

buktemirlnk, 2021-12-11:
Hello @jpo-stx,
Today we tried the solution you suggested on Renault Zoe and it solved the problem.The problem was with the energy transfer mode. When we did only 1, the problem was solved. Thank you very much for your interest and solution suggestions.

closed 2021-12-24 by buktemirlnk