Sevenstax/FreeV2G #17: Session Stopped issue

marsisyazilim, 2021-12-06:
Hello 

I am trying to test whitebeet  dev board with Tesla Model 3. on the v2g stage I cant pass evse id requested part.
 I am trying with raspberry pi. Output firstly, said "No EVSE ID available" and I added this line 
if message['format'] == 0:
            evseid = "+90*123*456*789"

in this case said session stopped after send evseid here you can see output message. also I added attachment output message with serial debug messages.

Here I share output both network and serial debug. Where am I doing wrong?

Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: eth0, mac: C4:93:00:21:4E:96
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
128 b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x98\xed\\\x8a\x0b\xdb'
"Session started" received
Protocol: 0
Session ID: 0000000000000000
EVCC ID: 98ed5c8a0bdb
130 b'\x00\x00\x06\x08\x00'
"Request EVSE ID" received
Set EVSE ID: +90*123*456*789
129 b''
"Session stopped" received
EVSE loop finished
Goodbye!

[ 1255.341] [NOTICE] SLAC: stxSLAC_Start()
[ 1255.376] [NOTICE] V2GCPS: Change State from B to A
[ 1255.380] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 1255.386] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[ 1255.391] [NOTICE] V2GCPS: STATE A
[ 1257.346] [NOTICE] APL-ETH: Lost link:
[ 1257.349] [NOTICE] APL-ETH:     Interface: eth0
[ 1257.354] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:4E:97
[ 1257.360] [NOTICE] APL-MAIN: 
[ 1257.362] [NOTICE] APL-MAIN: =============================
[ 1257.368] [NOTICE] APL-MAIN: LAN CONFIGURATION DELETED: 
[ 1257.373] [NOTICE] APL-MAIN: for IP Config handle [fe00] 
[ 1257.378] [NOTICE] APL-MAIN: IPv6 Addresses:
[ 1257.382] [NOTICE] APL-MAIN:   Link-Local:  0:0:0:0 : 0:0:0:0
[ 1257.388] [NOTICE] APL-MAIN:   Local:	0:0:0:0 : 0:0:0:0
[ 1257.393] [NOTICE] APL-MAIN:   DNS1:	0:0:0:0 : 0:0:0:0
[ 1257.399] [NOTICE] APL-MAIN:   DNS2:	0:0:0:0 : 0:0:0:0
[ 1257.404] [NOTICE] APL-MAIN: =============================
[ 1257.409] [NOTICE] APL-MAIN: 
[ 1257.412] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[ 1259.356] [NOTICE] APL-ETH: Got link:
[ 1259.360] [NOTICE] APL-ETH:     Interface: eth0
[ 1259.364] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:4E:97
[ 1261.543] [NOTICE] V2GCPS: Change State from A to B
[ 1261.547] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 1261.553] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[ 1261.558] [NOTICE] V2GCPS: STATE B
[ 1261.720] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[ 1262.190] [WARN  ] SLAC:   Unexpected message!
[ 1262.214] [WARN  ] SLAC:   Unexpected message!
[ 1263.571] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[ 1263.774] [NOTICE] SLACSERV: # NC_SLAC_SUCCESS
[ 1264.361] [NOTICE] V2GSVC: Starting in EVSE mode.
[ 1264.366] [NOTICE] V2GSVC: TCP: Listen on port 58550 for non-TLS
[ 1264.586] [NOTICE] V2GSVC: TCP: A client connected on the TCP port
[ 1264.593] [NOTICE] V2GDIN12: EV requested to start session.
[ 1268.524] [NOTICE] APL-MAIN: 
[ 1268.527] [NOTICE] APL-MAIN: =============================
[ 1268.532] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6): 
[ 1268.537] [NOTICE] APL-MAIN: for IP Config handle [fe00] 
[ 1268.543] [NOTICE] APL-MAIN: IPv6 Addresses:
[ 1268.547] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe21:4e97
[ 1268.554] [NOTICE] APL-MAIN:   Local:	0:0:0:0 : 0:0:0:0
[ 1268.559] [NOTICE] APL-MAIN:   DNS1:	2001:4860:4860:0 : 0:0:0:8888
[ 1268.565] [NOTICE] APL-MAIN:   DNS2:	fd00:0:0:0 : be05:43ff:fe44:5280
[ 1268.572] [NOTICE] APL-MAIN: =============================
[ 1268.577] [NOTICE] APL-MAIN: 
[ 1268.742] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server
[ 1268.750] [NOTICE] V2GSVC: TCP: Listen on port 58550 for non-TLS
[ 1268.791] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server
[ 1269.010] [NOTICE] SLAC: stxSLAC_Stop()

jpo-stx, 2021-12-07:
Hi @marsisyazilim, I think that #16 is the same issue as yours, please check ot my comment there https://github.com/Sevenstax/FreeV2G/issues/16#issuecomment-987883224

marsisyazilim, 2021-12-08:
when I activete it, problem will solve or just we see what messages comes?

jpo-stx, 2021-12-08:
I think that the problem was already found: https://github.com/Sevenstax/FreeV2G/issues/16#issuecomment-988695769

It has to do with the energy transfer modes. Please set the modes as mentioned in the comment and check if the issue persits or if this solves the issue.

BR
Jan

jpo-stx, 2021-12-10:
@marsisyazilim please check out my latest commit https://github.com/Sevenstax/FreeV2G/commit/d9291a407258ef39e79ad02993e6f22365c1da22

The handling of signed values was not correct.

marsisyazilim, 2021-12-10:
replacing whitebeet.py file is enough right?

jpo-stx, 2021-12-10:
Yes, it should be sufficient to replace only this file.

anr32, 2021-12-10:
Hi @jpo-stx,
Just a little note to say thank you for the whitebeet.py update - it was a pleasant surprise to find the issue I discovered today (with the signed values handling) had been fixed on the repository 2 hours earlier!
Thanks, Alex

marsisyazilim, 2021-12-10:
Thank you solved. it is sending continously "Request Pre Charge Parameters" received message. is this happen because I dont apply voltage? do I have to apply voltage in this stage?

"Request Schedules" received
Max entries: 12
Set the schedule: [{'id': 1, 'tuple': [{'start': 0, 'interval': 1800, 'max_power': 25000}, {'start': 1800, 'interval': 1800, 'max_power': 18750}, {'start': 3600, 'interval': 82800, 'max_power': 12500}]}]
"Request Discovery Charge Parameters" received
EV maximum current: 499.0A
EV maximum power: 211000W
EV maximum voltage: 422.0V
SOC: 98%
"Request Cable Check Status" received
"Request Cable Check Parameters" received
SOC: 98%
"Request Pre Charge Parameters" received
EV target voltage: 351.0V
EV target current: 0.0A
SOC: 98%
"Request Pre Charge Parameters" received
EV target voltage: 351.0V
EV target current: 0.0A
SOC: 98%
"Request Pre Charge Parameters" received
EV target voltage: 351.0V
EV target current: 0.0A
SOC: 98%

jpo-stx, 2021-12-17:
> Thank you solved. it is sending continously "Request Pre Charge Parameters" received message and stop session. is this happen because I dont apply voltage? do I have to apply voltage in this stage?

Yes this is the phase where the EVSE should apply the voltage to the output. The EV will continue when the EVSE output volatge has reached the level of the battery voltage.

closed 2022-01-19 by jpo-stx