Sevenstax/FreeV2G #19: Interoperability: DIN SPEC 70121 support?

zacharyalutz, 2021-12-07:
It seems that Whitebeet struggles to communicate by way of DIN SPEC 70121. We have tested with and without our power electronics system, both with and without a real EV in the mix and cannot seem to get past the ServiceDiscoveryReq/ServiceDiscoveryRes message pair. As far as I can tell from your documentation, DIN SPEC 70121:2012 is supported and ServiceDiscoveryReq is handled by the Whitebeet EI application, without a direct response required from the host. Why would Whitebeet send “ResponseCode=FAILED” in ServiceDiscoveryRes? Does the Whitebeet-EI firmware support DIN SPEC 70121? 

For your reference, I have attached an archive [din70121-fail.zip](https://github.com/Sevenstax/FreeV2G/files/7672223/din70121-fail.zip) which contains the following:
  - *din70121_fail.pcapng* -- Wireshark log for the Host-Whitebeet interface during failing session
  - *din70121_fail.txt* -- PuTTY log for the debug interface during failing session
  - *din70121_ServiceDiscoveryReq_fail.txt* -- screen cap of emulator running DIN SPEC 70121 state machine during failing session
  - *port_mirror_fail.pcapng* -- Wireshark log showing Port Mirroring function not working once firmware is updated to v01_01_05

I configure whitebeet just as done in the FreeV2G application, and this same behavior was observed running on our real system with a real EV, running on our real system with an EV emulator platform, and running with FreeV2G python application driving Whitebeet to EV emulator. The data I captured was running v01_01_02 firmware, but I have updated to v01_01_05 and observed the same results. I noticed someone with similar issues posted to the FreeV2G GitHub Issues forum and made an attempt at port mirroring for sniffing the QCA700x traffic, but the result code 0x02 indicated that the Sub ID was not recognized (see *port_mirror_fail.pcapng*). I eagerly await your response, and thanks again for your help.

#### V2G Message Sequencing 
[image]

#### EV Emulator Trace (Keysight Charging Discover System)
[image]

jpo-stx, 2021-12-08:
Hi @zacharyalutz ,

the problem seems to be the selection of the energy transfer modes. When DIN70121 is set to supported you can only set the energy transfer modes to the following combinations:

- only 0 (DC Core)
- only 1 (DC extended)
- only 2 (DC combo core)
- 1 and 2 (DC extended and DC combo core)

I will add this as a note to the user manual and will come back to regarding the problem with the port mirror.

BR
Jan

jpo-stx, 2021-12-08:
Hi @zacharyalutz ,

the port mirror feature is currently only available in the EV firmware. I'll add a note in the user manual.
We are planning to release this feature in the next EVSE firmware.

BR
Jan

zacharyalutz, 2021-12-08:
Great, thanks for the speedy response. Tested this today where instead of configuring energy transfer modes as all DC types, i.e. {0,1,2,3}, I configured solely for DC Extended {1}, which absolutely works. Also thanks for the info re: port mirroring.

[image]

Curious if you are able to share which EVs (make/model) Whitebeet has been tested for ISO 15118 and which for DIN SPEC 70121?

closed 2022-02-24 by jpo-stx