Sevenstax/FreeV2G #21: Pausing a V2G communication session

zacharyalutz, 2021-12-14:
From ISO 15118-2 section 8.4.1, we have the following:

> In ISO 15118 a V2G Communication Session pausing is controlled by the parameter ChargingSession in SessionStopReq with its values “Terminate” and ”Pausing”. The parameter can be used independently from the charging profile. This means that an EV can initiate a pausing at any time after sending PowerDeliveryReq with ChargeProgress equal to 'Stop' and according to [V2G2-739]. 

[image]

How can EVSE know that a charging session is being paused by EV?  Here's my best guess at how this might go down:

1. Host receives _Request Stop Charging_ `0x8b` from WB
2. Wait  for indefinite period of time for one of the following messages from WB
    - _Session Started_ `0x80`
    - _Session Stopped_ `0x81` 
    - _Request Post Charge Parameters_ `0x8c`

If the above sequence is what is to be expected from WB, the host has no way to positively identify that we are "paused". One could imagine a scenario where host receives _Request Stop Charging_ `0x8b` from WB, and we have loss of communication for some reason. There's really no way to differentiate between a paused V2G session and loss of comm. Is it possible that WB provide a field within a message to indicate what the standard refers to as the parameter _ChargingSession_ ? Perhaps a new message pair could be added to EVSE host<==>WB side that mirrors _SessionStop.Req/Res_ on the WB<==>EV side. 

In particular, keep _Session Stopped_ `0x81` to indicate that the TCP link is actually down, but add something like _Request Session Stop_ `0x8d` where we have a field called _ChargeSession_ that takes on the values  [ _Terminate_ `0x00` | _Pause_ `0x01` ] so that EVSE host can understand what's going on. 

- If _ChargeSession=Terminate_, host would expect it to be followed soon after with _Session Stopped_ `0x81`.
- If _ChargeSession=Pause_, host can confidently wait for session to resume with _Session Started_ `0x80`.

zacharyalutz, 2021-12-17:
I also have questions about pause/sleep at the physical and datalink layers. In ISO 15118-3 section 7.6 _Sleep mode and wake-up_, requirements define expected behavior during all parts of the sleep phasis, i.e.  how to enter sleep, what to do while asleep, how to wake up, etc.  

I am curious if the coordination defined between the Control Pilot and SLAC are managed automatically by Whitebeet? Is the host ever made aware of such activity? For example, refer to the following requirement for EVSE entering sleep mode:

---
[image]

bbr-stx, 2021-12-21:
Hi @zacharyalutz, due to vacation, we will come back to the questions from your initial post not before the beginning of next year. Regarding the question about the Control Pilot and SLAC, I can already tell you that this is not automatically managed by the Whitebeet module. This is in the responsibility of the host.

jpo-stx, 2022-01-04:
Hi @zacharyalutz ,

you are right, there should be some information for the host whether a session was terminated or paused and if a session was resumed in the session started message.

Currently the way to decide if a pause happend is only to observe the state on the CP:

### Session Terminate
1. Request Stop Charging `0x8b`
2. Session Stopped `0x81`
3. CP state changes to A

### Session Pause
1. Request Stop Charging `0x8b`
2. Session Stopped `0x81`
3. No change of CP state
4. Session Started `0x80`

jpo-stx, 2022-03-22:
Hi @zacharyalutz ,

please checkout the latest version 1.1.7 with the added information in the session stop message.

When the SLAC service is not stopped and the start matching command is triggered again the same NID and NMK will be used again.

closed 2022-03-22 by jpo-stx