Sevenstax/FreeV2G #23: How to correctly configure service modules and restore service them to usable idle state? (CP,SLAC,V2G)

zacharyalutz, 2021-12-18:
How do I restore each service (CP/SLAC/V2G) to a state that is ready for the next session? I am familiar with the _Appication Examples_ section of the user manual, but I find them either lacking in information, or sometimes incorrect. Let's start with CP Service to address what I believe is a mistake, just to get that out of the way, as it's not a major concern of mine. 

### CP Service
The message sequence diagram in the user manual instructs the Host EVSE to stop the CP service after high-level communication with the EV is complete. If the service is turned off, it seems to me that we would not be able to detect the next plugin.

[image]

- _**Suppose you want Whitebeet to run indefinitely in EVSE mode in a DC fast charger...how would you handle the end of a session with respect to the CP service?**_
  - _My method to manage end of session is to return CP to the x1 state by issuing "Set PWM duty cycle to 100%" (CP service left on)._
- **_Besides changing the EV/EVSE mode, what is an intended use case for stopping CP service?_** 
- **_What happens when CP service is turned off besides turning off 12V power supply?_**

---

### V2G Service
Between the user manual, experience with testing, the FreeV2G source code, and threads on this forum, I have been able to piece together correct usage at the beginning and end of a session. On the other hand, it's not clear to me what the "best practices" for V2G service usage are. 

For example, in the configuration portion, we set some (but not all) parameters that remain constant to the station. A good example is the EVSEID. We shouldn't expect the EVSEID value to change from session to session, only the format changes according to which protocol is selected (ISO vs DIN). So why not set each value once during configuration, and let it be?

Speaking of only configuring Whitebeet once, from the user manual I would assume that configuration is performed after reset release, but in the FreeV2G application, configuration happens every single charging session!

[image]

- **_What is the best method to configure V2G service: once after reset or once after each successful SLAC match?_**
  - _I have tested both, and both seem to work, but what is your recommendation?_
- _**When should the V2G service be stopped??**_
  - _I turn it off either after receiving_ "Session Stopped (0x81)" _or in the case of emergency shutdown and loss of comm, I turn it off after safe voltage level is observed. (I allow an interval for response from EV, but after the timeout, service is closed.)_  
  - _According to the diagram, it should not be stopped: is this because the message "Session Stopped (0x81)" is an indication that the V2G *service* is stopped as well as the session? What gives?_
- _**As I understand it, when V2G Service is stopped, TCP port is closed..what else happens when the service is stopped?**_

---

### SLAC Service
It is clear to me how to _initialize_ the SLAC service, but not so much how to restore it to a proper idle state. The application example sequence diagrams for CP and V2G at least depicted the end of the charging session, but there is no such diagram for SLAC. My current understanding is to leave SLAC running at the end of charging session until plugout. After plugout, I turn SLAC service off, wait, then turn it back on. 

- _**What is your recommendation for starting and stopping the SLAC service?**_
- _**What actually happens when the SLAC service is stopped?**_
- **_At what point are NID/NMK generated_**?
- **_At what point is the HomePlug Green PHY logical network established? Disestablished?_** 

---

### Closing Remarks
In my first pass as designing the host interface, my assumption was that each service needed to be configured, but only configured _once_. And after configuration, my assumption was that I would turn the service on _once_. The only reason I conceived to turn the service off was to re-configure it. However, it seems that each service requires a different approach with respect to (1) configuration and (2) restoring the services to an idle/standby state such that is ready for the next session. 

### TL;DR

1. When do I start service?
2. What actually happens when I start service?
3. When do I stop service?
4. What actually happens when I stop service?

bbr-stx, 2021-12-21:
Hi @zacharyalutz, thanks for your detailed description. However, due to vacation we will not be able to answer it until the beginning of next year. I apologize for this.

jpo-stx, 2022-01-04:
Hi @zacharyalutz ,

### CP Service
Your handling of session ending is correct.
The only use case for stopping the service is when you do not want to accept any further charging sessions. The voltage on CP is -12V in this case.

### V2G Service
After startup, you only need to configure the service once.
When the service is started the listen ports for V2G (TCP) and SDP (UDP) are opened.
When the service is stopped an open connection and the ports are closed.
I would recommend to start the service after plug-in and stop it after plug-out.

### SLAC Service
What I know is that after startup the service needs to be started and slac matching needs to be started to be ready for a slac matching request from the EV. For more details I need to ask my collegue who will return from his holidays at the end of the week. I will come back to you with a more detailed response ;)

jpo-stx, 2022-03-29:
Hi @zacharyalutz ,
I have checked with my collegue and he recommends to have a delay of 2s between start slac and start matching, because of communication beween Whitebeet and QCA. Furthermore, it is recommended to start matching and after a delay of ~200ms the duty cycle should be activated i.e. to 5%, because the EV will start it's matching sequence when it sees the duty cycle and the SLAC service needs a few ms to be ready for matching. I'll add this info to manual.

closed 2022-03-29 by jpo-stx