Sevenstax/FreeV2G #25: SLAC matching timed out

johnsonWu0528, 2021-12-23:
Hello to everyone,

I have been testing on the WB-CARRIER-BOARD-EI ISO 15118(EVSE) module and  WB-CARRIER-BOARD-PI ISO 15118(EVSE) module for a while. 

I used Sevenstax/FreeV2G/master and /EV support for testing and follow as that README.md.

In testing i got two log, follow as:

> Log 1
> D:\work\ev\seven_satx\FreeV2G> py .\Application.py "Realtek USB GbE Family Controller #2"
> Welcome to Codico Whitebeet EVSE reference implementation
> Initiating framing interface
> iface: Realtek USB GbE Family Controller #2, mac: C4:93:00:21:51:12
> Set the CP mode to EVSE
> Set the CP duty cycle to 100%
> Start the CP service
> Start SLAC in EVSE mode
> Wait until an EV connects
> EV connected
> Start SLAC matching
> Set duty cycle to 5%
> SLAC matching timed out
> EVSE loop finished
> Goodbye!

> Log 2
> [  331.551] [NOTICE] SLAC: stxSLAC_Start()
> [  331.664] [NOTICE] V2GCPS: Change State from B to A
> [  331.668] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
> [  331.674] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
> [  331.679] [NOTICE] V2GCPS: STATE A
> [  342.121] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
> [  342.221] [NOTICE] V2GCPS: Change State from A to B
> [  342.225] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
> [  342.231] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
> [  342.236] [NOTICE] V2GCPS: STATE B
> [  392.127] [ERROR ] SLAC:  Timeout: No SLAC matching within defined time window!
> [  392.134] [NOTICE] SLACSERV: # NC_SLAC_FAILED - 1
> [  392.142] [NOTICE] SLAC: stxSLAC_Stop()

In the log you can see "NC_SLAC_FAILED", can you help me fix this to establish SLAC connection?

johnsonWu0528, 2021-12-24:
Hello,

I keep testing and had some finding, the log message print "V2GCPS: Duty cycle is 56 permill" from WB-CARRIER-BOARD-PI ISO 15118(EVSE) board , but in the oscilloscope monitor is 5.06% , follow as :

> [2021-12-24 17:13:30.508] APPLICATION:
> [2021-12-24 17:13:30.508]   Name:      ISO15118 (EV)
> [2021-12-24 17:13:30.524]   Version:   V01_00_03
> [2021-12-24 17:13:30.524]   Target:    WHITE-BEET
> [2021-12-24 17:13:30.524]   CPU:       STM32F745
> [2021-12-24 17:13:30.524]  
> [2021-12-24 17:13:30.524] PRODUCT VERSION:
> [2021-12-24 17:13:30.524]  String: SEVENSTAX_LIB_V_10_02_01
> [2021-12-24 17:13:30.524]  Hex:    0x000a0201
> [2021-12-24 17:13:30.524]  Dec:    100201
> [2021-12-24 17:13:30.524]  
> [2021-12-24 17:13:30.524]  Hostcontroller Interface: ETHERNET!
> [2021-12-24 17:13:30.524]  
> [2021-12-24 17:13:30.524] APPLICATION RUNNING ...
> [2021-12-24 17:13:30.524] [    0.004] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
> [2021-12-24 17:13:30.524] [    0.005] [NOTICE] APL-MAIN: BOOT-CFG: FW2
> [2021-12-24 17:13:30.524] [    0.006] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EV
> [2021-12-24 17:13:30.524] [    0.006] [NOTICE] Q7KCOM:  Boot from host selected
> [2021-12-24 17:13:30.524] [    0.007] [NOTICE] SLAC: stxSLAC_Init()
> [2021-12-24 17:13:30.524] [    0.008] [NOTICE] APL-ETH:   MAC of NIC[0]:    00:01:01:63:77:30
> [2021-12-24 17:13:32.889] [    2.375] [NOTICE] APL-ETH:   MAC of NIC[1]:    00:01:01:63:77:31
> [2021-12-24 17:13:32.889] [    2.381] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
> [2021-12-24 17:13:32.905] [    2.384] [NOTICE] APL-ETH: Got link:
> [2021-12-24 17:13:32.905] [    2.385] [NOTICE] APL-ETH:     Interface: eth1
> [2021-12-24 17:13:32.905] [    2.386] [NOTICE] APL-ETH:     MAC:       00:01:01:63:77:31
> [2021-12-24 17:13:33.098] [    2.591] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
> [2021-12-24 17:13:33.114] [    2.591] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:33.114] [    2.592] [NOTICE] Q7K:   QCA700x is READY
> [2021-12-24 17:13:33.114] [    2.593] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:33.114] [    2.593] [NOTICE] Q7KCOM: Reset QCA700x
> [2021-12-24 17:13:33.114] [    2.595] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:33.114] [    2.596] [NOTICE] Q7K:   CPU ON RECEIVED
> [2021-12-24 17:13:33.114] [    2.596] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:33.114] [    2.597] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
> [2021-12-24 17:13:33.114] [    2.598] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:33.114] [    2.599] [NOTICE] Q7K:   QCA700x is READY
> [2021-12-24 17:13:33.114] [    2.599] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:33.957] [    3.437] [NOTICE] Q7KCOM:  ======================================= 
> [2021-12-24 17:13:33.957] [    3.438] [NOTICE] Q7KCOM:    FW and PIB was loaded successful 
> [2021-12-24 17:13:34.270] [    3.754] [NOTICE] APL-MAIN: 
> [2021-12-24 17:13:34.270] [    3.754] [NOTICE] APL-MAIN: =============================
> [2021-12-24 17:13:34.270] [    3.755] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6): 
> [2021-12-24 17:13:34.270] [    3.755] [NOTICE] APL-MAIN: for IP Config handle [fe01] 
> [2021-12-24 17:13:34.270] [    3.756] [NOTICE] APL-MAIN: IPv6 Addresses:
> [2021-12-24 17:13:34.270] [    3.757] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : 201:1ff:fe63:7731
> [2021-12-24 17:13:34.270] [    3.758] [NOTICE] APL-MAIN:   Local:   0:0:0:0 : 0:0:0:0
> [2021-12-24 17:13:34.270] [    3.758] [NOTICE] APL-MAIN:   DNS1:    2001:4860:4860:0 : 0:0:0:8888
> [2021-12-24 17:13:34.270] [    3.759] [NOTICE] APL-MAIN:   DNS2:    fd00:0:0:0 : be05:43ff:fe44:5280
> [2021-12-24 17:13:34.270] [    3.760] [NOTICE] APL-MAIN: =============================
> [2021-12-24 17:13:34.270] [    3.761] [NOTICE] APL-MAIN: 
> [2021-12-24 17:13:34.473] [    3.953] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:34.473] [    3.954] [NOTICE] Q7K:   CPU ON RECEIVED
> [2021-12-24 17:13:34.473] [    3.954] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:34.473] [    3.955] [NOTICE] Q7KCOM:  ======================================= 
> [2021-12-24 17:13:34.478] [    3.956] [NOTICE] Q7KCOM:    BOOT FROM HOST READY 
> [2021-12-24 17:13:34.478] [    3.956] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
> [2021-12-24 17:13:34.478] [    3.957] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:34.478] [    3.958] [NOTICE] Q7K:   QCA700x is READY
> [2021-12-24 17:13:34.481] [    3.958] [NOTICE] Q7K: =========================================
> [2021-12-24 17:13:34.544] [    4.038] [NOTICE] Q7KCOM:  Own QCA7K MAC = 00:B0:52:00:00:01
> [2021-12-24 17:13:42.272] [   11.762] [WARN  ] APL-ETH: # Unhandled Ethernet notifycode 365
> [2021-12-24 17:13:52.270] [   21.758] [NOTICE] SLAC: stxSLAC_Stop()
> [2021-12-24 17:13:52.270] [   21.761] [WARN  ] V2GCPS: stxV2GCPS_Stop() - Failed
> [2021-12-24 17:13:52.286] [   21.773] [NOTICE] SLAC: stxSLAC_Start()
> [2021-12-24 17:14:04.320] [   33.811] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
> [2021-12-24 17:14:04.320] [   33.811] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
> [2021-12-24 17:14:04.320] [   33.812] [NOTICE] V2GCPS: Duty cycle is 0 permill
[... cut ...]
> [2021-12-24 17:14:23.026] [   52.508] [WARN  ] SLAC: # SLAC_STATE_CHECK_CONN - TIMEOUT
> [2021-12-24 17:14:23.026] [   52.508] [ERROR ] SLAC: TICK (SLAC_STATE_SEND_SET_KEY_REQ): Timeout (max retries)! 
> [2021-12-24 17:14:23.026] [   52.509] [NOTICE] SLACSERV: # NC_SLAC_FAILED - 7
> [2021-12-24 17:14:23.026] [   52.518] [NOTICE] SLAC: stxSLAC_Stop()
> [2021-12-24 17:14:23.088] [   52.572] [ERROR ] V2GCPS: Failed to get duty cycle!

[image]

My question is WB-CARRIER-BOARD-PI ISO 15118(EVSE) board print message does not meet an oscilloscope monitor. why?

adirschl, 2021-12-27:
Hi Johnson,
your oscilloscope measurement seems to be in the unconnected state, the measurement of the board is in the connected state. So I assume this is the reason for the little difference in the measurement from 5,06% to 5,6%.
BR,
Andreas (CODICO)

bbr-stx, 2022-01-06:
Hi, I came back from vacation today and will have a look at it soon.

bbr-stx, 2022-01-10:
Hi, I just did a test with two boards and could see that the SLAC matching process was completed successfully.

To analyze the SLAC problem I would like to ask you to repeat the test and add logs for both sides (EV+EVSE) from USB-UART-Outputs and FreeV2G-Outputs?

---------------------------------------------
My setup was as follows:

WHITBEET-EVSE:
- Firmware V01_01_06

WHITBEET-EV:
- Firmware V01_00_03

Log-EVSE:

```
APPLICATION:
  Name:      ISO15118 (EVSE)
  Version:   V01_01_06
  Target:    WHITE-BEET
  CPU:       STM32F745

PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_00_03
 Hex:    0x000a0003
 Dec:    100003

 Hostcontroller Interface: ETHERNET!

APPLICATION RUNNING ...
[    0.026] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
[    0.033] [NOTICE] APL-MAIN: BOOT-CFG: FW2
[    0.037] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EVSE
[    0.044] [NOTICE] Q7KCOM:  Boot from host selected
[    0.051] [NOTICE] SLAC: stxSLAC_Init()
[    1.806] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
[    1.816] [NOTICE] APL-ETH: Got link:
[    1.820] [NOTICE] APL-ETH:     Interface: eth1
[    1.824] [NOTICE] APL-ETH:     MAC:       00:01:01:63:77:33
[    2.023] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    2.028] [NOTICE] Q7K: =========================================
[    2.034] [NOTICE] Q7K:   QCA700x is READY
[    2.038] [NOTICE] Q7K: =========================================
[    2.044] [NOTICE] Q7KCOM: Reset QCA700x
[    2.049] [NOTICE] Q7K: =========================================
[    2.055] [NOTICE] Q7K:   CPU ON RECEIVED
[    2.059] [NOTICE] Q7K: =========================================
[    2.066] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    2.071] [NOTICE] Q7K: =========================================
[    2.077] [NOTICE] Q7K:   QCA700x is READY
[    2.081] [NOTICE] Q7K: =========================================
[    2.988] [NOTICE] Q7KCOM:  =======================================
[    2.994] [NOTICE] Q7KCOM:    FW and PIB was loaded successful
[    3.503] [NOTICE] Q7K: =========================================
[    3.509] [NOTICE] Q7K:   CPU ON RECEIVED
[    3.513] [NOTICE] Q7K: =========================================
[    3.519] [NOTICE] Q7KCOM:  =======================================
[    3.525] [NOTICE] Q7KCOM:    BOOT FROM HOST READY
[    3.531] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    3.536] [NOTICE] Q7K: =========================================
[    3.542] [NOTICE] Q7K:   QCA700x is READY
[    3.546] [NOTICE] Q7K: =========================================
[    3.681] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:00:00:00:01
[    4.758] [NOTICE] APL-ETH: Got link:
[    4.762] [NOTICE] APL-ETH:     Interface: eth0
[    4.766] [NOTICE] APL-ETH:     MAC:       00:01:01:63:77:32
[    5.740] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[   10.853] [NOTICE] SLAC: stxSLAC_Stop()
[   10.905] [WARN  ] V2GCPS: stxV2GCPS_Stop() - Failed
[   11.108] [NOTICE] SLAC: stxSLAC_Start()
[... 133 lines cut ...]
[   20.754] [NOTICE] APL-MAIN:   DNS1:  0:0:0:0 : 0:0:0:0
[   20.755] [NOTICE] APL-MAIN:   DNS2:  0:0:0:0 : 0:0:0:0
[   20.756] [NOTICE] APL-MAIN: =============================
[   20.756] [NOTICE] APL-MAIN:
[   29.290] [NOTICE] SLAC: stxSLAC_Stop()
[   29.341] [WARN  ] V2GCPS: stxV2GCPS_Stop() - Failed
[   29.533] [NOTICE] V2GCPS: Changed Max Current EV State (1)
[   29.533] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   29.534] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   29.535] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[   29.535] [NOTICE] V2GCPS: Duty cycle is 57 permill
[   29.599] [NOTICE] SLAC: stxSLAC_Start()
[   36.687] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[   36.795] [WARN  ] SLAC: # SLAC_STATE_RECV_SLAC_PARAM_CNF - TIMEOUT
[   40.004] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[   40.158] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[   40.245] [NOTICE] APL-ETH: Got link:
[   40.245] [NOTICE] APL-ETH:     Interface: eth0
[   40.246] [NOTICE] APL-ETH:     MAC:       00:01:01:63:77:30
[   40.247] [NOTICE] SLACSERV: # NC_SLAC_SUCCESS
[   40.500] [NOTICE] V2GSVC: Starting in EV mode.
[   40.501] [INFO  ] V2GSVC: State transition from "INITIALIZED" to "EV_SDP_REQUEST".
[   40.502] [INFO  ] V2GSVC: State transition from "EV_SDP_REQUEST" to "EV_SDP_WAIT_RESPONSE".
[   40.503] [INFO  ] V2GSVC: Sent SDP request
```

johnsonWu0528, 2022-01-11:
HI Bjoern,

I follow your recommendation and attachments are test logs.

Johnson

zhiduozhang, 2022-01-13:
Hi,

I'm experiencing a similar issue, with SLAC matching failing. I've been able to bypass the SLAC matching by setting validation to false on the EVSE side, however I'm not seeing the V2GSVC messages on the debug module despite the 0x00 response codes on executing EV side V2G module initialisation commands.

Not sure if they're interconnected.

Thanks,
Duo

bbr-stx, 2022-01-13:
Hi @johnsonWu0528,

unfortunately attachments from mails are not taken over. Thus, I do not have access to the logs you sent.

Can you please add the logs to the issue in GitHub?

johnsonWu0528, 2022-01-18:
HI ,

We  send the attached updated file with PIB and MAC files into the board using the command . That step is a suggestion from CODICO - Vasily Budko and he also provides a config update file for us. Currently WB-CARRIER-BOARD-EI ISO 15118(EVSE) module and WB-CARRIER-BOARD-PI ISO 15118(EVSE) module work fine, thank you for your help.

Johnson

closed 2022-01-18 by bbr-stx