Sevenstax/FreeV2G #27: SPI interface issue- WhiteBeet

PawelBiela3, 2022-01-11:
Hello ,

I'm beginning adventure with White Beet and at the start I have issue with simple communication over SPI.
I have evalboard WB-CARRIER-BOARD-PI(EV and ISO15118) and I configured him to SPI interface, power supply over USB.
After sending simple SPI transfer, device doesn't react. What I found on user Manual, line after chip select to low, should set line RxReady also to low and in during transfer of data on MOSI line, on MISO we should also see some transfer of data. I attached images and logs(Saleae) from this situation. In my case is completly silence

After reset I have such traces and it is all:

APPLICATION:<\n>
  Name:      ISO15118 (EV)<\n>
  Version:   V01_00_03<\n>
  Target:    WHITE-BEET<\n>
  CPU:       STM32F745<\n>
 <\n>
PRODUCT VERSION:<\n>
 String: SEVENSTAX_LIB_V_10_02_01<\n>
 Hex:    0x000a0201<\n>
 Dec:    100201<\n>

 <\n>
 Hostcontroller Interface: SPI!<\n>
 <\n>
APPLICATION RUNNING ...<\n>
[    0.004] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host<\n>
[    0.005] [NOTICE] APL-MAIN: BOOT-CFG: FW2<\n>
[    0.005] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EV<\n>
[    0.006] [NOTICE] Q7KCOM:  Boot from host selected<\n>
[    0.007] [NOTICE] SLAC: stxSLAC_Init()<\n>
[    0.008] [NOTICE] APL-ETH:   MAC of NIC[0]:<9>c4:93:00:21:55:de<\n>

[    5.286] [NOTICE] APL-ETH:   MAC of NIC[1]:<9>c4:93:00:21:55:dd<\n>
[    5.292] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()<\n>
[    5.502] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55<\n>
[    5.502] [NOTICE] Q7K: ==
=======================================<\n>
[    5.503] [NOTICE] Q7K:   QCA700x is READY<\n>
[    5.504] [NOTICE] Q7K: =========================================<\n>
[    5.504] [NOTICE] Q7KCOM: Reset QCA700x<\n>
[    5.506] [NOTICE] Q7K: =========================================<\n>
[    5.507] [NOTICE] Q7K:   CPU ON RECEIVED<\n>
[    5.507] [NOTICE] Q7K: =========================================<\n>
[    5.508] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55<\n>
[    5.509] [NOTICE] Q7K: =========================================<\n>
[    5.510] [NOTICE] Q7K:   QCA700x is READY<\n>
[    5.510] [NOTICE] Q7K: =========================================<\n>
[    6.348] [NOTICE] Q7KCOM:  ======================================= <\n>
[    6.348] [NOTICE] Q7KCOM:    FW and PIB was loaded successful <\n>
[    6.863] [NOTICE] Q7K: =========================================<\n>
[    6.864] [NOTICE] Q7K:   CPU ON RECEIVED<\n>
[    6.865] [NOTICE] Q7K: =========================================<\n>
[    6.865] [NOTICE] Q7KCOM:  ======================================= <\n>
[    6.866] [NOTICE] Q7KCOM:    BOOT FROM HOST READY <\n>
[    6.867] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55<\n>
[    6.868] [NOTICE] Q7K: =========================================<\n>
[    6.868] [NOTICE] Q7K:   QCA700x is READY<\n>
[    6.869] [NOTICE] Q7K: ===============
==========================<\n>
[    7.035] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:00:21:55:D
[image]
[image]

bbr-stx, 2022-01-12:
Hello,

I was able to reproduce the error, but unfortunately there is no workaround to get it working with the current version. We are working on a solution and will provide a new version soon.

PawelBiela3, 2022-01-12:
Hello,

It means, current version of software on STM32F745VGH6?! What I understood from datasheet is possible flash new version by SWD, because we have no ethernet connection on our PCB's only SPI is available. After flashing new firmware to STM is needed use extra python tools(ethernet) to do something extra ?? Or, is needed only new firmware on STM.

PawelBiela3, 2022-01-12:
> Hello,
> 
> I was able to reproduce the error, but unfortunately there is no workaround to get it working with the current version. We are working on a solution and will provide a new version soon.

It is possible estimate date of release of new version?

bbr-stx, 2022-01-13:
According to current planning, there will be a new version next week.

Joenoodoe, 2022-01-21:
Hi @bbr-stx 

We have the same problem with SPI interface.
When will the new firmware version be released ?

Thanks.

ttu-stx, 2022-01-21:
Next EVSE Release (also including Plug&Charge) is planned for coming month.
BR
Tolga

bbr-stx, 2022-01-26:
Hi,

the new EV firmware with SPI support should be available now.

PawelBiela3, 2022-01-26:
> Hi,
> 
> the new EV firmware with SPI support should be available now.

Hi,
I'm testing new firmware (ISO15118 (EV)) V01_00_04. It looks better, I see some reaction of device.
But it is not what I expected.(image).On the MISO I see not properly bytes. Line RX_ready is also not stabil.
Question. Here are needed some pullups/pulldowns on the SPI lines??? 
Tomorrow, I'll check again my connections to evalboard. Maybe GND? Do You have some hints?

[image]
.

bbr-stx, 2022-01-27:
Hi,
for me it looks like there is a problem with the chip select timing. Maybe the 20 µs mentioned in the Whitebeet manual do not fit anymore.

Could you please try to increase the interval from chip select low to the start of data transmission (CLK) to 40 µs.
I suspect that this will fix the problem.

PawelBiela3, 2022-01-27:
> Hi, for me it looks like there is a problem with the chip select timing. Maybe the 20 µs mentioned in the Whitebeet manual do not fit anymore.
>
> Could you please try to increase the interval from [...]

I don't think, because actual we have 75 µs. I'll check double wiring connections.

PawelBiela3, 2022-01-27:
Hi,

I resolved issue. I changed place of GND on our PCB and WhiteBeet start correctly react.
Many Thanks for Your support and SW :)!
[image]

closed 2022-01-28 by bbr-stx