Sevenstax/FreeV2G #28: SPI interface RDY issiue on Whitebeet-EI

ElectronicMinds-uk, 2022-01-13:
There appears to be 2 issues regarding the RDY signal for SPI communication on Whitebeet EI

The first, if I am loading code via the JTAG to my host processor with the RDY pin connected, then it occasionally fails to function at all, remaining low all the time. Resetting the board and powering it off does nothing and it will not function again until it has been left powered off for some time ( over 30 minutes ).

The second

The second issue is during operation where the RDY signal stays asserted when I believe it should not. I attach a screen grab from my oscilloscope showing this operation.

 
[image]

[image]

D0 MISO
D4 RDY
D5 MOSI
D6 NSS
D7 CLK

 

As you can see, after completion of a transaction the RDY pin stays high so the next transaction begins but the header from the whitebeet is not populated in a meaningful way and then the payload is filled with 0xaa. My code checks the RDY before starting the transaction using it to throttle the SPI link, as it is already high then the CS deasserted is quite short.

bbr-stx, 2022-01-13:
Thank you for the detailed description. I will have a look at the problem soon.

bbr-stx, 2022-02-09:
Hi, could you please try the following two things.

1.) Increase the delay between setting NSS low and starting the CLK. Please try with a delay of 100 µs.

2.) Wait a short time before the next SPI transfer after the Rx-Ready pin goes High.

bbr-stx, 2022-03-07:
I assume that the problem has been fixed by now. If the problem is still not solved, you are welcome to reopen the ticket.

closed 2022-03-07 by bbr-stx