Sevenstax/FreeV2G #30: Session Stop when setting discovery charge parameters

Joenoodoe, 2022-01-13:
Hello,

We connected module-EI and module-PI using ethernet. 

[Python code]
module-EI: https://github.com/Sevenstax/FreeV2G
module-PI: https://github.com/lukhof/FreeV2G/tree/EV-support

[Situation]
Now, After module-EI send  "Set_Discovery_Charge_Parameters" command, each board received "Session Stop" command from each module.

[Question]
1. Do we need to set V2G configurations additionally in each module before testing ?
2. Is there any reasons cause this issue ?

Please refer to the attachments including Module debug, Application debug and Wireshark debug

BR
JoeChen
[PEV_DBG.zip](https://github.com/Sevenstax/FreeV2G/files/7862124/PEV_DBG.zip)
[EI_DBG.zip](https://github.com/Sevenstax/FreeV2G/files/7862131/EI_DBG.zip)

jpo-stx, 2022-01-13:
Hi @Joenoodoe ,

have you changed the default configuration of EV and EVSE? In this case, can you please provide your changes to the sources as a patch file.

What I have seen is that you set the battery capacity to -15536Wh. The exponential value is a signed short and `c3 50 00` is a negative value. If you want to set the battery capacity to 50kWh use 50 for the value and 3 for the exponent: `00 32 03`.

Joenoodoe, 2022-01-14:
Hi @jpo-stx 

1. No, we didn't changed the default configurations of EV and EVSE.

2. After modify the code for changing the battery capacity( **Payload**: 0xc4, 0x93, 0x0, 0x21, 0x55, 0xc5, 0x2, 0x0, 0x1, 0x1, 0x0, 0x1, 0x0, 0x0, 0x32, 0x3), the error still show up. Trace the document and EV code
    a. Tracing the document I found that:  EV -> Set configuration command
        The document shows two byte value (C3 50) which is different from you said. Which is the real value we need to send ?
        
    b. Tracing the code I found that: EV -> Whitebeet.py -> Set configuration command
        The code didn't caculate the exponent value. Have the code been test already ?

              if isinstance(config["battery_capacity"], int):
                payload += config["battery_capacity"].to_bytes(2, "big")
                payload += b"\x00"
            else:
                payload += config["battery_capacity"][0].to_bytes(2, "big")
                payload += config["battery_capacity"][1].to_bytes(1, "big")

3. Currently, we just want to test the whole procedures including CP, SLAC and V2G using Ethernet. Could you provide the working code for each module-EI and module-PI ? 

BR
JoeChen

Joenoodoe, 2022-01-14:
The firmware version for PEV and EVSE side.

[WB-module-EI Side]
APPLICATION:
  Name:      ISO15118 (EVSE)
  Version:   V01_01_02
  Target:    WHITE-BEET
  CPU:       STM32F745
 
PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_00_03
 Hex:    0x000a0003
 Dec:    100003

[WB-module-PI Side]
APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_03
  Target:    WHITE-BEET
  CPU:       STM32F745
 
PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_02_01
 Hex:    0x000a0201
 Dec:    100201

jpo-stx, 2022-01-17:
Hi @Joenoodoe , my collegue is currently analysing the issue. I will come back to you with the results

jpo-stx, 2022-01-17:
In the meantime, can you update your EVSE to the latest firmware version V01_01_06 from december 2021.

Joenoodoe, 2022-01-18:
Hi @jpo-stx,

We test the whole procedures successfully between module-EI and module-PI after modifying both code.
This issue can be closed for sure.

If we have problems, we'll open another issue in github.

Thanks for help and analyzing.

BR.
JoeChen

jpo-stx, 2022-01-18:
@Joenoodoe nice to hear that you could could solve the issue.
Nevertheless, we will update FreeV2G so that it is working out of the box. Sorry for the inconvenience.

closed 2022-01-18 by jpo-stx