Sevenstax/FreeV2G #31: Session Stops after "Set EVSE ID" when Hyundai Kona is connected

nikhilnezodi, 2022-01-14:
Hello All,
I have been working on the WB-CARRIER-BOARD-EI ISO 15118(EVSE) module for a while. The vehicle I tested on is Hyundai Kona. On running the Application.py file and observing the logs the session stops after "SET EVSE ID". The format is "0" so I set the EVSE ID in "+49*XXX*XXX*XXX" format as mentioned in the Sevenstax Usermanual. 
This seemed similar to issue #16 so I tried by changing Energy Transfer modes. I also tried to set EVSE ID as 0 which also did not help in the process. I had not faced any such issues while testing with Tata Nexon.

I have attached screenshots for the log observations - 
1. Initial Log
[image]
 
2. When EVSE ID was set to "0"
[image]

3. When Energy Transfer Mode was set to "1"
[image]

4. When Transfer modes were set as [0,1,2,3]
[image]

jpo-stx, 2022-01-19:
@nikhilnezodi , it seems a bit strange that the session id is all zeros. Can you activate the port mirror and do a wireshark log of the ethernet interface? See Whitebeet User Manual chapter 12.3.10.

nikhilnezodi, 2022-01-26:
@jpo-stx I have attached the screenshot of the latest logs. 
[image]

Also activated the port mirror and attached the wireshark logs obtained.
[Hyundai_Kona_wireshark_logs.zip](https://github.com/Sevenstax/FreeV2G/files/7938823/Hyundai_Kona_wireshark_logs.zip)

jpo-stx, 2022-01-26:
Hi @nikhilnezodi , are you sure that you activated the port mirror? How is your setup, are you connected directly to the Whitebeet or do you have a switch in between?
The EV wants to charge with the DIN protocol, there is an issue when the energy transfer modes are not set correctly. Please check my comment in https://github.com/Sevenstax/FreeV2G/issues/16#issuecomment-988695769 and chapter **17.10.7 Set energy transfer modes** in the manual and verify that you have set the modes correctly.

nikhilnezodi, 2022-02-03:
@jpo-stx Thanks. Setting the Energy transfer mode to 1 (DC Extended) worked for me.

closed 2022-02-03 by nikhilnezodi