Sevenstax/FreeV2G #35: Start charging session with whitebeet EI and PI

markus-kl, 2022-02-02:


jpo-stx, 2022-02-03:
Can you please share the logs of the whitebeet debug uart port?

Which firmware versions are running on your whitebeets? The latest versions are 1.1.6 for EVSE and 1.0.4 for EV.

Btw: the EV functionality was merged to the FreeV2G respository today.

markus-kl, 2022-02-03:
EV has V 1.0.3, i can not find an updatefile at codico website. EVSE i updated to 1.1.6
[output_2022-02-03_15-51-42.log](https://github.com/Sevenstax/FreeV2G/files/7995664/output_2022-02-03_15-51-42.log)
.

adirschl, 2022-02-03:
You should find the update file now in your CODICO PLC download area. Unfortunately your account was configured only for WHITEbeet EVSE side, not PEV side. Now I enabled it. BR Andreas Dirschl, CODICO

markus-kl, 2022-02-07:
It worked well with the latest Version of FreeV2G from Github and my EV with FW 1.0.3. It simulated a charging session without an error. After updating to 1.0.4 i have an error with timeout again. This is similar to what i described at first. Please check the log of EVSE.
[output_2022-02-07_10-23-58.log](https://github.com/Sevenstax/FreeV2G/files/8013987/output_2022-02-07_10-23-58.log)

jpo-stx, 2022-02-08:
Hello @markus-kl , so now you are using EV with 1.0.4, an EVSE with 1.1.6.
Are you using the latest github version of FreeV2G for EV and EVSE?
Can you please check this.

markus-kl, 2022-02-08:
Hello, of course i have updated and now i am using the latest Versions as i already wrote. For the first and only time it worked with EV FW 1.0.3, EVSE FW 1.1.6, but now i updated EV to 1.0.4. For the moment it would help to roll back to EV 1.0.3 until FW 1.0.4 is fixed, so maybe you can provide this FW file.

adirschl, 2022-02-08:
Hello @markus-kl, please check your codico PLC Documentation Area. In the USER_FILES folder I provided the FW_1.0.3 update file.

jpo-stx, 2022-02-11:
Hello @markus-kl , I have asked again because the setup you described is tested and should be working. Can you describe your setup, how the whitebeets are connected to their host applications. Are they connected directly or do you use a switch, Both host applications on the same PC?

adirschl, 2022-02-11:
Hello, I tested as well with EVSE FW V01.01.06 and EV FW V01.00.04 and can confirm it works.
Andreas Dirschl - CODICO

markus-kl, 2022-02-17:
I tried on a different pc, it also doesnt work. It only worked one time, before and after everytime timeout or and other errors. Setup is still the same. Yes, Both host applications on the same PC. I have connected one board directly to the pc via ethernet port. The other board is connected to the ethernetport of my usb-c hub.

jpo-stx, 2022-03-29:
Were you able to get your setup running? If not please enable the port mirror on one of you boards and do a wireshark trace. See manual chapter "13.3.10 Set Port Mirror State" for how to activate it.

lho-stx, 2022-05-02:
I assume this issue is solved, if not please re-open.

closed 2022-05-02 by lho-stx