Sevenstax/FreeV2G #37: Missing Status Messages V2G Schedule Received and DC Charge Parameters Changed in during V2G start

PawelBiela3, 2022-02-08:
Hello,

I'm using evalboard ISO 15118 EV and SW V_01_00_04. and SPI interface.
I started V2G session (Chapter 19.9.2 Start Session) and I observed that WhiteBeet doesn't send extra two status messages.
Additionally WhiteBeet should send  also **Cable Check Ready** . I'm noticing only positive respond "Session Started" and it is all.
In logs from device I see such events, but it looks, that this events are not forward to SPI interface. 
I did small experiment and I sent "Start Cable Check" but WhiteBeet nothing answer.

**Traces from WhiteBeet**
APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_04
  Target:    WHITE-BEET
  CPU:       STM32F745
 
PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_02_01
 Hex:    0x000a0201
 Dec:    100201
 
 Hostcontroller Interface: SPI!
 
APPLICATION RUNNING ...
[    0.005] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
[    0.005] [NOTICE] APL-MAIN: BOOT-CFG: FW2
[    0.006] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EV
[    0.008] [NOTICE] Q7KCOM:  Boot from host selected
[    0.012] [NOTICE] SLAC: stxSLAC_Init()
[    0.013] [NOTICE] APL-ETH:   MAC of NIC[0]:	c4:93:00:21:55:de
[    5.291] [NOTICE] APL-ETH:   MAC of NIC[1]:	c4:93:00:21:55:dd
[    5.297] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
[    5.510] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    5.511] [NOTICE] Q7K: =========================================
[    5.511] [NOTICE] Q7K:   QCA700x is READY
[    5.512] [NOTICE] Q7K: =========================================
[    5.513] [NOTICE] Q7KCOM: Reset QCA700x
[    5.515] [NOTICE] Q7K: =========================================
[    5.515] [NOTICE] Q7K:   CPU ON RECEIVED
[    5.516] [NOTICE] Q7K: =========================================
[    5.517] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    5.518] [NOTICE] Q7K: =========================================
[    5.519] [NOTICE] Q7K:   QCA700x is READY
[    5.519] [NOTICE] Q7K: =========================================
[    6.385] [NOTICE] Q7KCOM:  ======================================= 
[    6.386] [NOTICE] Q7KCOM:    FW and PIB was loaded successful 
[    6.901] [NOTICE] Q7K: =========================================
[    6.901] [NOTICE] Q7K:   CPU ON RECEIVED
[    6.902] [NOTICE] Q7K: =========================================
[    6.903] [NOTICE] Q7KCOM:  ======================================= 
[    6.903] [NOTICE] Q7KCOM:    BOOT FROM HOST READY 
[    6.905] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    6.905] [NOTICE] Q7K: =========================================
[    6.906] [NOTICE] Q7K:   QCA700x is READY
[    6.907] [NOTICE] Q7K: =========================================
[    7.073] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:00:21:55:DF
[   18.290] [NOTICE] SLAC: stxSLAC_Start()
[   29.428] [NOTICE] V2GCPS: Changed Max Current EV State (1)
[   29.428] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   29.429] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   29.430] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[   29.430] [NOTICE] V2GCPS: Duty cycle is 56 permill
[   29.495] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[   29.602] [WARN  ] SLAC: # SLAC_STATE_RECV_SLAC_PARAM_CNF - TIMEOUT
[   32.809] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[   35.628] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[   35.652] [NOTICE] APL-ETH: Got link:
[   35.653] [NOTICE] APL-ETH:     Interface: eth0
[   35.654] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:55:DE
[   35.654] [NOTICE] SLACSERV: # NC_SLAC_SUCCESS
[   35.973] [NOTICE] V2GSVC: Starting in EV mode.
[   37.350] [NOTICE] APL-MAIN: 
[   37.350] [NOTICE] APL-MAIN: =============================
[   37.351] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6): 
[   37.351] [NOTICE] APL-MAIN: for IP Config handle [fe00] 
[   37.352] [NOTICE] APL-MAIN: IPv6 Addresses:
[   37.353] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe21:55de
[   37.354] [NOTICE] APL-MAIN:   Local:	0:0:0:0 : 0:0:0:0
[   37.354] [NOTICE] APL-MAIN:   DNS1:	2001:4860:4860:0 : 0:0:0:8888
[   37.355] [NOTICE] APL-MAIN:   DNS2:	fd00:0:0:0 : be05:43ff:fe44:5280
[   37.356] [NOTICE] APL-MAIN: =============================
[   37.357] [NOTICE] APL-MAIN: 
[   37.608] [WARN  ] V2GSVC: Security options satisfied, connect!
[   37.615] [NOTICE] V2GSVC: TCP: Connected to server without TLS
[   39.542] [NOTICE] V2GAPP: Received event 'Session started'
**[   40.355] [NOTICE] V2GAPP: Received event 'Schedule received'
[   40.356] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   40.357] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   40.358] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   40.359] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'**

Last sniffs from V2G session
**1. EV request **
 [-] 10.694932   Eth 1   Charge Parameter Discovery Request   Rx    v2g        FE80::C693:FF:FE21:55DE   FE80::11         57843         51111              TP version: 1, Session ID: 17608499347531563676   52              
       |  Session ID                    17608499347531563676   
      [-] DC_EVChargeParameter          
           [-] DC_EVStatus             
                 |  EVReady       TRUE       
                 |  EVErrorCode   NO_ERROR   
                 |  EVRESSSOC     50 %       
            |  EVMaximumCurrentLimit   50 A       
            |  EVMaximumPowerLimit     20000 W    
            |  EVMaximumVoltageLimit   400 V      
            |  EVEnergyCapacity        50000 Wh   
            |  EVEnergyRequest         25000 Wh   
            |  FullSOC                 100 %      
            |  BulkSOC                 80 %       
       |  RequestedEnergyTransferMode   DC_core                
       |  MaxEntriesSAScheduleTuple     5                      

 **2. EVSE response**
 [-] 10.995293   Eth 1   Charge Parameter Discovery Response (OK)   Tx    v2g        FE80::11    FE80::C693:FF:FE21:55DE   51111         57843              TP version: 1, Session ID: 17608499347531563676   253             
       |  Session ID               17608499347531563676   
       |  ResponseCode             OK                     
       |  EVSEProcessing           Finished               
      [-] SAScheduleList           
           [+] SAScheduleTuple (0)   
           [+] SAScheduleTuple (1)   
      [-] DC_EVSEChargeParameter   
           [-] DC_EVSEStatus                    
                 |  NotificationMaxDelay   0            
                 |  EVSEIsolationStatus    Invalid      
                 |  EVSEStatusCode         EVSE_Ready   
                 |  EVSENotification       None         
            |  EVSEMaximumCurrentLimit          700 A      
            |  EVSEMaximumPowerLimit            500000 W   
            |  EVSEMaximumVoltageLimit          920 V      
            |  EVSEMinimumCurrentLimit          0 A        
            |  EVSEMinimumVoltageLimit          0 V        
            |  EVSECurrentRegulationTolerance   50 A       
            |  EVSEPeakCurrentRipple            12 A       
            |  EVSEEnergyToBeDelivered          25000 Wh

PawelBiela3, 2022-02-09:
I checked this scenario over ethernet, 
I used Your EV python application and also I'm stoping in the same place

**PYTHON**
Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: Ethernet 4, mac: C4:93:00:21:55:DD
EV configuration: {'mac': 'C4:93:00:21:55:DD', 'ev': {'evid': '00:01:01:63:77:31', 'protocol_count': 2, 'protocols': [0, 1], 'payment_method_count': 1, 'payment_method': [0], 'energy_transfer_mode_count': 2, 'energy_transfer_mode': [0, 4]}, 'battery': {'timestep': 10000, 'capacity': 50000, 'level': 49000, 'full_soc': 100, 'bulk_soc': 80, 'max_current': 100, 'max_power': 12000, 'max_voltage': 300, 'target_current': 40, 'target_voltage': 200, 'target_voltage_delta': 10, 'max_current_AC': 0, 'max_voltage_AC': 0, 'min_current_AC': 0}}
Set the CP mode to EV
Start the CP service
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Set AC charging parameters
Change State to State C
Create new charging session
ControlPilot duty cycle: 5.6
"Session started" received
        Protocol: 1
        Session ID: 208e84ff40782599
        EVSE ID: 44452a5645432a455637373731412a3700000000000000000000000000000000000000000000
        Payment method: 0
        Energy transfer mode: 0

**TRACES**
[    2.789] [NOTICE] Q7KCOM:  ======================================= 
[    2.790] [NOTICE] Q7KCOM:    FW and PIB was loaded successful 
[    3.241] [NOTICE] APL-MAIN: 
[    3.241] [NOTICE] APL-MAIN: =============================
[    3.242] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6): 
[    3.242] [NOTICE] APL-MAIN: for IP Config handle [fe01] 
[    3.243] [NOTICE] APL-MAIN: IPv6 Addresses:
[    3.244] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe21:55dd
[    3.245] [NOTICE] APL-MAIN:   Local:	0:0:0:0 : 0:0:0:0
[    3.245] [NOTICE] APL-MAIN:   DNS1:	2001:4860:4860:0 : 0:0:0:8888
[    3.246] [NOTICE] APL-MAIN:   DNS2:	fd00:0:0:0 : be05:43ff:fe44:5280
[    3.247] [NOTICE] APL-MAIN: =============================
[    3.248] [NOTICE] APL-MAIN: 
[    3.304] [NOTICE] Q7K: =========================================
[    3.305] [NOTICE] Q7K:   CPU ON RECEIVED
[    3.306] [NOTICE] Q7K: =========================================
[    3.307] [NOTICE] Q7KCOM:  ======================================= 
[    3.307] [NOTICE] Q7KCOM:    BOOT FROM HOST READY 
[    3.309] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    3.309] [NOTICE] Q7K: =========================================
[    3.310] [NOTICE] Q7K:   QCA700x is READY
[    3.310] [NOTICE] Q7K: =========================================
[    3.478] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:00:21:55:DF
[   11.249] [WARN  ] APL-ETH: # Unhandled Ethernet notifycode 365
[  124.142] [NOTICE] SLAC: stxSLAC_Stop()
[  124.144] [WARN  ] V2GCPS: stxV2GCPS_Stop() - Failed
[  124.157] [NOTICE] SLAC: stxSLAC_Start()
[  124.162] [NOTICE] V2GCPS: Changed Max Current EV State (1)
[  124.162] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  124.163] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  124.164] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  124.164] [NOTICE] V2GCPS: Duty cycle is 56 permill
[  131.164] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[  131.272] [WARN  ] SLAC: # SLAC_STATE_RECV_SLAC_PARAM_CNF - TIMEOUT
[  134.480] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[  134.670] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[  134.765] [NOTICE] APL-ETH: Got link:
[  134.766] [NOTICE] APL-ETH:     Interface: eth0
[  134.766] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:55:DE
[  134.767] [NOTICE] SLACSERV: # NC_SLAC_SUCCESS
[... 103 lines cut ...]
[  138.239] [NOTICE] V2GCPS: Duty cycle is 56 permill
[  138.442] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  138.442] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  138.443] [NOTICE] V2GCPS: Duty cycle is 55 permill
[  138.544] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  138.544] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  138.545] [NOTICE] V2GCPS: Duty cycle is 56 permill
[  138.638] [NOTICE] V2GAPP: Received event 'Session started'
[  138.697] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  138.697] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  138.698] [NOTICE] V2GCPS: Duty cycle is 55 permill
[  138.850] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  138.850] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  138.851] [NOTICE] V2GCPS: Duty cycle is 56 permill
[  139.207] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  139.207] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  139.208] [NOTICE] V2GCPS: Duty cycle is 55 permill
[  139.309] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  139.309] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  139.310] [NOTICE] V2GCPS: Duty cycle is 56 permill
[  139.451] [NOTICE] V2GAPP: Received event 'Schedule received'
[  139.452] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  139.452] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  139.453] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  139.454] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'

jpo-stx, 2022-02-11:
Hello @PawelBiela3 , when the EVSE has finished the charge parameter discovery the cable check ready status message should be received. Can you please activate the port mirror and do a wireshark trace of the ethernet interface? Please refer to the manual chapter 12.3.10 - 12.3.13 for port mirror activation and configuration.

PawelBiela3, 2022-02-11:
Hello, 

I repeated exercise, please look to sniffs
**Timestamp (V2G Session Started Succesfully):** 
276	66.532647	8Devices_21:55:dd	Dell_05:47:1e	DEC DNA	76			msg nr. 3623: continuation segment , bytes this segment: 43, total so far:5423

**+Traces from WhiteBeet**
Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: Ethernet 4, mac: C4:93:00:21:55:DD
EV configuration: {'mac': 'C4:93:00:21:55:DD', 'ev': {'evid': '00:01:01:63:77:31', 'protocol_count': 2, 'protocols': [0, 1], 'payment_method_count': 1, 'payment_method': [0], 'energy_transfer_mode_count': 2, 'energy_transfer_mode': [0, 4]}, 'battery': {'timestep': 10000, 'capacity': 50000, 'level': 49000, 'full_soc': 100, 'bulk_soc': 80, 'max_current': 100, 'max_power': 12000, 'max_voltage': 300, 'target_current': 40, 'target_voltage': 200, 'target_voltage_delta': 10, 'max_current_AC': 0, 'max_voltage_AC': 0, 'min_current_AC': 0}}
Set the CP mode to EV
Start the CP service
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Set AC charging parameters
Change State to State C
Create new charging session
ControlPilot duty cycle: 5.6
"Session started" received
        Protocol: 1
        Session ID: 5096d0906cae27ee
        EVSE ID: 44452a5645432a455637373731412a3700000000000000000000000000000000000000000000
        Payment method: 0
        Energy transfer mode: 0

jpo-stx, 2022-02-11:
Thanks @PawelBiela3 I will check the log!

PawelBiela3, 2022-02-18:
> Hello @PawelBiela3 , to be clear, in the logs we see the events for 'Session started', 'Schedule received' and 'DC charge parameters changed' but do you receive these messages on the SPI interface, or is the Whitebeet completely silent regarding the event messages?

Hello, Issue is, that on device(logs) we see all events. But WhiteBeet send to host only one message 'postive Session started', Situation is the same when I'm using SPI and Ethernet(python app).  Missing are 'Schedule received' ,'DC charge parameters changed and "Cable Check Ready",completely silent.(in case of SPI, TX pending is low,in case of ethernet I don't see events on wireshark)

What I see on traces, and what I suppose, that this events are detect every 1ms, maybe some queue is blocked.?!
Do You have some unofficial SW with more debugs to identify what is exactly going on device?? 
Could You check source code in this area, maybe we have some unfulfilled conditional??? . For example , I detected issue with payment and immediately WhiteBeet reported issue. Here situation is other, it looks(base on traces), that all was ok, but WhiteBeet after this event is unresponsive.
I have possiblity control of parameters of EVSE, but I don't know, what exactly I have to set to get around problem and continue work. 

[ 139.451] [NOTICE] V2GAPP: Received event 'Schedule received'
[ 139.452] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 139.452] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 139.453] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 139.454] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'

jpo-stx, 2022-02-18:
Hallo @PawelBiela3 ,
1. When the Whitebeet went silent, is it responding to configuration commands or other framing commands like a ping? (see 10.3.1)
2. Can you try to omit the sales tariffs in the charge parameter discovery response fro EVSE side?

PawelBiela3, 2022-02-18:
Hello @jpo-stx 
1). When I have this silent , I'm trying send 'Start Cable Check' , after 10 seconds I'm trying other messages, for example CP_set Mode, zero reaction

2) I thought also about the same, But without success, I attached Discovery Response without Sales Tariff. But I think, when will be something wrong with EVSE , White Beet should not return positive message "Start Session" 

[image]

PawelBiela3, 2022-02-18:
Hi @jpo-stx 
Behaviour is the same. What I see additionally in traces, after 20-30 seconds we have reset of WhiteBeet

[   41.686] [NOTICE] APL-MAIN: =============================
[   41.687] [NOTICE] APL-MAIN: 
[   41.938] [WARN  ] V2GSVC: Security options satisfied, connect!
[   41.945] [NOTICE] V2GSVC: TCP: Connected to server without TLS
[   43.876] [NOTICE] V2GAPP: Received event 'Session started'
[   44.687] [NOTICE] V2GAPP: Received event 'Schedule received'
[   44.687] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   44.688] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   44.689] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   44.690] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'

 
APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_04
  Target:    WHITE-BEET
  CPU:       STM32F745
 
PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_02_01
 Hex:    0x000a0201
 Dec:    100201
 
 Hostcontroller Interface: SPI!
 
APPLICATION RUNNING ...
[    0.005] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from hos
[image]
t

jpo-stx, 2022-02-21:
Hello @PawelBiela3 can you disable the EVSECurrentRegulationTolerance parameter on your EVSE? I think there is an issue with this parameter on the EV side.
BR Jan

PawelBiela3, 2022-02-21:
> Hello @PawelBiela3 can you disable the EVSECurrentRegulationTolerance parameter on your EVSE? I think there is an issue with this parameter on the EV side. BR Jan

Hello @jpo-stx  , It was not easy, but I disabled this parameter. After this WhiteBeet was unblocked, even I started DC loop without any problems. Many Thanks for hints!!!

Are You planning provide some patch for this situation???

jpo-stx, 2022-02-24:
> Hello @jpo-stx , It was not easy, but I disabled this parameter. After this WhiteBeet was unblocked, even I started DC loop without any problems. Many Thanks for hints!!!

Nice to hear that you can now continue.

> Are You planning provide some patch for this situation???

We will fix this issue with the next release!

PawelBiela3, 2022-03-14:
Hello @jpo-stx 
Do You know date, when this release will be available???

jpo-stx, 2022-04-05:
Hi @PawelBiela3 ,
the fix is already implemented and tested, but I don't know an exact date for the release.
We will announce the release in the discussions section of this repository: https://github.com/Sevenstax/FreeV2G/discussions

jpo-stx, 2022-04-26:
Hi @PawelBiela3 , please update to the new firmware version V01_00_05.

You should be able to use the current regulation tolerance parameter on your EVSE. You will receive this parameter on EV side in the "DC charge parameters changed" message. Please refer to the manual for the new parameters in the message.

closed 2022-04-26 by jpo-stx