Sevenstax/FreeV2G #38: Code is Stuck at Session start request

Shekhar-devise, 2022-02-09:
Hello,

I'm using the Gloquad PLC module to interface with the Whitebeet PEV module.
Also, we are interfacing the white beet module with an Ethernet interface.

When I have started the repository I got a response up to CP voltage.

Create a new charging session
ControlPilot duty cycle: 5.5
after this message, the white beet PEV module should send SessionStarted (C0) status and SessionSetupReq to EVSE.
But when we checked its response. we are unable to get it.
I am sending the results that we got after running the code.

Python code Output messages:-------------------------------------------------------------------
Initiating framing interface
iface: enp4s0, mac: C4:93:00:21:56:64
Ev configuration: {'mac': 'C4:93:00:21:56:64', 'ev': {'evid': 'C4:93:00:21:56:64', 'protocol_count': 2,
'protocols': [0, 1], 'payment_method_count': 1, 'payment_method': [0],
'energy_transfer_mode_count': 2, 'energy_transfer_mode': [0, 4]}, 'battery': {'timestep': 10000,
'capacity': 50000, 'level': 49000, 'full_soc': 100, 'bulk_soc': 80, 'max_current': 100, 'max_power':
12000, 'max_voltage': 300, 'target_current': 40, 'target_voltage': 200, 'target_voltage_delta': 10,
'max_current_AC': 0, 'max_voltage_AC': 0, 'min_current_AC': 0}}
Set the CP mode to EV
Start the CP service
Set SLAC to EV mode
Start SLAC
Wait until an EVSE connects
cp_dc = 5.5
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Set AC charging parameters
Start V2G
Change State to State C
controlPilotSetResistorValue res = 0
Create new charging session
ControlPilot duty cycle: 5.5

------------------------------------------------------------------------------------------------

White beet debug log

[  217.247] [NOTICE] SLAC: stxSLAC_Stop()
[  217.348] [WARN  ] V2GCPS: stxV2GCPS_Stop() - Failed
[  217.692] [NOTICE] V2GCPS: Changed Max Current EV State (7)
[  217.692] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  217.869] [NOTICE] SLAC: stxSLAC_Start()
[  221.466] [NOTICE] V2GCPS: Changed Max Current EV State (1)
[  221.466] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  221.467] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  221.468] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  221.468] [NOTICE] V2GCPS: Duty cycle is 56 permill
[  226.755] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[  226.861] [WARN  ] SLAC: # SLAC_STATE_RECV_SLAC_PARAM_CNF - TIMEOUT
[  230.068] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[  231.458] [NOTICE] Q7KCOM: PIB File was updated => Download PIB without reset
[  231.488] [NOTICE] APL-ETH: Got link:
[  231.488] [NOTICE] APL-ETH:     Interface: eth0
[  231.489] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:56:65
[  231.490] [NOTICE] SLACSERV: # NC_SLAC_SUCCESS
[  231.514] [NOTICE] V2GCPS: Changed Max Current EV State (7)
[  231.514] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  231.515] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  231.516] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[  231.516] [NOTICE] V2GCPS: Duty cycle is 1000 permill
[  232.122] [NOTICE] V2GSVC: Starting in EV mode.
[  233.389] [NOTICE] APL-MAIN: 
[... 156 lines cut ...]
[  249.469] [NOTICE] Q7K:   CPU ON RECEIVED
[  249.469] [NOTICE] Q7K: =========================================
[  249.470] [NOTICE] Q7KCOM:  ======================================= 
[  249.471] [NOTICE] Q7KCOM:    BOOT FROM HOST READY 
[  249.472] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[  249.473] [NOTICE] Q7K: =========================================
[  249.474] [NOTICE] Q7K:   QCA700x is READY
[  249.474] [NOTICE] Q7K: =========================================
[  251.490] [NOTICE] APL-ETH: Lost link:
[  251.491] [NOTICE] APL-ETH:     Interface: eth0
[  251.491] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:56:65
[  251.492] [NOTICE] APL-MAIN: 
[  251.492] [NOTICE] APL-MAIN: =============================
[  251.493] [NOTICE] APL-MAIN: LAN CONFIGURATION DELETED: 
[  251.494] [NOTICE] APL-MAIN: for IP Config handle [fe00] 
[  251.494] [NOTICE] APL-MAIN: IPv6 Addresses:
[  251.495] [NOTICE] APL-MAIN:   Link-Local:  0:0:0:0 : 0:0:0:0
[  251.496] [NOTICE] APL-MAIN:   Local: 0:0:0:0 : 0:0:0:0
[  251.497] [NOTICE] APL-MAIN:   DNS1:  0:0:0:0 : 0:0:0:0
[  251.497] [NOTICE] APL-MAIN:   DNS2:  0:0:0:0 : 0:0:0:0
[  251.498] [NOTICE] APL-MAIN: =============================
[  251.499] [NOTICE] APL-MAIN: 
---------------------------------------------------------------------------------------------------------------

Can you please help us to rectify and solve this issue.

Shekhar-devise, 2022-02-12:
I am sending the wire shark logs with code output.please check.
I have already set mirror state as per the white beet manual.
Getting same response again.

Initiating framing interface
iface: eth0, mac: C4:93:00:21:56:64
Ev configuration: {'mac': 'C4:93:00:21:56:64', 'ev': {'evid': 'C4:93:00:21:56:64', 'protocol_count': 2, 'protocols': [0, 1], 'payment_method_count': 1, 'payment_method': [0], 'energy_transfer_mode_count': 2, 'energy_transfer_mode': [0, 4]}, 'battery': {'timestep': 10000, 'capacity': 50000, 'level': 49000, 'full_soc': 100, 'bulk_soc': 80, 'max_current': 100, 'max_power': 12000, 'max_voltage': 300, 'target_current': 40, 'target_voltage': 200, 'target_voltage_delta': 10, 'max_current_AC': 0, 'max_voltage_AC': 0, 'min_current_AC': 0}}
Set the CP mode to EV
Start the CP service
Set SLAC to EV mode
Start SLAC
Wait until an EVSE connects
cp_dc = 5.5
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
get v2g configuration
get conf res = {'evid': b'\x01\x02\x03\x04\x05\x06', 'protocol_count': 2, 'protocol': [0, 1], 'payment_method_count': 1, 'payment_method': [0], 'energy_transfer_mode_count': 1, 'energy_transfer_mode': [1], 'battery_capacity': 50000}
Set DC charging parameters
Set AC charging parameters
Start V2G
Change State to State C
controlPilotSetResistorValue res = 0
Create new charging session
ControlPilot duty cycle: 100.0

[wireshark_logs.pcapng.zip](https://github.com/Sevenstax/FreeV2G/files/8053026/wireshark_logs.pcapng.zip)

Shekhar-devise, 2022-02-17:
@jpo-stx Hi have you found anything strange?

jpo-stx, 2022-02-24:
Hi @Shekhar-devise it seems that the QCA is restarted and therefore looses the link:
[ 247.356] [NOTICE] Q7K: CPU ON RECEIVED
My collegue is currently checking why the QCA is restarted.
Have you started a v2g session before the one that is shown in the log?

Shekhar-devise, 2022-03-18:
Have you found why QCA is restarting?
Have you started a v2g session before the one that is shown in the log?
yes, we have started the V2G session before the log.

jpo-stx, 2022-03-29:
Hello @Shekhar-devise ,

we were not yet able to reproduce the issue. Further Questions:

- Which firmware version are you using?
- How often does it happen?
- Is it also happening in the first session after a reboot, or only in the following sessions?

Shekhar-devise, 2022-04-01:
HI,
We are using the following firmware:
WHITEBEET_ISO15118_EV_V01_00_04.fwu

It happens every time when we are running it.

Yes it is also happening in the first senssion after a reboot.

jpo-stx, 2022-04-01:
Can you please send me a complete log of the UART and wiresshark trace from boot on?

Shekhar-devise, 2022-04-17:
Sharing a screenshot of the Firmware version that we are using
[image]
g

jpo-stx, 2022-04-19:
Hi @Shekhar-devise ,

in the second line I see that you are using the old v1 of the QCA firmware. Can you try with the new version, by setting the DIP switch SW1 to "FW VER **NEW**".

Have you switched to the old version intentionally?

Shekhar-devise, 2022-04-22:
We have changed that switch to "FW VER NEW" but the problem is the same.

jpo-stx, 2022-04-22:
Do you have the possibility to use another counterpart on EVSE side?

Since we cannot reproduce this issue here can please do what I asked for in https://github.com/Sevenstax/FreeV2G/issues/38#issuecomment-1085591446:

> Can you please send me a complete log of the UART and wiresshark trace from boot on?

Shekhar-devise, 2022-04-22:
@jpo-stx please check I have added the the whitebeet log from boot and EVSE simulator log.pleaes check
[EVSElog.txt](https://github.com/Sevenstax/FreeV2G/files/8541561/EVSElog.txt)
[PEV.txt](https://github.com/Sevenstax/FreeV2G/files/8541564/PEV.txt)
.

jpo-stx, 2022-05-02:
@Shekhar-devise can you also add the wireshark log with port mirror enabled. It is importent to have both logs of the same process. So please start logging of the debug port and the ethernet with port mirror enabled an do a clean start of the Whitebeet module and attach the files to this issue. This would be really helpful to analyze the issue.

lho-stx, 2022-10-04:
@Shekhar-devise,
as there is no progress in this issue it will be closed. If you still encounter this issue, please reopen!
Best regards,
lho

closed 2022-10-04 by lho-stx