Sevenstax/FreeV2G #40: warning at the handleRequestChargeLoopParameters

codderembedded, 2022-02-22:
Hi,

We are working on car charging system. We are testing on zoe vehicle. We use freeV2G. We are coming to stage of request charge loop parameters. But we are getting the warning on v2gSetDcChargeLoopParameters.
Can you look at the txt? And Why are we getting the warning?

[Log-21.txt](https://github.com/Sevenstax/FreeV2G/files/8116783/Log-21.txt)

klth, 2022-02-23:
Hi, 

first, I don't use the python script.

but, the loading process seems to be aborted because of a timeout.
`[   95.680] [NOTICE] V2GDIN12: Message performance timeout of 75ms reached.`

The loading loop needs a response within the specified time, see manual 17.12.11 parameter timeout.

In your case, the answer comes after this timeout. At this point, the Whitebeet is no longer in the charge loop state because it is already terminating the process.

The reaction from the whitebeet is: 
`Warning: Module did not accept command 27:70, return code: 5`
`[   95.729] [ERROR ] V2GAPP: Not in the state to receive this message`

Return code 5 means (manual 10.1.5):
` 0x05  Unexpected message, module is not in a state to execute the command`

what kind of machine is the script running on? raspberry pi? 
maybe try running the script in a higher priority?

jpo-stx, 2022-02-24:
@codderembedded , @klth is right, you get the timeout because you do not respond to the request message in time.
If you are on a linux system you can try with the branch https://github.com/Sevenstax/FreeV2G/tree/eth_pylibpcap which uses the pylibcap directly instead of using scapy to receive ethernet frames.

codderembedded, 2022-02-25:
Hi, thank you. We changed scapy with pylibcap. and we tryed again. After a few times charge loop we got an error. And we changed ethernet cable with cat7. And it worked for 20 seconds. and we deleted all of prints. it worked nicely. we use super 10.1" PPC with linux ubuntu.

duverg, 2022-02-26:
Hi Sevenstax,

( updated from issue #41 )

We are test the branch eth_pylibpcap intead of master (#41), but although we see much more exchange of information we obtain the same warning output:
```
Warning: Module did not accept command 27:70, return code: 5
"Session stopped" received
EVSE loop finished
Goodbye!
```
[image]

We have attached all the logs, both from the Terminal and from the UART, also the version of the repository with which we have run the applications EV and EVSE.

--------------------------------------------------------------------------------------

**Additional information:**

I run the application in following device

**cat /proc/device-tree/model**
`Raspberry Pi 4 Model B Rev 1.4`

**cat /etc/os-release**
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

With the following testbench: 
```
- WHITE-BEET-EI (EVSE): Name: ISO15118 (EV), Version: V01_00_04
- WHITE-BEET-PI (EV): Name: ISO15118 (EVSE), Version: V01_01_06
- cat5 shielded cables
- eth0 port: directly to raspberry pi
- eth1 port: connected with adapter eth to usb3.0
- usb cables of supply and uart connected to hub to usb 3.0 port
```
[image]

[LOG_n3_WHITEBEET_TERMINAL_EV-PI.txt](https://github.com/Sevenstax/FreeV2G/files/8146730/LOG_n3_WHITEBEET_TERMINAL_EV-PI.txt)
[LOG_n3_WHITEBEET_TERMINAL_EVSE-EI.txt](https://github.com/Sevenstax/FreeV2G/files/8146731/LOG_n3_WHITEBEET_TERMINAL_EVSE-EI.txt)
[LOG_n3_WHITEBEET_UART_EV-PI.txt](https://github.com/Sevenstax/FreeV2G/files/8146732/LOG_n3_WHITEBEET_UART_EV-PI.txt)
[FreeV2G-eth_pylibpcap.zip](https://github.com/Sevenstax/FreeV2G/files/8146738/FreeV2G-eth_pylibpcap.zip)

[LOG_n3_WHITEBEET_UART_EVSE-EI.txt](https://github.com/Sevenstax/FreeV2G/files/8146733/LOG_n3_WHITEBEET_UART_EVSE-EI.txt)

jpo-stx, 2022-03-07:
@codderembedded could you proceed with your project or are you still experiencing this issue with the https://github.com/Sevenstax/FreeV2G/tree/eth_pylibpcap branch?

duverg, 2022-03-15:
Hi @jpo-stx @vbudko, any news about the issue with the https://github.com/Sevenstax/FreeV2G/tree/eth_pylibpcap branch?

duverg, 2022-03-16:
Hi @jpo-stx @vbudko,

---

We have tested the branch eth_pylibpcap with the following commands:

```
sudo nice -n -20 python3 Application.py eth1 -m c4:93:00:21:56:b8 -r EV
sudo nice -n -20 python3 Application.py eth0 -m c4:93:00:21:51:f6 -r EVSE
```
and
```
sudo python3 Application.py eth1 -m c4:93:00:21:56:b8 -r EV
sudo nice -n -20 python3 Application.py eth0 -m c4:93:00:21:51:f6 -r EVSE
```
_(obviously resetting the two boards before launching the command and connecting only the control pilot when we see in both terminals the "Wait until an EV/EVSE connects" lines )_

---

But unfortunately we get the same warning output at the end:

```
Warning: Module did not accept command 27:70, return code: 5
"Session stopped" received
EVSE loop finished
Goodbye!
```

[image]

---

**Do you have any other ideas to solve this problem, in order not to stop the application because of the warning?**

duverg, 2022-03-23:
Hi @jpo-stx @vbudko, any news about the issue ?

ColinCGC, 2022-04-01:
I have seen the same behaviour (presumably the same issue), so would also be very interested in any news, as this is somewhat of a show-stopper.

duverg, 2022-04-04:
Hi @ColinCGC where can i find the firmware release for ISO15118 (EV) ?

vbudko, 2022-04-04:
> Hi @ColinCGC where can i find the firmware release for ISO15118 (EV) ?

Hi Alexander,
Please check Sevenstax folder at Codico PLC Docs https://downloads.codico.com/misc/plc/index.php
BR, Vasily

codderembedded, 2022-04-06:
Hi, we thought this problem was fixed. We haven't tested in a long time. but when we make a new test, it falls into error again within 5 10 minutes. why could this be?

ttu-stx, 2022-04-06:
We have changed the Host Controller Interface in v2.0.0 firmware so that this problem shall no longer occur.

closed 2022-05-02 by lho-stx