Sevenstax/FreeV2G #41: Warning: Module did not accept command 27:70, return code: 5

duverg, 2022-02-25:
Hi Sevenstax,

We are trying to connect the connect the two WHITEBEET Boards EV/PI and EVSE/EI, in order to understand its functioning, with the latest FreeV2G-master repository (_attached_).

**_APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_04
  Target:    WHITE-BEET
  CPU:       STM32F745

APPLICATION:
  Name:      ISO15118 (EVSE)
  Version:   V01_01_06
  Target:    WHITE-BEET
  CPU:       STM32F745_**

**_Testbench_**
[image]

With the following steps:
1. Reset both boards with the button reset
2. Launch  EV and EVSE demo applications.
3. Connect PE & CP wires.

**_Terminal_**
[image]

But we have always the same warning output:
_**Warning: Module did not accept command 27:70, return code: 5
"Session stopped" received
EVSE loop finished
Goodbye!**_

We would like to know if this warning is normal in the demo application, or it is a problem of our testbench?

Attached: Uart-Log and Terminal-Log for EVSE- EI 

[WHITEBEET_EVSE-EI_TERMINAL_N2.txt](https://github.com/Sevenstax/FreeV2G/files/8139842/WHITEBEET_EVSE-EI_TERMINAL_N2.txt)
[WHITEBEET_EVSE-EI_UART_N2.txt](https://github.com/Sevenstax/FreeV2G/files/8139843/WHITEBEET_EVSE-EI_UART_N2.txt)
[WHITEBEET_EV-PI_TERMINAL_N2.txt](https://github.com/Sevenstax/FreeV2G/files/8139844/WHITEBEET_EV-PI_TERMINAL_N2.txt)
[WHITEBEET_EV-PI_UART_N2.txt](https://github.com/Sevenstax/FreeV2G/files/8139845/WHITEBEET_EV-PI_UART_N2.txt)
[FreeV2G-master_Beond.zip](https://github.com/Sevenstax/FreeV2G/files/8140047/FreeV2G-master_Beond.zip)

klth, 2022-02-25:
Hi @duverg,

please have a look at issue #40

jpo-stx, 2022-03-07:
@duverg have you tried with the https://github.com/Sevenstax/FreeV2G/tree/eth_pylibpcap branch?

duverg, 2022-03-07:
Yes, i have the same warning (although I had more information exchange). I update 10 days ago the information to issue [#40](https://github.com/Sevenstax/FreeV2G/issues/40#issuecomment-1052135272) in order to avoid two equal issues.