Sevenstax/FreeV2G #42: WB V.02_00_00 V2G commands problem

IonelFratila, 2022-03-08:
Hi @jpo-stx,
I'm trying to adapt the host controller SW to the new version of WB V.02_00_00 (EVSE). Most of the V2G Service commands are changed or new, so I have to rewrite the charging process almost entirely.
Again many details are missing from the user manual, therefore I need some information about how the new introduced V2G commands are working:
1. If I try to send Start Listen command (0x27:0x6A), I always get the result code 0x11 (undocumented code). I suppose this is because no parameters are set into WB module before this command, but I need a confirmation from You (because I cannot sent the configuration commands, see below). Please tell me what are the prerequisites for WB in order to accept the V2G command StartListen.
2. If I try to send Set Configuration command (0x27:0x60), I have to enter EVSEID for both formats DIN and ISO, but there is no longer exists the possibility of not using one or both of these IDs (former code "No EVSE ID available"). Please tell me how can I send the Set Configuration command without an EVSEID?
3. In Set Configuration command (0x27:0x60), I believe that the actual length of the EVSEID fields should also be sent (the same problem as in the former User Manual, not solved yet). Please confirm.
4. In Set Configuration command again (0x27:0x60), the Supported protocols should be send on 1-2 bytes. Please explain how? What happens if one of the two bytes is omitted? (the former Count parameter is missing now). Can you tell me the correct format of this command?
5. If I try to send Get Configuration command (0x27:0x61), I always get the Result code 0xFF (Internal error). Please explain why?
6.Please explain what means the warning "EXPERIMENTAL" attached to the SW V.02_00_00. It means that it is a beta version and is not completely tested or what? 

Due to the above issues (at the first test), I am stuck and cannot continue to implement the new version. Please help!
Thank You,
Ionel Fratila

IonelFratila, 2022-03-08:
Hi @jpo-stx ,
Please review also the details regarding command Set DC Charge Parameters (0x27:0x62) format (especially Current Regulation Tolerance field, that can be optional, but how?). What is the meaning of the EVSE Status field? If Set DC Charge Parameters should be sent only before StartListen (next session), EVSE can be in other states than Ready? If so, EVSE will not decide to start a new session, so the command Set DC Charge Parameters will not be transmitted!!! Please explain.

Because the new commands introduced in WB V.02_00_00 are more different than former version regarding the standard V2G messages between EVSE and EV, can you describe in more detail how these command were intended to be used and when ?
Thank You,
Best Regards,
Ionel Fratila

jpo-stx, 2022-03-09:
Hi @IonelFratila , first of all thank you for using the new 2.0.0 firmware.

1. You need to set the "basic" configuration, protocols that you want to support, energy transfer mode, etc. After that you need to set the AC and/or DC parameters, depending on the energy transfer modes you selected. If you only support DC modes you only need to provide the DC parameters. After you set  the parameters you can start a session.
2. You can set the length of the EVSE ID to 0. So you can send 0x00 0x00 for both parameters if you do not want to provide an EVSE ID.
3. Yes, you send the ID in the format 1 byte length followed by the ID itself with the number of bytes specified.
4. I think there is some confusion about the notation. when the Type for a parameter is defined as i.e. uint8[1-2], this means an array of uint8s with a dynamic length of 1-2 bytes. Since the length is dynamic, it needs to be prefixed with the size of the array. In this case if you want to use the array with length 1: 0x01 0x00, if you use length 2: 0x02 0x00 0x01. Hopefully this also clarifies the usage of the EVSE IDs. I will add this type definition to the manual.
5. I will check this one. Can you give me the data of the set configuration command you used?
6. This version is fully tested. Experimental due to the new plug and charge functionality where we still need feedback from the field. The EIM functionality from the previous versions should not be affected.

jpo-stx, 2022-03-09:
> Hi @jpo-stx , Please review also the details regarding command Set DC Charge Parameters (0x27:0x62) format (especially Current Regulation Tolerance field, that can be optional, but how?).

The current regulation tolerance can be omitted if you set the perameter to 0x00. If you want to use the parameter send i.e. 0x01 0x00 0x01 0x00 for a value of 1A

> What is the meaning of the EVSE Status field?

This is simply the status of your EVSE. If everything is ok set it to Ready. If you have an error of somecase you can use this parameter to inform the EV about it.

> If Set DC Charge Parameters should be sent only before StartListen (next session), EVSE can be in other states than Ready? If so, EVSE will not decide to start a new session, so the command Set DC Charge Parameters will not be transmitted!!! Please explain.

If I understand you correctly you want to change the state later on? If so you can change the state by using the "Update DC charge parameters" command.

> Because the new commands introduced in WB V.02_00_00 are more different than former version regarding the standard V2G messages between EVSE and EV, can you describe in more detail how these command were intended to be used and when ? Thank You, Best Regards, Ionel Fratila

We introduced the change of the interface to reduce the timing requirements on host side. In the charging loop there was a very short timeout where the EVSE needs to respond to the request message of the EV.

IonelFratila, 2022-03-14:
Hi @jpo-stx ,
I have more problems related to the WB V.02_00_00 commands:
1. When the Authorization Status Request message is received (0x27:0x82), the TimeOut parameter contained is always equal to 0.
   ERx: 123456789ABC C49300215136 6003 0004000C C02782000004 [00000000]CFC1

1. When the Schedules Requested message is received (0x27:0x84), the TimeOut and MaxEntries parameters contained are always equal to 0 {!!!}.
ERx: 123456789ABC C49300215136 6003 0004000E C02784000006 [000000000000]CBC1

Actually, I think that all messages containing the TimeOut parameter have this parameter set to 0. Can you check, please?
Because MaxEntries parameter is zero, I cannot continue testing (cannot send SetSchedules command with valid parameters).
Please help!
Thank You,
Best Regards,
Ionel Fratila

jpo-stx, 2022-03-14:
Hi @IonelFratila , thanks for the input I will check the points.

I have just I checked the manual and have seen that there is an error in the example.

But you should be able to set at least one schedule entry. A minimal set schedule command shoul look like this:

```
00           Result code OK
01           1 schedule tuple
01           Tuple ID of the 1st schedule
01           1 schedule entry
00 00 00 00  1st entry starts now
00 01 51 80  Duration of 1st entry: 86400 (24h)
00 0B 03     Available power: 11kW
00           No energy to be delivered
00           No sales tariffs
```

IonelFratila, 2022-03-14:
hI @jpo-stx ,
I think the Tuple ID is sint16, so has to be transmitted as 00 01, right? What happens if the MaxEntries received in the Schedule Requested message is 0 and I still transmit one Tuple with 1 schedule Entry? It is not rejected by WB? It is only an error in the transmision of the Schedule Requested message?
Thank You,
Ionel Fratila

IonelFratila, 2022-03-14:
Hi @jpo-stx ,
I have some major problems with WB V.02_00_00 (which you said that it was fully tested!!!). I expected to encounter some problems (bugs) with this new version, but not so big!

1.When Cable Check Requested message is received from WB (again with TimeOut =0), the host controller does not have to confirm this message in any way, but should start cable check execution and answer after this process with the Set Cable Check Finished message (this I understand to be the normal behavior). Instead of this, after Cable Check Request message was sent from EV to WB, WB send the Cable Check Requested message to host controller but does not answer to EV with Cable Check Confirmation (no further V@G messages exchanged between EV and WB). After few seconds, WB send the Session Stopped and Session Error messages to host controller and everything is over. Please observe also the error code in the Session Error message bellow! (undocumented code!)
.....
ERx: FFFFFFFFFFFF C49300215138 88E1
ERx: FFFFFFFFFFFF C49300215138 88E1
ERx: FFFFFFFFFFFF C49300215138 88E1
ERx: 123456789ABC C49300215136 6003 00040008 C02880FF0000 D4C1   ;here SLAC matching process was successful
ERx: 3333FF03950A 007DFA03950A 86DD
ERx: 333300000002 007DFA03950A 86DD
ERx: 333300000001 007DFA03950A 86DD
ERx: 333300000001 007DFA03950A 86DD
ERx: 123456789ABC C49300215136 6003 00040018 C02780000010 01C7901B3B80817DF906007DFA03950A 7BC1   ;Session started
ERx: 123456789ABC C49300215136 6003 00040009 C02781000001 00 D3C1
ERx: 123456789ABC C49300215136 6003 0004000C C02782000004 00000000 CFC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040009 C0276B120001  00 D7C1
ERx: 123456789ABC C49300215136 6003 00040009 C0276B120001 00 D7C1
ERx: 123456789ABC C49300215136 6003 0004001B C02783000013 00000DF2FF0004B0FF00010000000000010028 E0C1
ERx: 4000C000 123456789ABC C49300215136 6003 0004000E C02784000006 000000000000 CBC1
ERx: 123456789ABC C49300215136 6003 0004001E C02785000016 0DF2FF04B0FF00010028000000000000000000000000 DCC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001E C0276C130016 00010000000100000000FFFFFFFF0032030101F40300 8FC1
ERx: 123456789ABC C49300215136 6003 00040009 C0276C130001 00 D5C1
ERx: 123456789ABC C49300215136 6003 0004000C C02787000004 00000000 CAC1
ERx: 123456789ABC C49300215136 6003 00040009 C02981FF0001 02 CFC1
ERx: 333300000002 007DFA03950A 86DD
ERx: 123456789ABC C49300215136 6003 00040009 C0278C000001 00 C8C1
ERx: 123456789ABC C49300215136 6003 00040009 C0278E000001 17 AFC1
ERx: 400F 4A00C001 333300000002 007DFA03950A 86DD
ERx: 920F 4C00C001 333300000001 007DFA03950A 86DD

2. An old problem regarding the Session ID not zero in the Session Setup Request message from EV is raised again (I signaled this several months ago when using WB V.01_01_02, see issue #14). I found that at every new session (after the above sequence, abnormally terminated), the Session Setup Request from EV contains the Session ID from the previous session (not zero). Even the Session ID sent from EV is not zero, the V2G communication continues and WB respond to EV with Session Setup Response (OK) message containing a new Session ID. 

Please help!!!
Ionel Fratila

jpo-stx, 2022-03-14:
> 1. When Cable Check Requested message is received from WB (again with TimeOut =0), the host controller does not have to confirm this message in any way, but should start cable check execution and [...]

I think this is due to the charge parameters not being updated in time. After cable check requested was received you need to update the parameters at least every 500ms, so the whitebeet can be sure that the host is still alive. I'll add this error code and timeout to the manual. The infomation that an update of the parameters is needed regularily is present in the manual in the application examples (2.8.6 Cable Check)

>2. An old problem regarding the Session ID not zero in the Session Setup Request message from EV is raised again (I signaled this several months ago when using WB V.01_01_02, see issue [White-Beet-EI strange problem regarding Session Setup Request #14](https://github.com/Sevenstax/FreeV2G/issues/14)). I found that at every new session (after the above sequence, abnormally terminated), the Session Setup Request from EV contains the Session ID from the previous session (not zero). Even the Session ID sent from EV is not zero, the V2G communication continues and WB respond to EV with Session Setup Response (OK) message containing a new Session ID.

This is as expected. When a session wasn't closed successfully with session stop (pause), it cannot be resumed and a new session with a new session ID is started. If the EV wants to resume a session at a later point in time in needs to pause a session successfully. We have seen some EVs that always use the session ID from the previous session, even if it was from a different charging station.

IonelFratila, 2022-03-16:
Hi @jpo-stx ,
Finally, I was able to run a full charge session, ended correctly with user stop interaction. Still, I have some problems to solve:

1. It seems that EVSEStatus information in the Update DC Charging Parameters command (0x27, 0x63) is not handled right by WB V.02_00_00. If I update this parameter at the end of Cable Check sequence to Isolation Monitoring Active and right after that I send the Cable Check Finished message (0x27,0x6D), the last Cable Check Response sent by WB to the EV does not contains this parameter updated. I tried also with a 100ms delay between Update DC Charging Parameters (0x27, 0x63) and Cable Check Finished (0x27,0x6D) and also repeating the Update DC Charging Parameters message, but the result is the same.
At the same time, when during Precharge sequence I try to stop the charging process, I send the Update DC Charging Parameters message with EVSEStatus=EVSEShutDown and after that Send Notification message (0x27, 0x75) message. In this case, The last message PrechargeResponse sent from WB to EV does contain the updated value of EVSEStatus. You can check this in the following log between host controller and WB module, and also in the capture of the message log between WB and EV.
Please tell me when the EVSEStatus value should be updated, in order to have the updated value transmitted to EV every time.

2. The problem signaled in WB V.01_00_06 related to the correct answer to Session Stop Request message from EV is still present also in WB V.02_00_00. When EV send this message (due to PWM change to 100%), WB does not answer correctly to this message and the communication with the car is not ended properly (see the capture of the messages between WB and EV bellow). PWM change to 100% cannot be delayed (according to the standard, at least for emergency stop situations) and the answer to Session Stop Request message from the EV should be transmitted every time (again standard requirements). We have already discussed this problem in issue #14.
Please check the behavior of WB in this situation.

3. Please tell me if you have some news about the problem with TimeOut=0 (and other parameters, like TupleCount) in WB messages. When do you think it will be solved?
4. 
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763060014 001392FF0000FD0101F4000130D4FE0100320300 03C1
ERx: 9A02 4000C000 123456789ABC C49300215136 6003 00040009 C02763060001  00 EBC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763070014 001392FF0000FD0101F4000130D4FE0100320300 02C1	;Update DC Charge Prm, EVSE Status=Ready
ERx: 123456789ABC C49300215136 6003 00040009 C02763070001 00 EAC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763080014 0207D0FD0000FD0101F4000130D4FE0100320304 CAC1	;Update DC Charge Prm, EVSE Status=Isolation Monitoring Active
ERx: 123456789ABC C49300215136 6003 00040009 C02763080001 00 E9C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763090014 0207D0FD0000FD0101F4000130D4FE0100320304 C9C1	;Update DC Charge Prm, EVSE Status=Isolation Monitoring Active
ERx: 123456789ABC C49300215136 6003 00040009 C02763090001 00 E8C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040009 C0276D0A0001 00 DDC1			;Set Cable Check Finished=OK
ERx: 123456789ABC7899300215136 6003 00040009 C0276D0A0001  00 DDC1
ERx: 123456789ABC C49300215136 6003 0004001E C02785000016 0DF2FF04B0FF0001004F0CFDFF000AFF000000000000 A1C1	;DC Parameters Changed
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C027630B0014 0207D0FD0000FD0101F4000130D4FE0100320304 C7C1	;Update DC Charge Prm, EVSE Status=Isolation Monitoring Active
ERx: 123456789ABC C49300215136 6003 00040008 C02788000000 CDC1		;PreCharge Started
ERx: 123456789ABC C49300215136 6003 00040009 C027630B0001 00 E6C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C027630C0014 0207D0FD0000FD0101F4000130D4FE0100320304 C6C1	;Update DC Charge Prm, EVSE Status=Isolation Monitoring Active
ERx: 123456789ABC C49300215136 6003 00040009 C027630C0001 00 E5C1
...
...
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763130014 021770FD0032FD0101F4000130D4FE0100320304 DDC1	;Update DC Charge Prm, EVSE Status=Isolation Monitoring Active
ERx: 123456789ABC C49300215136 6003 00040009 C02763130001 00 DEC1
											;User Stop Condition

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004000A C02944140002 03E8 0EC1			;Set PWM=100%
ERx: 123456789ABC C49300215136 6003 00040009 C02944140001 00 FAC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763150014 021770FD0032FD0101F4130D4FE0100320302 DDC1	;Update DC Charge Prm, EVSE Status=ShutDown
ERx: 123456789ABC C49300215136 6003 00040009 C02763150001 00 DCC1
ERx: 123456789ABC C49300215136 6003 00040009 C0278E000001 01 C5C1	;Session Error=0x01=Sequence Error!!!
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763160014 021770FD0032FD0101F4000130D4FE0100320302 DCC1	;Update DC Charge Prm, EVSE Status=ShutDown
ERx: 123456789ABC C49300215136 6003 00040009 C02763160001 00 DBC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004000B C02775170003 000000 C6C1			;Send Notification=Stop Charge
ERx: 123456789ABC C49300215136 6003 00040009 C02775170001 00 C8C1
ERx: 123456789ABC C49300215136 6003 00040009 C02981FF0001 01 D0C1	;CPState=B
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763180014 021770FD0036FD0101F4000130D4FE0100320302 D6C1	;Update DC Charge Prm, EVSE Status=ShutDown
ERx: 123456789ABC C49300215136 6003 00040009 C02763180001 00 D9C1
...
...
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C027632B0014 0207D0FD0000FD0101F4000130D4FE0100320302 A9C1	;Update DC Charge Prm, EVSE Status=ShutDown
ERx: 123456789ABC C49300215136 6003 00040009 C027632B0001 00 C6C1
ERx: 123456789ABC C49300215136 6003 00040009 C0278C000001 00 C8C1	;Session Stop=Terminate

[image]
[image]

Thank You, 
Ionel Fratila

IonelFratila, 2022-03-20:
Hi @jpo-stx ,
After many tests performed on the WB V.02_00_00, I have found that none of the problems I signaled in issue #20  were solved. These were, in short:
1.WB does not answer with Session Stop Response=OK to the Session Stop Request=Terminate from the EV, during all stages except charge. The error code transmitted to EV and also to host controller is "Sequence Error", even it is not actually an error and EV can send Session Stop Request messages at any time (even before Power Delivery Request=Start).
Because the session stop request is not confirmed by the EVSE, this can be the cause for a lot of other abnormal situations, like for example when EV sends a SessionID not equal to zero in the next Session Start Request.
2. After Session Stop Request from EV (to which WB responds with Sequence Error code), if the entire process is restarted by sending again PWM=5% and restarting the SLAC Service and the V2G Service and then sending Start SLAC Matching Process (0x44), EV send again SLAC_Param.Req, WB answer to this message and the entire SLAC Matching is performed OK, terminated with SLAC_Match.Req from EV and SLAC_Match.Confirm from EVSE, containing NID and NMK parameters. The problem is that the NID and NMK parameters are changed from the first SLAC Matching process, even if the EV has not been disconnected (no CP STate=A has been detected).
According to the standard, the NID and NMK parameters must be initialized before plug-in the EV and should remain unchanged until the EV is plugged out (CP State =A), no matter how it is handled by WB module internally (even if the SLAC Service is restarted or not).

In order not to waste extra time with unnecessary tests, please confirm that there are no changes in the new version WB V.02_00_00 regarding these issues, or explain what should be done to make them work properly.

In addition to these two older issues, I have reported two other issues regarding WB V.02_00_00, to which I have not yet received a response (see above in this issue #42):
3.TimeOut and other parameters (like TupleCount or MaxEntries) in the messages from WB to host controller are transmitted as 0 (incorrect values).
4.WB does not transmit actual EVSEStatus information (last received from host controller) to EV every time.

Please tell me if and when do you think these problems will be corrected?
Thank You,
Ionel Fratila

jpo-stx, 2022-03-22:
Hi @IonelFratila ,

> 1. WB does not answer with Session Stop Response=OK to the Session Stop Request=Terminate from the EV, during all stages except charge. The error code transmitted to EV and also to host controller [...]

To be clear, you have a DIN70121 charging session where the Whitebeet is responding with a sequence error to a SessionStopReq message before charging was started. This should be fixed and the Whitebeet should respond with OK. However, if the EV does this in a ISO15118 session this would lead to a sequence error, because the specification was changed regarding the session stop. If you do have this issue in the DIN70121 protocol, it would be helpful if you provide a wireshark log and uart log of the Whitebeet, because I cannot reproduce the issue.

> 2. After Session Stop Request from EV (to which WB responds with Sequence Error code), if the entire process is restarted by sending again PWM=5% and restarting the SLAC Service and the V2G Service [...]

The SLAC service should only be restarted when the EV was disconnected. If you want to continue with the same NID and NMK just start the matching again without restarting the service.

> 3. TimeOut and other parameters (like TupleCount or MaxEntries) in the messages from WB to host controller are transmitted as 0 (incorrect values).

This should be fixed in the next version. Some news about the max entries parameter. When this parameter is 0 the EV is not limited regarding the amount of entries in each schedule tuple (The max entries parameter is optional in the V2G protocol).

> 4. WB does not transmit actual EVSEStatus information (last received from host controller) to EV every time.

What you do is correct, I could reproduce the issue. It seems that the status is set when the next request is received from the EV.

Let me summarize the issues you found so far:
- Get Configuration command fails when both EVSE IDs are set to null
- Timeout parameters are zero in multiple messages
- EVSE status not set immediatly after it was updated
- User Manual "18.10.13 Set Schedules" Example is not correct and it does not mention the 500ms timeout in the charge loop

IonelFratila, 2022-03-22:
Hi @jpo-stx ,
1. To be clear, I'm running both DIN 70121 and ISO15118-2:2014 charging sessions. For both types, the answer to Session Stop Request = Terminate before charging is started (before Power Delivery Request=Start from the EV) is the same, Session Stop Res=Failed (Sequence Error). At least you agree that this is a problem according to DIN 70121. I think the same apply also according to ISO 15118 (because Session Stop Req message is transmitted by EV after V2G session is started but before V2G charging loop is started, the message Power Delivery Req=Stop and also Welding Detection Request will not be transmitted). I will check again the exact requirement in ISO 15118-2:2014.
Unfortunately I don't have a Wireshark log, but you told me once that I can run a DIN 70121 session by limiting the supported protocols in Set Configuration message. Maybe you can reproduce the situation that way.

If I don't restart SLAC Service, EV does not start a new SLAC matching process. It is not enough to send the Start SLAC Matching command to WB. I think it is a problem with the re-initialization of EVSE network status and the low level messages sent to EV after Session Stop. Can you please check?

And I have a new bug discovered today: The values for MinCurrent and MaxVoltage received by WB in Set DC Charging Parameters message are exchanged when they are transmitted to EV in Discovery Charge Parameters Response. Here is the message sent to WB and the message that WB sends to EV. Please check!

Set DC Charging Parameters, Vmin=200V, MinCurrent=0A, MaxVoltage=500V, ...
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040020 C027620D0018 0000C80000000001F40030D4FE0032030100010000040000 D0C1
                                              
ERx: 123456789ABC C49300215136 6003 00040009 C027620D0001 00 E5C1

[image]

Thank you,
Ionel Fratila

jpo-stx, 2022-03-23:
> If I don't restart SLAC Service, EV does not start a new SLAC matching process. It is not enough to send the Start SLAC Matching command to WB. I think it is a problem with the re-initialization of EVSE network status and the low level messages sent to EV after Session Stop. Can you please check?

Have you tried to set the duty cycle to 0% (CP State F) and back to 5%?

IonelFratila, 2022-03-23:
Hi @jpo-stx ,
First of all, I think we need to be more careful and more clear about the issues we test and the conditions used for testing.

1.Regarding the problem with Session Stop Response, under DIN 70121 conditions:
_"I retested with the DIN protocol and everything is fine:"_
Can you tell me please what SW version did you use for WB (EVSE)?
After retesting with DIN 70121 protocol, WB V.01_01_06 is NOT working well (send Session Stop Response=Failed, error code=Sequence Error), and WB V.02_00_00 is OK (send Session Stop Response=OK). So the conclusion is that this was corrected in WB V.02_00_00 (this information would have been enough to close the subject).
But there is still a problem with the answer to Session Stop Request from EV, in both versions V.01_01_06 and V.02_00_00:
If the result of Cable Check is not OK (isolation status Faulted), host controller send the Result Code=Failed to WB, WB send Cable Check Response message with ResponseCode=Failed to EV and EV send the message Session Stop Request to WB. WB does not answer at all with a Session Stop response message.

2.Regarding the problem with Session Stop Response, under ISO15118-2:2014 conditions:
_"For ISO15118 a PowerDeliveryReq with Start Charging need to be sent, then CurrentDemandReq, after this PowerDeliveryReq with Stop Charging and then SessionStopReq."_
I do not agree with this interpretation of the standard, for the following reasons:
Some requirements from the ISO15118-2:2014:
"**8.4.3.12.3.SessionStopRes**
_After receiving the SessionStopReq of the EVCC the SECC sends the SessionStopRes informing the EVCC if terminating the charging process was successful."_
So the SessionStopRes message shall be transmitted all the time, regardless of the conditions and charging stage.
Also, from [V2G2-458], [V2G2-459], [V2G2-524], the EV can send SessionStopReq directly (without PowerDeliveryReq=Stop message) (otherwise haw the V2G_Communication_Session can be stopped?) and If EVSE answer with a ResponseCode=Failed, this will tell to EV that the request message has not been processed successfully (in this case the communication was not ended?)
_"[V2G2-538] The SECC shall respond with the corresponding response message containing a 'ResponseCode=FAILED_SequenceError' if request message was received that the SECC does not expect in the wait state".__
But who says that the message SessionStopReq is not expected during Cable Check?
"[V2G2-571]" requires that SECC shall respond with SessionStopRes with ResponseCode=OK if the processing of the information is successfully passed (and then shall terminate the communication by applying [V2G-043].

There are some other requirements regarding to SessionStopReq message, but none says that this message cannot be received during Cable Check. As you can see, [V2G2-]
Moreover, let's think about what happens if the Cable Check is finished with Isolation Status=Faulted. Under no circumstances EV will not send in this case PrechargeReq, PowerDeliveryReq or CurrentDemandReq messages, because energy cannot be applied to the EV connector (safety reason). In this case SessionStopRes will always be transmitted with ResponseCode=FAILED_SequenceError? What is your opinion on this situation?
Of course after starting the charging loop, EV will send all these messages before sending SessionStopReq, in order to stop the communication session, but this is not true before starting charging loop.

3.Regarding the SLAC matching re-initialization, I tried with PWM=100% but not with 0%. I will try this, but I still think that this is a problem with the communication session stopping (maybe related to the SessionStopReq=Failed ???)

Thank You,
Ionel Fratila

IonelFratila, 2022-03-24:
Hi @jpo-stx ,
Until yerstaday I have completely tested and run complete charging sessions only using WB V.02_00_00 according to ISO15118. When I tried to run according to DIN, every time since now I stopped the process during or at end of Cable Check stage, just to see if SessionStopRes is transmitted OK to EV. Now I'm trying to run a complete session on DIN and I found a new issue:
When running on DIN70121 (by limiting the EVSE supported protocols), when the Cable Check is finished OK and Precharge is started, WB does not transmit any DC Charge Parameters Changed (0x27:0x85 message) to host controller, it only transmit Precharge Started message (0x27:0x88). When running using ISO15118 everything goes normally.

This is the log between host controller and WB when running ISO15118 (the end of Cable Check and start of Precharge - OK):
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763E00014 006979FE028AFD0101F400011D4CFD0100320300 FBC1
ERx: 123456789ABC C49300215136 6003 00040009 C02763E00001 00 11C1
....
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F30014 0207D0FD029EFD0101F400011D4CFD0100320304 DAC1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F30001 00 FDC1
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040009 C0276DF40001 00 F2C1			;Set CableCheck=Succes
ERx: 123456789ABC C49300215136 6003 00040009 C0276DF40001 00 F2C1
ERx: 123456789ABC C49300215136 6003 0004001E C02785000016 0DF2FF04B0FF000100370C4BFF000AFF000000000000 6CC1	;DCChgPrm Changed
ERx: 123456789ABC C49300215136 6003 00040008 C02788000000 CDC1		;Precharge Started
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F50014 0207D0FD02A8FD0101F400011D4CFD0100320304 CEC1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F50001 00 FBC1
.....
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763FE0014 0207D0FD067CFD0101F400011D4CFD0100320304 EDC1
ERx: 1C02 4000C000 123456789ABC C49300215136 6003 00040009 C02763FE0001 00 F2C1
.....

And this is the log between host controller and WB when running DIN70121 (the end of Cable Check and start of Precharge -not OK):
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763CA0014 001392FF0258FD0101F400011D4CFD0100320300 80C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763CA0001 00 27C1
.......                                                                              
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763D20014 001392FF0262FD0101F400013C28FD0100320300 73C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763D20001 00 1FC1
........                                                                              
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F00014 0203E8FD028AFD0101F400011D4CFD0100320304 DDC1
;Update DCChgPrm, Isolation=2(Warning-OK), PresV=1V, PresI=0.65A, MaxV=500V, MaxI=7.5A, MaxP=50kW, Status=4-IsoMonitActive
ERx: 123456789ABC C49300215136 6003 00040009 C02763F00001 00 01C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040009 C0276DF10001 00 F5C1			;Set CableCheck=Succes
ERx: 123456789ABC C49300215136 6003 00040009 C0276DF10001 00 F5C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F20014 0203E8FD0262FD0101F400011D4CFD0100320304 04C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F20001 00 FEC1
;There is no DC Charge Paramaters Changed!!!
ERx: 123456789ABC C49300215136 6003 00040008 C02788000000 CDC1		;Precharge Started
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F30014 0203E8FD0262FD0101F400011D4CFD0100320304 03C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F30001 00 FDC1
...........                                                                              
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763FA0014 021770FD0276FD0101F400011D4CFD0100320304 4CC1
;Update DCChgPrm, Iso=2, PresV=6V, PresI=0.63A, MaxV=500V, MaxI=7.5A, MaxP=50kW, Status=4-IsoMonitActive
ERx: 123456789ABC C49300215136 6003 00040009 C02763FA0001 00 F6C1
;here Host controller does not control the voltage for Precharge because it was not asked by WB!!!

Please confirm if you can reproduce this!

Let's summarize the list of open issues found on WB V.02_00_00:
1. Get Configuration command fails when no parameters have been set before (you said when both EVSE IDs are set to null, I haven't checked yet)
2. Timeout parameters are zero in all messages that contain it, and also MaxEntries in Schedules Requested (0x27:0x84)
3. EVSE status not set immediately after it was updated
4. WB does not answer with a Session Stop response message to SessionStopReq=Terminatefrom EV, when Cable Check is ended with ResponseCode=Failed.
5. Values for MinCurrent and MaxVoltage received by WB in Set DC Charging Parameters message are exchanged with each other when transmitted to EV in Discovery Charge Parameters Response
6. WB send SessionStopRes with ResponseCode=FAILED_SequenceError (when using ISO5118) when SessionStopReq is not preceded by PowerDeliveryReq from EV (here we have different opinion on standard requirements, have to discuss). This can be the cause for SessionID not 0 from EV after plugged-in or the impossibility to start a new SLAC matching!?
7. Cannot start a new SLAC matching process or SLAC matching is not followed by a new charging session, when the EV is not plugged-out and then plugged-in again (I have to test further with PWM=0%, as you suggested)
8. When running DIN protocol, the message DC Charge Parameters Changed (0x27:0x85) is not transmitted by EV before Precharge start

When do you think these issues will be corrected? Do you have a time estimation for the next release? I know that there is a new release V.01_01_07, but according to the Release Notes file it does not bring new things compared to the previous version (only Closure Type from SessionStopped message), moreover the changes introduced in V.02_00_00 (regarding for example EVSEStatus or Send Notification) are not available there. What will happen with V.02_00_00 version? It will continue to be updated in the future or we need to go back to the old version?

Thank You,
Ionel Fratila

IonelFratila, 2022-03-24:
Hi @jpo-stx ,
I have an update on the issues list above:
_5. Values for MinCurrent and MaxVoltage received by WB in Set DC Charging Parameters message are exchanged with each other when transmitted to EV in Discovery Charge Parameters Response_
This is happens both with ISO and DIN protocol, and most of the time Vmax and Imin are transmitted exchanged to the EV, but I have seen also situations when Vmax=500V and Imin=500A. I have checked every time the message Set DC Charging Parameters that host contrller send to WB and is correct. The initial values I use all the time are Vmax=500V, Imin=0A. So I think there is something random when the Discovery Charge Parameters Response is sent to EV.
_7.When running DIN protocol, the message DC Charge Parameters Changed (0x27:0x85) is not transmitted by EV before Precharge start_
This is happens all the time when using DIN protocol. But something random happens also when using ISO: sometime (not all the time) when Cable Check is finished, host controller send the new Isolation Status and EVSEStatus to WB and then send Set Cable Check Finished - Success, WB keep sending CableCheckRes to EV with EVSEProcessing=Ongoing, but with the updated values for EVSEIsolationStatus and EVSEStatusCode. The commands from host controller are correct (I checked) and WB confirm the receiving of Set Cable Check Finished message (which anyway is set to be repeated 3 times if not confirmed).
Please check what happens in the WB around the end of Cable Check, because there are too many strange things at that moment.

Thank You,
Ionel Fratila

IonelFratila, 2022-03-25:
Hi @jpo-stx ,
Update to issue 7 on the list above:
I have tested again the procedure for re-setup the SLAC Matching process after a SessionStopReq=Terminate message, following the exact requirements in ISO15118-3:2014 standard, Chapter 7.5 (CPState=F for at least C_step_EF). The conclusions are:
a. If SLACService is stopped and then restarted, then WB transmits CM_Unassociated_Sta.Ind messages and EV sends again CM_SLAC_Param.Req message to start SLAC matching process. When SLAC matching is successful, a new NMK and NID are sent to EV, after that the V2G communication is not started (EV does not send SECC Discovery Request). Probably this is due to the change of NMK and NID network parameters.
b. If SLAC Servcie is not restarted, no message CM_Unassociated_Sta.Ind appears and EV does not start SLAC matching anymore.
c. Can you please check if the parameters NMK and NID can remain unchanged regardless of SLAC Service restart, in order to fulfill ISO15118 requirements regarding signaling of the Unmatched state of EVSE node and also NID/NMK unchanged until EV is plugged out? Or do you have any other ideas how to solve it?

Thank you,
Ionel Fratila

IonelFratila, 2022-03-30:
Hi @jpo-stx ,
Please tell me if you have any comments about the issues list above (you can reproduce them or not, consider them problems or not, if and when you intend to solve them). We need an answer to the questions I have sent, in order to be able to decide what to do next and to plan our activity in the near future, related to this project.
_When do you think these issues will be corrected? Do you have a time estimation for the next release? I know that there is a new release V.01_01_07, but according to the Release Notes file it does not bring new things compared to the previous version (only Closure Type from SessionStopped message), moreover the changes introduced in V.02_00_00 (regarding for example EVSEStatus or Send Notification) are not available there. What will happen with V.02_00_00 version? It will continue to be updated in the future or we need to go back to the old version?_

Thank You,
Ionel Fratila

jpo-stx, 2022-03-30:
Hi @IonelFratila , I will answer all your questions, but I need some time on that.

There will be updates with bugfixes coming for version V02_00_00. I'll come back to you ASAP!

IonelFratila, 2022-03-31:
Hi @jpo-stx ,
Thank you for the feedback. I look forward for your answers.
Still, I have to ask you if you have an estimate of the time needed for solving the reported problems. I must remind you that apart from the problems that appeared with V.02_00_00, the other issues have been reported since the end of November 2021, that means 4 months ago. Then we had the software for host controller almost ready and since then we have only rewritten the software according to the new specification of WB V.02_00_00, but with the same functional problems. We need to know more accurately when you plan to have an ISO15118/DIN70121 compliant industrial product (at least in terms of main requirements) usable in mass production, (as it was presented a year ago by CODICO), not just an experimental product, still in development. I hope you understand that we also have some deadlines to meet and we can't wait indefinitely to solve these problems.
Thank you,
Ionel Fratila

ttu-stx, 2022-03-31:
I hided your comment because it contained unnecessary discussions.
This issue tracker is used to improve FreeV2G sw and belonging firmware.
If you have general product related feedback: pls contact you distributor.

Regarding update roadmap: we will announce very soon the time plan for the coming versions.

Best Regards
Tolga

jpo-stx, 2022-04-05:
Hi @IonelFratila ,

>1. Get Configuration command fails when no parameters have been set before (you said when both EVSE IDs are set to null, I haven't checked yet)

When one of the EVSE IDs is zero length the get command fails. This is fixed in the next version.

>2. Timeout parameters are zero in all messages that contain it, and also MaxEntries in Schedules Requested (0x27:0x84)

This is fixed in the next version.

>3. EVSE status not set immediately after it was updated

I was able to reproduce the issue and it is fixed in the next version. I.e. for cable check when the status is set and immediatly after that the cable check finished command is sent, the new status is present in the cable check response message.

>4. WB does not answer with a Session Stop response message to SessionStopReq=Terminatefrom EV, when Cable Check is ended with ResponseCode=Failed.

When the EVSE answered with a FAILED response code the session has finished and further messages are ignored. Is your EV throwing an error?

>5. Values for MinCurrent and MaxVoltage received by WB in Set DC Charging Parameters message are exchanged with each other when transmitted to EV in Discovery Charge Parameters Response

I could reproduce the issue and it will be fixed in the next release. However, I didn't see the randomness you described in the update of this issue.

> 6. WB send SessionStopRes with ResponseCode=FAILED_SequenceError (when using ISO5118) when SessionStopReq is not preceded by PowerDeliveryReq from EV (here we have different opinion on standard [...]

Can you open a new discussion for this in https://github.com/Sevenstax/FreeV2G/discussions

>7. Cannot start a new SLAC matching process or SLAC matching is not followed by a new charging session, when the EV is not plugged-out and then plugged-in again (I have to test further with PWM=0%, as you suggested)

Let me answer this seperately.

>8. When running DIN protocol, the message DC Charge Parameters Changed (0x27:0x85) is not transmitted by WB before Precharge start

I could reproduce the issue, it seems that the status message is not sent everytime the EV parameters change. This is fixed in the next version.

IonelFratila, 2022-04-05:
Hi @jpo-stx ,
Thank you for the answers. But you did not answer to the main question: When the next version you mention will be available?

4.
> When the EVSE answered with a FAILED response code the session has finished and further messages are ignored. Is your EV throwing an error?

Why do you consider that the session has finished after EVSE responds with FAILED code in Cable Check Response message? I can find nothing in the standards regarding this. Instead, chapter 8.4.3.12.3 ask that SessionStopRes message to be send (regardless of the Cable Check Status!), in order to confirm that the charging session was terminated succesfully. So I don't understand your question about the EV.
In the real case, after the Cable Check Res message EV sends Session Stop Req message = Terminate and does not receive any answer. Of course in this case the session will be forced to close due to the Message Time-Out, but it is not the standard behavior and the EV signals an error.

5. As I mentioned in the update of this issue, I had tests in which after finishing Cable Check OK, host controller transmitted the new values for  Isolation Status and EVSEStatus to WB and then transmitted Set Cable Check Finished - Success. After this, host controller continues to update DC Charge Parameters (at every 100ms) and WB kept sending CableCheckRes messages to EV with EVSEProcessing=Ongoing, but with the updated values for EVSEIsolationStatus and EVSEStatusCode. This happened in about 20% of the tests performed. I checked every time that the messages from host controller are correct.

6. I think this can be related also with issue 7, so I'm waiting for an answer there first.
> Can you open a new discussion for this in https://github.com/Sevenstax/FreeV2G/discussions

My goal is not to have a discussion, but to find a solution to this problem. Please let me know if you still believe that these are the requirements of ISO15118-2.
 
7. I'm still waiting for the answer to issue 7 (the only true issue I had, the rest are only bugs that can be easily fixed, I quess), because it can be related also with issue 6. I wonder if SLAC matching procedure could be restarted without plugging-out the EV (as stated in ISO15118-3), even when EVSE sends SessionStopRes with FAILED_SequenceError code.

Thank you,
Best Regards,
Ionel Fratila

ttu-stx, 2022-04-05:
JPO asked to move the conversation to the discussion section as the issue tracker becomes confusing with too much back and forth discusssions. Details can be exchanged there without blowing up the issue tracker.
Furthermore it would be helpful to create an issue here per real existing issue instead of handling several problems within a single thread.

Regarding the release date for updates: we will announce them in the discussions section as well.

jpo-stx, 2022-04-05:
> Why do you consider that the session has finished after EVSE responds with FAILED code in Cable Check Response message? I can find nothing in the standards regarding this. Instead, chapter [...]

It would be interesting what the EV is doing, because we do not want any errors on EV, even if it is not acting 100% according to spec::

[V2G2-539]  After  the  SECC  sent  a  "ResponseCode  =  FAILED"  it  shall  terminate  the  communication  by 
applying [V2G2-034]. 
[V2G2-034]  The SECC shall terminate the TLS or TCP connection after stopping the V2G Communication 
Session. 

> 5. As I mentioned in the update of this issue, I had tests in which after finishing Cable Check OK, host controller transmitted the new values for Isolation Status and EVSEStatus to WB and then [...]

Was able to reproduce this, it happens in rare case and is fixed in the next version.

IonelFratila, 2022-04-06:
Hi @ttu-stx ,
I'm sorry that "the issue tracker becomes confusing" due to the issues I reported. It is not my fault that are many in V.02_00_00. And they are all "real existing issues", as you can see. I cannot open a separate issue for every single bug I found. I thought you wanted to know about these problems. Or not?
Thank you,
BR,
Ionel Fratila

IonelFratila, 2022-04-06:
Hi @jpo-stx ,
Regarding the issue 4 from the list:

> It would be interesting what the EV is doing, because we do not want any errors on EV, even if it is not acting 100% according to spec::

I have to tell you that EV is acting 100% according to spec, at least in this case. I can't say the same about WB! Please see bellow.

 I retested this on Skoda CitiGo and the result is the same: -after WB send the message Cable Check Response (Failed), EV send Session Stop Req message (50miliseconds after) and WB does not answer to this message. From this point the EV is waiting (for an answer, meaning a confirmation from EVSE that the session was closed) for at least 30 seconds, while the EV LED is still ON and the connector is still locked (EV cannot be unplugged). You must agree that this is not a normal behavior of WB. I have attached the capture of the communication log between EVSE and EV at that moment.
[image]
And this is the communication log between host controller and WB:
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763EE0014 0203E8FD0262FD0101F400011D4CFD0100320304 08C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763EE0001 00 03C1

ETx: FFFFFFFFFFFF 123456789ABC 6003 00040009 C0276DEF0001 01 F6C1 ;Cable Check Finished with Failed result
ERx: 123456789ABC C49300215136 6003 00040009 C0276DEF0001 00 F7C1

ERx: 123456789ABC C49300215136 6003 00040009 C0278E000001 00 C6C1   ;Session Error (Unspecified)

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004000A C02944F00002 03E8 31C1 ;host controller ask for PWM=100% due the the Cable Check error
ERx: 123456789ABC C49300215136 6003 00040009 C02944F00001 00 1EC1

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F10014 0203E8FD0262FD0101F400011D4CFD0100320302 07C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F10001 00 FFC1

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F20014 0203E8FD0262FD0101F400011D4CFD0100320302 06C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F20001 00 FEC1

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004000B C02775F30003 000000 E9C1 ;host controller send notification Stop Charging (because it does know that the session is not closed yet)
ERx: 123456789ABC C49300215136 6003 00040009 C02775F30001 00 EBC1  ;Confirmation OK!!!

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F40014 0203E8FD026CFD0101F400011D4CFD0100320302 F9C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F40001 00 FCC1

ERx: 123456789ABC C49300215136 6003 00040009 C02981FF0001 01 D0C1  ;CPState=B

ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763F50014 0203E8FD026CFD0101F400011D4CFD0100320302 F8C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763F50001 00 FBC1

..... around 2.8 seconds when host controller keep updating DC parameters (0x27:0x63)
                                                                              
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001C C02763100014 020000FD029EFD0101F400011D4CFD0100320302 98C1
ERx: 123456789ABC C49300215136 6003 00040009 C02763100001 00 E1C1

ERx: 123456789ABC C49300215136 6003 00040009 C0278C000001 00 C8C1  ;WB send "Session Stopped" message

AS you can see, WB decided a session error (unspecified) right after it receives the message Cable Check Finished (Failed)
and after 3 seconds sends the Session Stopped mesage to host controller. During this time, the message Session Stop Req from EV is completely ignored!!! Please check more carefully this sequence, because something is obviously wrong in the WB behavior.

Regarding the standard requirements, I agree that they are a little confusing. Please check again the [V2G2-034] requirement that you quoted. As you can see, [V2G2-034] states that "The SECC shall terminate the TLS or TCP connection after stopping the V2G Communication Session". So the communication has to be stopped **ONLY AFTER** stopping the V2G Communication Session (or due to a sequence time-out, if nothing is received from EV)! Also please check following requirements:
a. [V2G2-570] and [V2G2-571] - [V2G2-571] states that "The V2G Communication session is then stopped without error". This imply that, if [V2G-539] is applied directly, without [V2G2-034], an error will be raised by EV, and this happens indeed with the real EV.
b. [V2G2-524]. In this case EVCC will stop the V2G Communication Session by sending Session Stop Req message to EVSE, so EVSE has to receive and confirm the message and only after that the communication will be closed!

I hope that is clear now that the EV has nothing wrong and it is acting 100% according to spec!

Best Regards,
Ionel Fratila

jpo-stx, 2022-04-06:
Hi @IonelFratila , thank you for the detailed information.

The requirements ([V2G2-570], [V2G2-571]) apply in a specific state of the statemachine, see Figure 104. There you see that the requirement only applies after PowerDelivery or WeldingDetection was processed.

In the case of CableCheckRes with FAILED response code [V2G2-585] applies, which leads to [V2G2-539] and [V2G2-034].

Nevertheless, sending a SessionStopRes could be helpful, even if it is not according to spec.

Regarding the session error. When a session error is received by the host the session is already terminated and we are just waiting for the connection to be closed by the EV. When the EV is not closing the connection withing 5s, EVSE closes the connection. I'll add this information to the user manual.

IonelFratila, 2022-04-06:
Hi @jpo-stx ,
> In the case of CableCheckRes with FAILED response code [V2G2-585] applies, which leads to [V2G2-539] and [V2G2-034].
> 
> Nevertheless, sending a SessionStopRes could be helpful, even if it is not according to spec.

I don't understand, did you read [V2G2-034] (you agree that applies here)? Sending a SessionStopRes is mandatory, according to spec.( The SECC shall terminate the TLS or TCP connection **after stopping the V2G Communication Session**.) and WB does not respect this.

In the case we discuss, EV ask for Session Stop and, if a confirmation is received from EVSE, then the connection will be closed. Because WB decides for himself that the Session is closed (???) without sending a Session Stop Response=OK, EV does not close the connection, but WB waits "for the connection to be closed by EV". Doesn't it look abnormal to you?
Best Regards,
Ionel Fratila

IonelFratila, 2022-04-10:
Hi @jpo-stx ,
Thank you for the feedback.

> > 13\. Cannot start a new SLAC matching process or SLAC matching is not followed by a new charging session, when the EV is not plugged-out and then plugged-in again (I have to test further with PWM=0%, as you suggested)
> 
> Let me answer this seperately.

1. You promised an answer regarding the restart of communication without plugging-out the EV, according to ISO15118-3, Chapter 7.5. As you can see in the update on issue 7 above, I have tested also with PWM=0% (CPState=F), but the result is the same (EV starts SLAC Matching only if SLAC Service in WB is restarted, but because NMK is changed, EV will not continue with SECC Discovery Request).
2. Regarding this, I saw an answer in another thread that said it is possible to "send a request set NMK in the WHITE-beet module, e.g. with plctool -M command, which would just set default HomePlug AV NMK using VS_SET_KEY.REQ". Do you think that I can use this in order to keep NMK unchanged after restarting the SLAC Service? Is this setting permanent?

Thank you,
Best Regards,
Ionel Fratila

lho-stx, 2022-04-12:
Hey,

@jpo-stx is on vacation this week. I will look into this and give you an answer soon.

Sincerely,
lho

jpo-stx, 2022-04-26:
Hi @IonelFratila ,

as a workaround you can try the following with the Open PLC tools (https://github.com/qca/open-plc-utils):

- When the V2G session stopped read out the current NMK, replace enp0s3 with the interface name you are using:
`plctool -i enp0s3 -I`
- The key will be shown as:
`NMK xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx (HomePlugAV)`
- Set the NMK to the default value:
`plctool -i enp0s3 -M`
- Set the NMK to the value from the previous session:
`plctool -i enp0s3 -M -K xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx`
- Send the Start Matching command to the Whitebeet.

IonelFratila, 2022-04-29:
Hi @jpo-stx ,
I wonder how I can use Open PLC tools you recommend right after session stop is raised, switch the Eth connection from Host Controller to PC, send all this commands manually (including copy the NMK from one response to the next command), switch back Eth to Host Controller and send Start Matching in the time specified in the standard for restarting the communication with EV. And during this time also CP state have to be controlled (by Host Controller, which keep trying to communicate with WB!). May I remind you that I'm not using FreeV2G tool running from PC for testing purposes, but I'm using a Hot Controller who is in charge of the entire charging process? And the EV I'm using is a real EV (not a SW simulation I can control), which will not wait indefinitely to restart the communication.
I cannot see how to apply the workaround you proposed, unless you have a possibility to set the NMK to a fixed value, used for every communication session. Even so, if this NMK can be set only from PC, I cannot switch the Eth connection from Host Controller to PC and back without resetting the WB module, as I described in issue #56.
Please tell me if you have another idea on how to keep the NMK unchanged after session stop, until EV is plugged-out. Do you have any information about when this will be corrected in WB firmware?

Thank You,
Ionel Fratila

jpo-stx, 2022-05-02:
@IonelFratila I moved the still open SLAC pause issue to #61, if you have anything to add please comment there.
Thank you,
Jan

closed 2022-05-02 by jpo-stx