Sevenstax/FreeV2G #45: Stop charging initiation from EVSE

pubaek, 2022-03-21:
This is a query about user manual.  
It is clear that the request to stop charging session comes from EV side is handled inside EVSE and session is stopped.
It is not clear how user can manually stop the charging from EVSE.  
Is it using v2gstop(), slacStop() and controlPilotStop() ?  
There is API in EVSE configuration - Set Stop Charging Status - i.e. in response to the EV request to stop charging but no specific command obvious from manual, for stopping charging from EVSE.

jpo-stx, 2022-03-22:
Hi @pubaek,
in the commands for the cable check, pre charge and charge loop you can set the Code parameter to 3: Stop charging. I.e. have a look at chapter 17.10.17.

jpo-stx, 2022-03-29:
Hi @pubaek , I think that the answer has solved your problem. Please feel free to reopen this issue if it is not solved yet, or open another one if you have further issues.

pubaek, 2022-04-01:
Hello @jpo-stx , 

The problem is not solved yet as there is no response from the Whitebeet to the sent command and the charging continues.

Tried to send charge loop parameters with stop charging command.  Scanned ethernet frame being sent and it is verified that the corresponding frame is gone but there is no response from Whitebeet.  Eventually the SendRecevieAck function returns connection error.  

The changes done in the source are present in the attached file.  Also attached is the scanned ethernet frame which is going out from application to Whitebeet but there is response frame.

On the contrary, there is corresponding response frame for every ChargeLoopParameters call with code parameter = 0 (i.e. continue charging).

Can you suggest if there is some issue ?

Thanks,
[image]
[DCChargeLoopParameters_withStopCharging.py.txt](https://github.com/Sevenstax/FreeV2G/files/8394958/DCChargeLoopParameters_withStopCharging.py.txt)

jpo-stx, 2022-04-01:
When you set the code 3 or 4 the Whitebeet sets a notification in the V2G response message that is sent to the EV. The EV then needs to stop the charging. Do you have the possibility to log the messages on EV side? If not and you interface the Whitebeet via ethernet you can activate the port mirror, the PLC traffic is then sent out on the ethernet port and you can monitor it with wireshark. It would be helpful if you post the wireshark trace here.

jpo-stx, 2022-04-05:
Thanks @pubaek , but unfortunately I do not see the V2G communication in this log. Is the Whitebeet connected directly to the PC where you are sniffing with Wireshark?

pubaek, 2022-04-05:
Whitebeet is connected to PC through switch.  Should it be directly connected?  As seen in the logs, Homeplug AV packets are also captured although Whitebeet is not connected directly to PC (with Wireshark).

ttu-stx, 2022-04-05:
Pls connect directly!

nikhilnezodi, 2022-04-06:
Wireshark Logs with Whitebeet directly connected to PC. Port mirror state was also set to ON.
[WiresharkLogs_6Apr.zip](https://github.com/Sevenstax/FreeV2G/files/8425230/WiresharkLogs_6Apr.zip)

pubaek, 2022-04-07:
> Wireshark Logs with Whitebeet directly connected to PC. Port mirror state was also set to ON. [WiresharkLogs_6Apr.zip](https://github.com/Sevenstax/FreeV2G/files/8425230/WiresharkLogs_6Apr.zip)

@jpo-stx please suggest if the logs can help in analyzing the issue, thanks

nikhilnezodi, 2022-04-07:
@jpo-stx through the wireshark logs what we observed was that code parameter to stop charging was sent at 
no.1208. But then at no.1212 codico board responds with "0x05 module not in state to accept command". Can you please tell why this warning comes and how to avoid getting this warning.

[image]

jpo-stx, 2022-04-08:
Hello @nikhilnezodi ,

what I see here seems ok:

1207: WB requests charge loop parameters
1208: Host sends charge loop parameters
1209: WB acknowledges the chargeloop parameters with OK
1211: Host sends charge loop parameters again, without them being requested
1212: WB does not acknowledge the charge loop parameters as it is expected, because they were not requested.

So the question is, why do you send the parameters again, without them being requested.

pubaek, 2022-04-11:
Yes WB was not requesting the charge loop parameters. Thanks @jpo-stx for the support,

jpo-stx, 2022-04-19:
You're welcome!

closed 2022-04-19 by jpo-stx