Sevenstax/FreeV2G #46: [Query] Could the Timeout be changed in handleRequestChargeLoopParameters command ?

Joenoodoe, 2022-03-23:
Hi,

Although we have done the both module-EI and module-PI with SPI interface, we're curious about the timeout issue in handleRequestChargeLoopParameters command.

> **[NOTICE] V2GDIN12: Message performance timeout of 75ms reached.**

Could we manually set this timeout by ourselves if we need it longer than 75ms ?
If yes, where should we modify because we didn't find any parameter to set this timeout ?  

Many Thanks for your support.

BR
JoeChen

jpo-stx, 2022-03-29:
Hi @Joenoodoe ,
unfortunately this is not possible, There is a requirement in the specification that the EVSE needs to answer within a certain timewindow to a request message. Therefore, we changed the interface in the V2 of the Whitebeet firmware so that we can get around this problem. There you need to send your current parameter set in the chargeloop phase at least every 500ms, but you can also send it more often if you would like to have more redundancy.

closed 2022-03-29 by jpo-stx

st4hli, 2022-08-09:
Hi @jpo-stx ,
why did you change this behaviour only in the V2 plug and charge firmware?
We're having the same timeout problem during the charge loop when running FreeV2G (tag EVSE_v1.1.7_1) on a Raspberry Pi 4 with the whitebeet version V01_01_07. At some point in time we see:

`[NOTICE] V2GISO13: Message performance timeout of 75ms reached.` (in UART debug output)

FreeV2G python output shows:

`Warning: Module did not accept command 27:70, return code: 5`

Do you have an idea what we can do as a workaround while still running it on a slow host PC? e.g: Is it possible to use the v2 firmware with external identification means (EIM) too?

BR,
st4hli

lho-stx, 2022-08-09:
Hey @st4hli,

yes, it is possible to use EIM with the EVSE V2. Actualy it is mandatory to support EIM. When using the whitbeet EVSE firmware V2 you can set the supported protocols in the `Set Configuration` command of the framing interface. Please have a look into the  [whitebeet manual](https://github.com/Sevenstax/FreeV2G/raw/EVSE_v2.0.1_3/SEVENSTAX_WhiteBeet_UserManual.pdf) , section 18.10.1

Regarding your performance question, I try to reproduce your problem and will come back to you next week.

Best regards,
lho