Sevenstax/FreeV2G #47: Problem with SLAC init

Ivangel1972, 2022-03-23:
I have Whitebeet module with SPI comm. I have problem with SLAC - after connecting car
this is a log:

DC Controller start 2013-1-1 1:36:47

C0 05 81 FF 00 01 01 F4 WhiteBeet interface UP 2013-1-1 1:36:48
C0 0E 00 BB 00 00 00 C1 2013-1-1 1:36:48

C0 0E 01 BB 00 00 B2 C1 WhiteBeet Synchro OK 2013-1-1 1:36:48
C0 28 43 BB 00 00 00 C1 2013-1-1 1:36:48
C0 10 41 BB 00 00 00 C1 2013-1-1 1:36:48
WhiteBeet FW V01_01_062013-1-1 1:36:48

C0 29 40 BB 00 01 01 00 C1 2013-1-1 1:36:48

C0 29 40 BB 00 01 00 57 C1 CCS CP_Set_mode OK, CCS=init1 2013-1-1 1:36:48
C0 29 44 BB 00 02 03 E8 00 C1 2013-1-1 1:36:48

C0 29 44 BB 00 01 00 53 C1 CCS CP_Set_PWM_duty OK , CCS=init2 2013-1-1 1:36:48
C0 29 42 BB 00 00 00 C1 2013-1-1 1:36:48

C0 29 42 BB 00 01 00 55 C1 CCS CP_start OK, CCS=init3 2013-1-1 1:36:48
C0 28 42 BB 00 01 01 00 C1 2013-1-1 1:36:48

C0 28 42 BB 00 01 00 56 C1 wb_SLAC_start_subID OK , CCS=idle 2013-1-1 1:36:48

_This is a resting state, I'm waiting for the car to be connected_

C0 29 81 FF 00 01 01 D0 C1 CCS CP state changed to 1 2013-1-1 1:48:12     _<-car is connected_
C0 28 44 BB 00 00 00 C1 2013-1-1 1:48:12   _<-start SLAC matching_
**C0 28 44 BB 00 01 10 44 C1 2013-1-1 1:48:12    _<- !!! status 0x10 - "Wrong State**_
C0 28 43 BB 00 00 00 C1 2013-1-1 1:48:12

C0 28 43 BB 00 01 00 55 C1 wb_SLAC_stop_subID OK 2013-1-1 1:48:12
C0 28 42 BB 00 01 01 00 C1 2013-1-1 1:48:12
C0 28 42 BB 00 01 00 56 C1 2013-1-1 1:48:12
C0 28 44 BB 00 00 00 C1 2013-1-1 1:48:12

C0 28 44 BB 00 01 00 54 C1 wb_SLAC_start_matching_process_subID OK 2013-1-1 1:48:12
C0 29 44 BB 00 02 00 32 00 C1 2013-1-1 1:48:12

C0 29 44 BB 00 01 00 53 C1 CCS CP_Set_PWM_duty OK 2013-1-1 1:48:12

But after restarting SLAC service and car is still connected will never SLAC matching success

What am I doing wrong or is there anything else I can do in addition?

jpo-stx, 2022-03-23:
Hi @Ivangel1972 ,
so it seems that the pause between the SLAC start and start matching commands seems to be an issue
`start -> 12min pause -> start matching -> fail`
and
`start -> start matching -> success`
I will check if I can reproduce this issue.
For now, you can work around that by starting the slac when you have detected the state change A->B on CP.

Ivangel1972, 2022-03-23:
Hello, thank you for the hint, it really helped but: first of all, the CP Get State function works badly. I now have a waiting state for a connected car with SLAC NOT turned on. To find out what is the state of the CP line, I either receive an "interrupt" (C0 29 81 FF 00 01 01 D0 C1 CCS CP state changed to 1 2013-1-1 1:48:12 <-car is connected) from the module that reports a change in state to me, or in a loop I query the module with the CP Get State function, and this causes the module to spontaneously move to the E state and then F = -12VDC. If I have installed firmware 01_01_06, the change in the state of the CP line generates this "interrupt" and my program reacts to it. In the latest firmware version, that is 01_01_07 it no longer works, physical change of the state of the CP line does not send this message. However, in both firmware versions, querying in the CP Get State loop causes the module to spontaneously go into the F state.
Next problem. When I start correctly, that is, I have confirmation of the correct activation of the SLAC Matching function and I try to periodically ask the module in what state is currently SLAC function Get State the module completely does not respond to it - there is no answer.
And finally - even with such a improved order of starting the function, I still fail to achieve success with SLAC matching - tests on the car and on the EVAL board from CODICO

closed 2022-03-31 by Ivangel1972