Sevenstax/FreeV2G #49: how to start/stop V2G service with PnC firmware (V02_00_00)

zacharyalutz, 2022-03-25:
According to the user manual and the response code from whitebeet, it seems that there is no longer a means to start or stop the V2G service. The FreeV2G test application ([tag EVSE_v2.0.0_0](https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.0_0)) still opens the session with a "stop v2g service" command, but as you can see in the Wireshark capture below, the response code from whitebeet is "Sub-ID is unknown".

[image]

Are the start/stop service commands intentionally left out? If so, how should one interact with the service? I raised a similar question a while back in issue  #23, where the following guidance was provided:

> **V2G Service**
After startup, you only need to configure the service once.
When the service is started the listen ports for V2G (TCP) and SDP (UDP) are opened.
When the service is stopped an open connection and the ports are closed.
I would recommend to start the service after plug-in and stop it after plug-out.  

Can you provide new guidance for safely and reliably interacting with this service? Thanks for your help

jpo-stx, 2022-04-08:
Hi @zacharyalutz ,

I just performed a charging session with the https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.0_0 tag and I cannot see that the FreeV2G application is using the old stop command. Do you have changes in your working copy?

We intentionally left out the start and stop command and replaced them by the more meaningful start listen and stop listen commands.

The same advice as previously applies:
I would recommend to start listen after plug-in and stop listen after plug-out.

lho-stx, 2022-05-02:
Hi @zacharyalutz,
i assume the issue is fixed, if not please re-open the issue.
Thanks,
lho

closed 2022-05-02 by lho-stx