Sevenstax/FreeV2G #50: [Query] Could the module upgrade and downgrade ?

Joenoodoe, 2022-03-29:
Hi,

Currently, I want to test the newest version(V.02_00_00) in module-EI. 

Because there's only one module-EI board to test, I want to make sure whether the module-EI could be upgrade and downgrade if I want ?

Thanks.
BR
JoeChen

jpo-stx, 2022-03-29:
Hi @Joenoodoe ,

yes, you can upgrade to V02_00_00 and then downgrade back to older versions. (i.e. V01_01_07)

closed 2022-03-29 by jpo-stx

ColinCGC, 2022-04-01:
I'm not so sure about that - I upgraded an EI to V02_00_00 (in a futile effort to get the reference app actually working), and now it refuses to downgrade.

```
18.819] APL: [ 0] FWU X -> 0 %
[   18.925] APL: [ 1] FWU X -> 0 %
[   18.928] APL: FWU STATUS: ERROR
[   18.937] APL: [ 2] FWU (NOT STARTED)
[   18.950] APL: [ 3] FWU (NOT STARTED)
```

```
Welcome the SEVENSTAX SFWU Updater

Initiating framing interface
Starting update process
Firmware update at 0 percent; Sent: 833 Bytes
[!] Error: Device rejected update file
```

jpo-stx, 2022-04-05:
Hi @ColinCGC the V02_00_00 checks the validity of the certificate before verfiying the signature of the firmware. Have you set the time on the Whitebeet? See user manual "12.3.1 Set Time"

ColinCGC, 2022-04-05:
No, I have not set the time, that should do the trick if it's checking the certificate. I won't try it now (as V02_00_00 is working ok), but will bear in mind for future updates, thank you.

Sertplas, 2022-04-11:
I can confirm that after setting the time, i successfully downgraded the module from v2 to v1.0.7. But i think it would be beter if software updater should do this by himself before starting the update procedure