Sevenstax/FreeV2G #52: Test mode for power spectral density calibration?

zacharyalutz, 2022-04-05:
Is there a software "test mode" that ensures the following behavior? 

- QCA7000/QCA7005 is accessible over CP
- Fixed and known NMK (i.e. randomization disabled)
- PWM generation on CP disabled (+12VDC is ok)

Such a mode would enable PSD calibration which is important for tuning SLAC attenuation parameters. Thanks for your time.

vbudko, 2022-04-05:
Hi Zach,
You can send a request set NMK in the WHITE-beet module, e.g. with plctool -M command, which would just set default HomePlug AV NMK using VS_SET_KEY.REQ.
When you do not initialize CP service, CP voltage would be -12V DC, if you do initialize, then it will turn to +12V DC.
BR, Vasily

> Is there a software "test mode" that ensures the following behavior?
> * QCA7000/QCA7005 is accessible over CP
> * Fixed and known NMK (i.e. randomization disabled)
> * PWM generation on CP disabled [...]

zacharyalutz, 2022-04-06:
Thanks for the super fast response :)

closed 2022-05-02 by lho-stx