Sevenstax/FreeV2G #54: How to migrate to new plug-and-charge firmware update.

zacharyalutz, 2022-04-07:
I have several questions about the V2G service API for the new plug-and-charge firmware. As others in this forum have pointed out, the V2G service has undergone a major overhaul. Migrating to this new update will come at a great cost time-wise, so to mitigate the inherent risk, I would appreciate your insight into the new design. I have several disjoint questions regarding the new FW, which I will try to keep organized within this thread.  Please consider these questions a follow-on to issue #49.  

### V2G: Set/Update Charge Parameters

1. There are two distinct though very similar _EVSE configuration commands_ regarding DC charging parameters.  Can you please illustrate the differences between them, describe when to use one versus when to use the other?
    - **V2G: Set DC Charging Parameters** `27 62`
    - **V2G: Update DC Charging Parameters** `27 63`

2. Speaking of `27 63` **Update DC Charging Parameters**, I am confused about when the host should send this command to WB. Usage is not explicitly prescribed in the user manual. The only context for using the command is in section 21.8.  It looks like once Cable Check is requested by the vehicle, host should periodically send this message every `x` milliseconds until the end of the session. 

    - How is interval `x` determined? 
        - Is this configurable? Constant?  
        - Just as a reference, according to ISO 15118-2 Table 109, the shortest `V2G_EVCC_Msg_Timeout` interval is 250ms for main charge loop 
    - Are there other times when WB expects updated charge parameters?

3. I have driven the new firmware with your FreeV2G application. After combing thru the ethernet frames, I notice your usage of the command **V2G: Update DC Charging Parameters `27 63`** actually first occurs before Cable Check is requested (see line 352 in wireshark log[^1]). It looks as though this is sent in response to certain messages sent by EV, but is repeated at somewhat irregular intervals (sometimes as short as 3ms apart if wireshark is to be believed). To put my cards on the table, here's my best guess at when this command should be used, please tell me what you think:
    - Send as a response to EV message `27 85` **DC Charging Parameters Changed**
    - Send any time host detects a change in one of the fields contained in `27 63` (i.e. Isolation Level, Present Voltage/Current, EVSE Status Code, Vmax, Pmax, Imax) 
    - Send periodically (i.e. every 100ms)

### V2G: Stop Listen

1. In  same FreeV2G application session as above, V2G: Stop Listen `27 70` was sent 2x by host with valid response code `00` from WB (see lines 382-385 in wireshark log[^1]). Is there a reason for this? It seemed like a command that would only need to be sent once if the response code was OK.

### V2G: Session Error

1. In  same FreeV2G application session as above, host received Session Error `27 8e` with decimal error code 23. Only error codes 0-22 are defined in the user manual. Can you tell me what this means? Consider the following sequence (~line 377 in log[^1])
    - `27 88` - Precharge started
    - `27 63` - Update d.c. charge parameters
    - `27 8c` - Session stopped (terminate)
    - `27 8e` - Session error (errcode=0x17)

---
Thats all for now, thanks for your time 
[^1]: [wireshark log](https://github.com/Sevenstax/FreeV2G/files/8446239/freev2g-wb-v2.0.0.pcapng.gz)

jpo-stx, 2022-04-08:
Hi @zacharyalutz ,

Thank you for the nice structure, I will try to keep it in my replies.

### V2G: Set/Update Charge Parameters

1. We have some parameters which cannot change during a session (i.e. peak current ripple or minimum values for voltage and current) and we have parameters that can change during the session (i.e. present voltage, status). Therefore we decided to have one command to initially set all the parameters and one command to update the parameters that can change during the session. The update command can be seen as a subset of the set command.
2. We added further information in the new manual [version 2.13](https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_0). The update command has to be sent at least every 500ms (not configurable) or when one of the parameters in the message changes. The 500ms timeout for this message applies from the command **Set Schedules** until **Session Stopped** or **Session Error**.
3. I think that 2. also answers this one.

### V2G: Stop Listen

1. The **Stop Listen** command only needs to be sent once. Seems to be due to the cleanup sequence of the FreeV2G application and is not intended.

### V2G: Session Error

1. This is the error code that the 500ms timeout has expired. This information is also added to the new [version 2.13](https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_0) of the user manual.
However, when I have a look at your wireshark look it seems a bit odd. The FreeV2G application should update the parameters every 100ms. In the log it seems to be every few ms and nonetheless the Whitebeet send the timeout error.

lho-stx, 2022-05-02:
I assume this issue is solved, if not please re-open.

closed 2022-05-02 by lho-stx