Sevenstax/FreeV2G #56: WB V02_00_XX firmware upgrade/downgrade problem

IonelFratila, 2022-04-15:
Hi @jpo-stx ,
1. I tried to upgrade my WB module from V02_00_00 to V02_00_01, using StxFwUpdater application, in the same way as for previous versions. As usually, after connecting WB module to the laptop using Eth cable, a Power-On reset of WB is needed in order to start the update, otherwise the following message is displayed (I've got this message every time also for previous SW versions if I don;t apply a Power-On reset of WB):
 " [!] Target does not answer to initial frame, check whether it has connectivity and has enabled framing over ethernet"

2.After Power-On reset, the update process is started normally, but the following error message is raised:
" Initiating framing interface
  Starting update process
  Firmware update at 0 percent; Sent: 833 bytes
  [!] Error: Device rejected update file"

I saw in another thread that a possible cause could be that the WB module does not have the internal clock updated, so I updated the clock using the SetUTCTime command, sent from the Host Controller. After that, I switched the Ethernet cable from Host Controller to Laptop, but the upgrade did not start, with the error message from point 1 (tested using StxFwUpdater application, both in Linux Ubuntu and Win10). If I apply a Power-On reset, the upgrade is started, but of course the internal clock of WB is lost and I get the error message from point 2. I tried to send the FWUpdate SetMode command (0x11:0x41) from Host Controller after setting the clock and is accepted every time.

3. I tried the same SW update to V02_00_01 for the second WB module, which is initially V01_01_06. Everything went well this time and the upgrade ended with "Done" message. But if I try now to run again the update procedure (over the new version V02_00_01), I get the same error message as above.

Having regard to the above, my conclusion is that WB module does not permit SW upgrade/downgrade over version V02_00_XX (V02_00_00 or V02_00_01), even the internal clock of the module is updated first.
Please advise what I'm doing wrong and how I can apply further upgrades/downgrades for these modules having SW version V02_00_XX.

Also, there are missing information in the User Manual:
a.What is the time-out for Firmware update Module (the time after sending SetMode command 0x11:0x41 and receiving the status message Time-Out 0x11:0x81)?
b. What happens if after FW Upgrade SetMode command (0x11:0x41) (but before Time-Out expires) other commands are sent to WB? The module will be still in update mode until Time-Out? It seems that other commands are accepted during this time. The module can be used normally or the Time-Out Error message should be expected first?
c. What happens with the WB module after the SW upgrade is ended, either by sending all Data Frames and receiving Firmware Update Ready message (0x11:0x80) or by receiving Error message (0x11:0x82)?
I ask this because after a successful update WB module does not even respond to PING messages unless a Power-On reset is applied.
d. Why the upgrade using StxFwUpdater application starts only after a Power-On reset?

Thank You,
Ionel Fratila

jpo-stx, 2022-04-26:
Hi @IonelFratila ,

1. What are your preconditions for the response message: " [!] Target does not answer to initial frame, check whether it has connectivity and has enabled framing over ethernet", so I can try to reproduce it.

2. Correct, you need to set the time before starting the update. We will update the tools to do this for the user. Regarding your issue, I tried to simulate this by using two different ethernet ports. What I did was:
- Connect WB to ethernet 1
- Reset WB
- Set the time on WB
- Disconnect from ethernet 1 and connect to ethernet 2
- Use firmware update tool to start the update
This was working for 2.0.1 -> 2.0.0 and 2.0.0 -> 2.0.1, can you try to follow this procedure?

3. See 2. and try the procedure. What I found was that if you start the update and not finish it and it times out, you get the error message "[!] Error: Device rejected update file" after trying to start it again. The workaround for this is to reset the WB before starting the firmware update.

User Manual:
a. Setting the timezone is not available on the Whitebeet, you should always use UTC. A note in the manual would be helpful.
b. The timeout for the firmware update is 30s
c. You are able to use the module normally.
d. When the firmware update succeeded the Whitebeet will restart after 15s and should respond to ethernet frames, I tested this with the "Set Time" command.
e. We need to further investigate why this happens, but I have only observed this when the firmware update was started but not completed. Please use the reset as a workaround.

IonelFratila, 2022-04-29:
Hi @jpo-stx ,
I tried to apply the procedure you described, but the conclusion is that when I switch the Eth connection from Host Controller to PC or back, the communication with WB cannot be continued unless a Power-On reset is applied. I suspect that there are some differences between the Eth port of the Host Controller and the Eth port of the PC, and the Eth interface used for WB cannot detect the new communication conditions (like automatic polarity detection), even Link and Activity LEDs are ON/clipping. I tried all cable combinations for TPIN+/TPIN- and TPOUT+/TPOUT-, but the result is the same. So I cannot set WB time using Host Controller, switch the Eth connection to PC and start the update. The only possible solutions I have are:
1. to modify the firmware update tool for setting the WB time before starting the update
2. to include all the upgrade procedure in the host controller
3. to wait for you to modify the firmware update tool for setting the WB time before update
4. to wait for you to change the change WB firmware and to eliminate the check for the validity of the certificate before firmware update (I don't understand why this is needed, an invalid certificate can at most to affect the Plug&Charge charging sessions).

In order to test further the procedure you described, I have to ask if you have a tool that can be used for setting the WB time from PC, before using the firmware update tool? Which of the methods above do you think is the fastest?

Thank You,
Ionel Fratila

jpo-stx, 2022-04-29:
Hi @IonelFratila ,

you could use a switch to connect the host and the PC at the same time.

But there will also be an update for the Update Tool available at the beginning of next week! This new version 2.4 will try to update the time on the WB if possible, so you can use it with the old and the latest version >= v02_00_00.

jpo-stx, 2022-05-02:
@IonelFratila the updated tool should be available by now. If you still encounter a problem feel free to reopen the issue.

closed 2022-05-02 by jpo-stx

IonelFratila, 2022-05-03:
Hi @jpo-stx ,
I had to reopen this issue because I think there is a problem with the new firmware update tool, StxFwUpdater.exe ver.2.4, when running in Win10 (I did not test under Linux). I tried to downgrade the firmware to V.01.01.06 and then back upgrade to V.02.00.01. The upgrade is executed normally but StxFWUpdater.exe does not properly finishes the upgrade (it displays continuously the update message, as you can see in the attached log) and I had no solution than Ctrl^C to stop execution. After changing firmware, WB seems to be OK (it reports the new FW version). Please confirm if the upgrade is OK and I can use the WB module, even in these conditions.
Thank You,
Ionel Fratila
[20220503_WB Update error.txt](https://github.com/Sevenstax/FreeV2G/files/8608597/20220503_WB.Update.error.txt)