Sevenstax/FreeV2G #58: WB V02_00_01 charging ID4 problem

IonelFratila, 2022-04-20:
Hi @jpo-stx ,
I have a problem trying to charge a VW ID4 using WB module V02_00_01. The problem is the same if I use DIN70121 or ISO15118 for charging:
When EV sends Charge Parameter Discovery Request, WB answers with Charge Parameters Discovery Response (OK). Right after this, EV change CPState to C, but WB closes the TCP communication (as you can see in the attached picture log). As a result, EV sends Session Stop Request, but it get no answer from WB (because TCP connection is already closed). The only difference I found from other EVs is that VW ID4 sends Charge Parameter Discovery Request message with parameter EVReady=False (I saw on other EVSEs that this parameter becomes True only when EV asks for Cable Check Request). Anyway, according to DIN7021, this parameter should not affect the charging session ([V2G-DC-639]).

Please tell me if the parameter EVReady=False can be the cause for WB to close TCP connection without no particular reason, or if you see another cause.
Attached you can find the log between EV and WB (EVSE) and the message log between Host Controller and WB.
[image]

ERx: 123456789ABC C493002150B5 6003 00040009 C02981FF0001 01 D0C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040008 C028440F0000 02C1
ERx: 123456789ABC C493002150B5 6003 00040009 C028440F0001 00 01C1
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004000A C02944100002 0032 CBC1     ;Set PWM=5%
ERx: 123456789ABC C493002150B5 6003 00040009 C02944100001 00 FEC1
ERx: 123456789ABC C493002150B5 6003 00040008 C02880FF0000 D4C1     ;SLAC Match
ERx: 123456789ABC C493002150B5 6003 00040018 C02780000010 00335BEF155962909A06007DFA068C71 C8C1	;Session Started, DIN70121 
ERx: 123456789ABC C493002150B5 6003 00040009 C02781000001 00 D3C1	;Payment Selected
ERx: 123456789ABC C493002150B5 6003 0004000C C02782000004 0000D6D6 22C1	;Authorization Requested
ETx: FFFFFFFFFFFF 123456789ABC 6003 00040009 C0276B110001 00 D8C1				;Set Authorization Status=OK
ERx: 123456789ABC C493002150B5 6003 00040009 C0276B110001 00 D8C1
ERx: 123456789ABC C493002150B5 6003 0004001E C02783000016 000101F4001054FF000DACFF00010000000000000050 56C1	;Energy Transfer Mode Selected, En.Req=0x01F400=500, MaxV=418V, MaxI=350A,DC_Extended, EV_Ready=False, SOC=80%
ERx: 123456789ABC C493002150B5 6003 0004000E C02784000006 0000D6D30000 21C1	;Schedules Requested
ERx: 123456789ABC C493002150B5 6003 0004001E C02785000016 1054FF0DACFF00000050000000000000000000000000 4CC1	;DC Charge Params Changed, MaxV=418V, MaxI=350A, TargetV=0V, TargetI=0A
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004001E C0276C120016 00010000000100000000000151800032030101F40300 BDC1		;Set Schedules=OK, TuppleID=0, No Sales Tariff Tuple
ERx: 123456789ABC C493002150B5 6003 00040009 C0276C120001 00 D6C1
ERx: 123456789ABC C493002150B5 6003 00040009 C02981FF0001 02 CFC1	;CPState=C
ERx: 123456789ABC C493002150B5 6003 00040009 C0278E000001 17 AFC1	;Sequence Error (Session Stop from EV received)
ERx: 123456789ABC C493002150B5 6003 00040009 C0278C000001 00 C8C1	;Session Stopped
ETx: FFFFFFFFFFFF 123456789ABC 6003 0004000A C02944130002 03E8 0FC1     ;Set PWM=100%
ERx: 123456789ABC C493002150B5 6003 00040009 C02944130001 00 FBC1
ERx: C008 4000C000 123456789ABC C493002150B5 6003 00040009 C02981FF0001 01 D0C1

Do you have any answer about the rest of the issues I submitted (see issue #42, issue #56)?

Thank You,
Best Regards,
Ionel Fratila

IonelFratila, 2022-04-29:
Hi @jpo-stx ,
I will try to set schedule tuple ID=1 and test again, even I don't agree with this. Why do you think that sending schedule tuple ID=0 can be a problem? I have used the same value=0 until now on other cars, and WB works well.
Again, when trying to charge VW ID4, is the WB decision to close the TCP/IP communication socket right after sending Charge Parameter Discovery Response (OK) to EV, so there is an information coming from EV that makes WB to decide this.
As I said, the only difference I can see in the Charge Parameter Discovery Req from EV is EVReady=False. VW ID4 (and also VW ID3) set this EVReady parameter to True only when Cable Check is started.
So I have to ask you again: Do you think that is possible this parameter to be the cause for WB to close TCP session suddenly? Did you check in firmware what happens if the EV send EVReady =False?

Thank You,
Ionel Fratila

jpo-stx, 2022-04-29:
> Hi @jpo-stx , I will try to set schedule tuple ID=1 and test again, even I don't agree with this. Why do you think that sending schedule tuple ID=0 can be a problem? I have used the same value=0 until now on other cars, and WB works well. 
Because it is defined in the ISO15118-2:2014 to be in the range 1-255:

[V2G2-773]  The SECC shall use the values 1 to 255 for the parameter SAScheduleTupleID.
But if you have used the value 0 in the past this isn't the problem.

> Again, when trying to charge VW ID4, is the WB decision to close the TCP/IP communication socket right after sending Charge Parameter Discovery Response (OK) to EV, so there is an information coming [...]

I have checked the firmware and the only thing that is done with this parameter is that it is communicated to the host.

The problem here is that the host isn't updating it's parameters in time.
- Please check the latest user manual regarding the error code in the session error message you received:
**23: Charge parameter timeout reached**
- It is also mentioned in "18.10.4 Update DC Charging Parameters":
**Note: This command needs to be sent at least every 500ms after the schedules were set with the “Set Schedules” command until “Session Stopped” or “Session Error” status message was received**

jpo-stx, 2022-05-02:
@IonelFratila the issue should be fixed by sending the "Update DC Charging Parameters" message in the correct interval. If you are still encounter an problem feel free to reopen the issue.

closed 2022-05-02 by jpo-stx