Sevenstax/FreeV2G #60: Updating to V02.00.01

lherreraalloza, 2022-04-28:
Hi,

Following suggestions on https://github.com/Sevenstax/FreeV2G/issues/40, I have updated the WB EVSE to the last available firmware release available in https://downloads.codico.com/misc/plc/index.php (V02.00.01) in order to solve timeout issues during the charge loop.

Firmware update was finished successfully, but after installing V02.00.01 I cannot communicate with the EVSE. Neither by ethernet nor by the console port. How can I perform a downgrade?

Firmware update to V02.00.01
[image]

Firmware downgrade to V01.01.07
[image]

lherreraalloza, 2022-04-28:
Correction: Console port is working. The issue is only in the ethernet port (duplicated issue with https://github.com/Sevenstax/FreeV2G/issues/59).

closed 2022-04-28 by lherreraalloza

jpo-stx, 2022-04-29:
V02_00_00 checks the validity of the certificate before verfiying the signature of the firmware. The time needs to be set on the Whitebeet? See user manual "12.3.1 Set Time"