Sevenstax/FreeV2G #61: No SLAC Matching possible after Session Pause

jpo-stx, 2022-05-02:
7. Cannot start a new SLAC matching process or SLAC matching is not followed by a new charging session, when the EV is not plugged-out and then plugged-in again (I have to test further with PWM=0%, as you suggested)

_Originally posted by @IonelFratila in https://github.com/Sevenstax/FreeV2G/issues/42#issuecomment-1077394996_

Update to issue 7 on the list above:
I have tested again the procedure for re-setup the SLAC Matching process after a SessionStopReq=Terminate message, following the exact requirements in ISO15118-3:2014 standard, Chapter 7.5 (CPState=F for at least C_step_EF). The conclusions are:
a. If SLACService is stopped and then restarted, then WB transmits CM_Unassociated_Sta.Ind messages and EV sends again CM_SLAC_Param.Req message to start SLAC matching process. When SLAC matching is successful, new values for NMK and NID are sent to EV, after that the V2G communication is not started (EV does not send SECC Discovery Request). Probably this is due to the change of NMK and NID network parameters.
b. If SLAC Servcie is not restarted, no message CM_Unassociated_Sta.Ind appears and EV does not start SLAC matching anymore.
c. Can you please check if the parameters NMK and NID can remain unchanged regardless of SLAC Service restart, in order to fulfill ISO15118 requirements regarding signaling of the Unmatched state of EVSE node and also NID/NMK unchanged until EV is plugged out? Or do you have any other ideas how to solve it?

_Originally posted by @IonelFratila in https://github.com/Sevenstax/FreeV2G/issues/42#issuecomment-1078903147_

as a workaround you can try the following with the Open PLC tools (https://github.com/qca/open-plc-utils):

- When the V2G session stopped read out the current NMK, replace enp0s3 with the interface name you are using:
`plctool -i enp0s3 -I`
- The key will be shown as:
`NMK xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx (HomePlugAV)`
- Set the NMK to the default value:
`plctool -i enp0s3 -M`
- Set the NMK to the value from the previous session:
`plctool -i enp0s3 -M -K xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx`
- Send the Start Matching command to the Whitebeet.

_Originally posted by @jpo-stx in https://github.com/Sevenstax/FreeV2G/issues/42#issuecomment-1109773580_

I wonder how I can use Open PLC tools you recommend right after session stop is raised, switch the Eth connection from Host Controller to PC, send all this commands manually (including copy the NMK from one response to the next command), switch back Eth to Host Controller and send Start Matching in the time specified in the standard for restarting the communication with EV. And during this time also CP state have to be controlled (by Host Controller, which keep trying to communicate with WB!). May I remind you that I'm not using FreeV2G tool running from PC for testing purposes, but I'm using a Hot Controller who is in charge of the entire charging process? And the EV I'm using is a real EV (not a SW simulation I can control), which will not wait indefinitely to restart the communication.
I cannot see how to apply the workaround you proposed, unless you have a possibility to set the NMK to a fixed value, used for every communication session. Even so, if this NMK can be set only from PC, I cannot switch the Eth connection from Host Controller to PC and back without resetting the WB module, as I described in issue #56.
Please tell me if you have another idea on how to keep the NMK unchanged after session stop, until EV is plugged-out. Do you have any information about when this will be corrected in WB firmware?

_Originally posted by @IonelFratila in https://github.com/Sevenstax/FreeV2G/issues/42#issuecomment-1113087802_

jpo-stx, 2022-06-03:
Hi @IonelFratila ,
we are planning to fix this issue and are currently working on a solution, but we do not have an exact date when an update will be available. I'll check back with you when I have more information.

lho-stx, 2022-06-14:
Hey @IonelFratila,
thank you for your patience!

I worked on your bug, but unfortunately I can not reproduce it. My EV and EVSE implementation always restarts the communication, even when the EVSE is not sending an CM_UNASSOCIATED_STA.IND. Furthermore, an EVSE must not send a CM_UNASSOCIATED_STA.IND, because it is configured as CCo always (as stated in the ISO15118). The high level entity (HLE) of the EV then has to send CM_SLAC_PARM.REQ which the high level entity of the EVSE answers with an CM_SLAC_PARM.CNF.

If it is possible to you, I kindly ask you to provide a wireshark log of the PLC communication between EV and EVSE and the communication between the HLE and EV/EVSE , then i can investigate further.

Best regards and thank you in advance,
lho

IonelFratila, 2022-06-15:
Hi @lho-stx ,
Thank you for your message.
I tried to acquire the Wireshark log you requested, for two different situations:
a.Without restarting SLAC Service
b.With SLAC Service restart (Stop and then Start again)

I have used a Skoda CitiGo for running the tests, which uses by default ISO15118 for charging. The entire sequence I run is as follow:
1.Start charging normally (SLAC, exchange parameters, Cable Check, Precharge, Charge)
2.After entering charge loop (around 4 seconds), I have simulated an interruption by physically switching the CPState from C2 to B2. At this point EVSE sees that CPState=B and switch PWM to 100%. EV send Session Stop Request message and EVSE (actually WB module) responds with Session Stop Response (Failed).
3.When EV asks for Session stop, PWM is switched to F State for 5 secs (according to ISO15118-3 Chapter 7.5.1)
4.After 5 seconds time-out, (in b scenario SLAC Service is restarted and then V2G Service is restarted), SLAC Matching Start is sent to WB and PWM is switched back to 5%.
5.In a scenario (without restarting SLAC Service), nothing visible happens further
6.In b scenario (with SLAC Service restart), after around 15 seconds EV sends a new SLAC request and the entire SLAC Matching process is finished OK (with SLAC Match confirm), but with values for NID and NMK different than the ones obtained after first SLAC (at the begining of the charge). After that, nothing happens further (EV does not send SECC Discovery Request).

For the PLC communication logs between EVSE (WB) and EV I have used a VH5110 device from Vector and I have attached wireshark logs for both (a) and (b) scenarios (in Zip file). If you think it can help, I also attached the logs from Vector application, converted in ASCII format for easy reading (these files are split in 2 parts, before CPState=F and after CPState=F).
For the communication between Host Controller and WB module, I cannot capture this using Wireshark (even by using a switch, the messages between HLC and WB are redirected only to the corresponding MACs). Unfortunately I don't have a simple Hub to connect trough. Therefore, I have attached the logs extracted from my Host Controller, in text format (at the moment when session is stopped).

[20220615_Without_SLACServ_Restart.txt](https://github.com/Sevenstax/FreeV2G/files/8909298/20220615_Without_SLACServ_Restart.txt)
[20220615_With_SLACServ_Restart.txt](https://github.com/Sevenstax/FreeV2G/files/8909302/20220615_With_SLACServ_Restart.txt)
[20220615_Log tests.zip](https://github.com/Sevenstax/FreeV2G/files/8909304/20220615_Log.tests.zip)

Hope this will help you to find the problem with Session restart. If you need some other logs or information about the tests I made, please let me know.
Thank You,
Best Regards,
Ionel Fratila

IonelFratila, 2022-06-20:
Hi @lho-stx ,
Please confirm if you received the files I sent and you can open them. Also, let me know if the information I sent is sufficient or you need any other details in order to investigate the issue.
Yhank you,
Best regards,
Ionel Fratila

lho-stx, 2022-06-21:
Hi @IonelFratila,
first of all, thank you very much for the logs!

In the wireshark log 20220615_WIth_SLACServ_restart.pcapng frame 479 the session stop request is captured. The EV is asking the EVSE to terminate the session. In frame 481 the sesstion stop reponse from the EVSE is captured. The EVSE reponses with FAILED_SequenceError.

The wireshark log 20220615_WIthout_SLACServ_restart.pcapng shows the same behavior of EV and EVSE, please see the frames 487 and 489.

The WB is not able to resume communication because of this sequence error. I am currently investigating why the WB does not answers the match requests when the SLAC service is not restarted. Thank you for your patience on this subject.

Best regards,
lho

IonelFratila, 2022-06-21:
Hi @lho-stx ,
Thank you for the feedback. Regarding the FAILED_SequenceError message from WB, as you probably know I pointed this out a long time ago, but the answer was that this is the right answer according to ISO 15118. Perhaps a closer communication between us on this subject would have saved us a lot of time and effort in solving this problem.
I think that beside the problem with the message Failed_SequenceError, the issue regarding the communcation restart is also related to the fact that WB does not keep NID and NMK unchanged until EV is plugged-out (CPState = A is observed), so please pay attention to this as well.
Thank you,
Best Regards,
Ionel Fratila

IonelFratila, 2022-06-21:
Hi @lho-stx ,
I totally agree with you, as long as WB will be able to resume communication with EV without removing the connector. 

> The WB is not able to resume communication because of this sequence error. I am currently investigating why the WB does not answers the match requests when the SLAC service is not restarted. Thank you for your patience on this subject.

To be very clear, this sequence error cannot be the reason why WB does not resume communication. EVSE must be able to resume communication according ISO15118-3, no matter how the previous session ended, right? So WB **must be made to be able to resume communication** even when this sequence error appears (especially since the situation simulated by me can happen at any time in real world).

Regarding your current investigation (why the WB does not answers the match requests when the SLAC service is not restarted), I think that when SLAC Service on WB is not restarted, the EV does not send SLAC Match requests (or maybe I missed), so the problem is not why WB does not answer, but why WB makes the EV to not send SLAC Match Req when the SLAC Service is not restarted.
I can see SLAC Match Req messages from EV only in the test when SLAC Service on WB is restarted, and in this case WB answers to these SLAC Match Req messages and the entire SLAC MAtch process is OK. In this case something bad is happens after SLAC Matching is done.

Thank you,
Best Regards,
Ionel Fratila

lho-stx, 2022-07-04:
Hey @IonelFratila,

please apologize my late reply!
As this issue does not relate to the V2G session pause i created a new issue. Please have look at #72.

Best Regards,
lho

closed 2022-07-04 by lho-stx

jpo-stx, 2022-08-12:
Hello @IonelFratila ,

there seems to be nothing wrong in the communication, EV and EVSE do SLAC matching and the EV assigns itself an IPv6 address. Therefore, it should be able to send an SECCDP Request next to acquire the details for the TCP connection.

My guess is that the EV detected that there was something strange happening on the CP (statechange to state B during charging) and will not continue to charge.

1. Do you see anything hint on the user interface of the car, why it's not charging?
2. Can you try to reset the whitebeet before starting the CP and SLAC again?
3. Can you also tell us how you get the EV to set the state B?

BR
Jan