Sevenstax/FreeV2G #63: WB V02_00_XX firmware upgrade/downgrade problem

IonelFratila, 2022-05-03:
Hi @jpo-stx ,
I had to reopen the issue #56 because I think there is a problem with the new firmware update tool, StxFwUpdater.exe ver.2.4, when running in Win10 (I did not test under Linux). I tried to downgrade the firmware to V.01.01.06 and then back upgrade to V.02.00.01. The upgrade is executed normally but StxFWUpdater.exe does not properly finishes the upgrade (it displays continuously the update message, as you can see in the attached log) and I had no solution than Ctrl^C to stop execution. After changing firmware, WB seems to be OK (it reports the new FW version). Please confirm if the upgrade is OK and I can use the WB module, even in these conditions.
Thank You,
Ionel Fratila
[20220503_WB Update error.txt](https://github.com/Sevenstax/FreeV2G/files/8608616/20220503_WB.Update.error.txt)

jpo-stx, 2022-05-03:
Hi @IonelFratila ,
this seems to be an issue with the tool, if the Whitebeet has received all the data it will send a status message and reboot to update the firmware,
It seems that the tool is missing this status message,

jpo-stx, 2022-05-06:
@IonelFratila , there should be a new version 2.5 of the tool available, which resolves this issue.

closed 2022-05-06 by jpo-stx