Sevenstax/FreeV2G #64: 2.0.1 firmware problem

codderembedded, 2022-05-10:
Hello,

we updated the software from 1.7 to 2.0.1. And we had some problems. we looked at debug and wireshark. These are;

1. It seems that precharge came in debug and wireshark. But didnt call handlePreChargeStarted function. Because; in receive_
next_frame function, satisfied is false because of frame.req_id. This issue occurs in some tests and not in some tests. 
[debug12.txt](https://github.com/Sevenstax/FreeV2G/files/8658909/debug12.txt)
2. It looks correct in wireshark and debug, but the code calls _handleRequestStartCharging first and then _handlePreChargeStarted. This issue occurs in some tests and not in some tests.
[debug7.txt](https://github.com/Sevenstax/FreeV2G/files/8658915/debug7.txt)
3. sometimes we get timeout error in cable check
[debug5.txt](https://github.com/Sevenstax/FreeV2G/files/8658921/debug5.txt)

jpo-stx, 2022-05-10:
Hi @codderembedded ,

it would be really helpful if you can provide a wireshark log for each case and even better if you would also activate the port mirror.

1. What do you have as request ID in this case?
2. Can you please test again with https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_1. There was an issue with frames read out in the wrong order.
3. Are you sure that you are sending your updated parameters at least every 500ms after setting the schedules?

codderembedded, 2022-05-10:
We have wireshark recording for each test
[wireshark.zip](https://github.com/Sevenstax/FreeV2G/files/8659939/wireshark.zip)

jpo-stx, 2022-05-11:
1. So you do not get the log `"Pre Charge Started" received` in your FreeV2G application? We try to receive frames with request ID 0, so this frame should be received:
https://github.com/Sevenstax/FreeV2G/blob/9c4fed33b531bc1167c249e2e41a84c7e7b4420e/Whitebeet.py#L2048 https://github.com/Sevenstax/FreeV2G/blob/9c4fed33b531bc1167c249e2e41a84c7e7b4420e/Whitebeet.py#L2073

codderembedded, 2022-05-11:
Yes I dont get the log "Pre Charge Started" received in your FreeV2G application.  But in debug, in whitebeet, we see that precharge is passed. In receive_next_frame function, satisfied was False. The loop is constantly spinning and after a while session stop comes. There is something wrong with that. most of the time it works wrong. we are rarely able to charge by working correctly. Today we constantly get this error. It wasn't just precharge, sometimes it was also authorization. We would like to sell our charging station. We need to solve this issue urgently. Please can you help us? 

Or for the 1.7 firmware, if possible can we just increase the 75 ms timeout. That was only issue in 1.7 firmware.

jpo-stx, 2022-05-11:
What is your setup like, on what system are you running FreeV2G?
How is the whitebeet connected? Is it connected directly or via a switch? If you are using a switch is there a DHCP server available in the network? If not try deactivating DHCP and use link-local only.

codderembedded, 2022-05-11:
For example;
`# got frame, apply filters
                if filter_req_id and not frame.req_id in filter_req_id:
                    if frame.sub_id == 0x88 or frame.sub_id == 0x82:
                        print("Line212", filter_req_id, frame.req_id)
                    satisfied = False
                if filter_mod and not frame.mod_id in filter_mod:
                    if frame.sub_id == 0x88 or frame.sub_id == 0x82:
                        print("Line218", filter_req_id, frame.req_id)
                    satisfied = False
                if filter_sub and not frame.sub_id in filter_sub:
                    if frame.sub_id == 0x88 or frame.sub_id == 0x82:
                        print("Line223", filter_req_id, frame.req_id)    
                    satisfied = False`

in the terminal;
Set the schedule: {'code': 0, 'schedule_tuples': [{'schedule_tuple_id': 1, 'schedules': [{'start': 0, 'interval': 1800, 'power': 5000}, {'start': 1800, 'interval': 1800, 'power': 3750}, {'start': 3600, 'interval': 82800, 'power': 2500}]}]}
Im over timeout 0.1: timeout_point is 1652277258.939689 and i am 1652277258.9397728
Im over timeout 0.1: timeout_point is 1652277259.463558 and i am 1652277259.4635942
Im over timeout 0.1: timeout_point is 1652277259.5676723 and i am 1652277259.5677125
Frame 0x88
Line212 [107] 0
Frame 0x88
Line212 [108] 0
Frame 0x88
Line212 [109] 0
Frame 0x88
Line212 [110] 0
Frame 0x88
Line212 [111] 0
Frame 0x88
Line212 [112] 0
Frame 0x88
Line212 [113] 0
.......
Frame 0x88
Line212 [213] 0
Frame 0x88
Line212 [214] 0
Frame 0x88
Line212 [215] 0
Frame 0x88
Line212 [216] 0
"Session stopped" received
Closure type: 0
Frame 0x88
Line212 [217] 0
EVSE loop finished

codderembedded, 2022-05-11:
We tested it in 2 places, one is a panel pc and the other is a thinkpad intel core i5 computer. The results are the same for both. Whitebeet connected with directly. Network isnt connected. We have tested it before with switch and network connection. When we got such a result, we connected it directly. But the result did not change.

jpo-stx, 2022-05-11:
Can you change
https://github.com/Sevenstax/FreeV2G/blob/9c4fed33b531bc1167c249e2e41a84c7e7b4420e/Whitebeet.py#L2073
`response = self._receiveSilent(self.v2g_mod_id, sub_id_list, None, 0.1)`

and

https://github.com/Sevenstax/FreeV2G/blob/9c4fed33b531bc1167c249e2e41a84c7e7b4420e/Whitebeet.py#L2098
`response = self._receive(self.v2g_mod_id, sub_id_list, None, 1)`

Then the request ID of the frame should not matter.

codderembedded, 2022-05-12:
Okey I will change, Thinkpad pc python version is 3.8.10. Panel pc python version is 3.9.9

panel pc is https://www.merpazar.com/urun/ninova-nv-076-android-anakart-A3530

jpo-stx, 2022-05-12:
Hi @codderembedded ,

I added your print to the FramingInterface.py and tested on linux with python 3.8.10. Pre charging message is received and charging is started successfully:

```
Set the schedule: {'code': 0, 'schedule_tuples': [{'schedule_tuple_id': 1, 'schedules': [{'start': 0, 'interval': 1800, 'power': 25000}, {'start': 1800, 'interval': 1800, 'power': 18750}, {'start': 3600, 'interval': 82800, 'power': 12500}]}]}
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 300V
EV maximum power: 12000W
EV ready: True
Error code: 0
SOC: 99%
EV target voltage: 0V
EV target current: 0A
Charging complete: 0
"Request Cable Check Status" received
Line212 [24] 0
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 300V
EV maximum power: 12000W
EV ready: True
Error code: 0
SOC: 99%
EV target voltage: 200V
EV target current: 40A
Charging complete: 0
Line212 [25] 0
"Pre Charge Started" received
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 300V
EV maximum power: 12000W
EV ready: True
Error code: 0
SOC: 99%
EV target voltage: 200V
EV target current: 40A
Charging complete: 0
"Start Charging Requested" received
```

Can you test with a clean clone of the repository?

codderembedded, 2022-05-12:
Did you test it like this?
response = self._receive(self.v2g_mod_id, sub_id_list, None, 30)
response = self._receiveSilent(self.v2g_mod_id, sub_id_list, None, 0.1)

jpo-stx, 2022-05-12:
No, I tested with a clean clone of https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_1 with just the print added.

codderembedded, 2022-05-12:
I am running the same code

jpo-stx, 2022-05-12:
I just updated the plug and charge branch, please check if this fixes your issue.

codderembedded, 2022-05-13:
I updated the code. And The problem fixed. I did a long charge test, about 2 hours. charging is not stopped. Thank you so much. But it was a question. in the Charge parameters change func, self.whitebeet.v2gEvseUpdateDcChargingParameters(charging_parameters),while charging I changed status 2 and charging did not stop. Why? how can i stop charging?

codderembedded, 2022-05-13:
Can i ask a question how should i write the stop command. It stops when you set status 2 in the Charge ParametersChanged function. Charge parameters change does not always come. when it comes will it be able to stop?

jpo-stx, 2022-05-13:
Setting the status to "shutdown" and send a notification for stop charging should do the trick:

**18.10.20 Send Notification**
```
C0 27 75 07 00 03 (ReqID: 7; Payload: 3 byte)
00                (Stop charging)
00 00             (Timeout is 0s)
00 C1
```

codderembedded, 2022-05-13:
Okay thak you so much

closed 2022-05-13 by codderembedded