Sevenstax/FreeV2G #65: Session ID not initialized correctly by Whitebeet (rejoins "old session" after powerup)

zacharyalutz, 2022-05-17:
### Setup
**Platform**: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; EVSE
**Firmware Version**: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;V01_01_07
**Host Controller Interface**: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Ethernet
**Host**: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Own Implementation
**Test Vehicle**: &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; Chevy Bolt
**Protocol Selected by EVCC**: &nbsp;&nbsp;&nbsp;DIN SPEC 70121-2:2012

### Description
1. In _Session Setup Request_, the EV correctly sets the Session ID to zero `00:00:00:00:00:00:00:00` according to [V2G-DC-241].  
2. In _Session Setup Response_, Whitebeet returns response code OK_OldSessionJoined `0x2`. 
3. _Service Discovery Request_ happens
4. In _Service Discovery Response_: FAILED_UnknownSession `0x7`
5. _Session Stop Request_ happens

### To Reproduce
Look thru the wireshark log attached to replicate my configuration. Run whitebeet firmware v01_01_07, attempt charging session. 

### Expected behavior
I would expect whitebeet to establish a non-zero Session ID to proceed with the charge parameter exchange and request a cable test. precharge, and enter the main charge loop.  

### Logs
[wireshark w/ port mirror](https://github.com/Sevenstax/FreeV2G/files/8704117/bolt-5-16-service-discovery-fail-unkown-session.pcapng.gz)

### Additional context
* DIN SPEC 70121 [_V2G-DC-391_]
* see ISO 15118-2 [_V2G2-750_]
* also see issue #31

jpo-stx, 2022-05-17:
Hi @zacharyalutz ,

thanks for using the template and for the detailed description. I was able to find the issue and it was already fixed in V02_00_00. Have you already started the migration to V02?

zacharyalutz, 2022-05-17:
Thanks for the quick response @jpo-stx. The migration to v2 is currently in the works, but since it requires a significant overhaul of our own firmware I am not ready to switch over yeet. Would you be able to release a v01_01_07 patch for this so I can get some more mileage in the meantime?

Thanks,
Zach

jpo-stx, 2022-05-17:
I need to clarify whether or not we are planning to further support the "old" firmware version.

zacharyalutz, 2022-05-24:
Hi @lho-stx @jpo-stx, curious if you are able to provide a patch for v01_01_07 or not? I see the issue has been left open and a "bug" tag was added, but not clear whether you will support this version nor what the timeline for a patch would be.

We have been able to charge DIN SPEC 70121 successfully with the old v01_01_02 firmware reliably, but unsuccessful for many ISO 15118 EVCCs. Firmware versions v01_01_05 thru v01_01_07 are promising for ISO 15118, but now I am having issues with DIN SPEC 70121 EVCCs. Like I said, the changes between v01 and v02 trigger a major overhaul in our own firmware, and we are still in the process of making the migration to v02. There are many new features in v02 which I would rather not test before pinning down core functionality in a firmware version that works reliably with both DIN and ISO state machines.

As always, thanks for your time and effort. Hope to hear from your soon!