Sevenstax/FreeV2G #66: v2.0.1_2 firmware status notification problem

codderembedded, 2022-05-18:
I'm sending "Send Notification" to stop the charging process. but it does not end successfully. It shows up in the terminal "Module did not accept command 27:75, return code: 4"
Why? 4: Malformed data
I used v2gEvseSendNotification(False,0) function.

jpo-stx, 2022-05-19:
Hi @codderembedded , the timeout value needs to be a uint16 value. See manual **18.10.20 Send Notification**.

Please change:
https://github.com/Sevenstax/FreeV2G/blob/7d1429caac9f85477c5769106d8ec0ef6733b697/Whitebeet.py#L1674 to:
```
payload += timeout.to_bytes(2, 'big')
```

jpo-stx, 2022-05-19:
Please use the new tag https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_3

closed 2022-05-19 by jpo-stx