Sevenstax/FreeV2G #67: firmware 2.0.1_3 problem session error

codderembedded, 2022-05-20:
I waited until the car was 100% charged. I charged it without a problem. But when it was 99 percent I got an error before it was exactly one hundred percent.

[ 5897.419] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 5897.599] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 5897.779] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 5897.959] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 5909.120] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[ 5914.337] [NOTICE] V2GCPS: Change State from C to B
[ 5914.337] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 5914.338] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[ 5914.338] [NOTICE] V2GCPS: STATE B
[ 5914.340] [NOTICE] V2GAPP: Received event 'Session error'
[ 5914.351] [NOTICE] SLAC: stxSLAC_Stop()
[ 5914.379] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server
[ 5914.380] [NOTICE] V2GAPP: Received event 'Session stopped'

Then I ran it again, this time it worked for 5 10 seconds and stopped successfully. Why do I get session stop not always successful?

jpo-stx, 2022-05-20:
Hi @codderembedded , please attach a debug and wireshark log with port mirror activated. You can also check the code transmitted in the session error message.

codderembedded, 2022-05-24:
When I first run it, I got a session error after timeout. Why did I get this error?
[wireshark.zip](https://github.com/Sevenstax/FreeV2G/files/8763429/wireshark.zip)

Then i run again and charged it for 1 hour without error. But I couldn't charge it to 100% today. Because I did not have time. Tomorrow i will send you the error which is one hundred percent.

codderembedded, 2022-05-27:
We solved the error that happens when it is 100%. it was because of our side. The other error is "wireshark1", if it happens in any way "session error", it is necessary to reset the whitebeet. if reset is not done,it occurs "Problem send receive".

closed 2022-06-16 by codderembedded