Sevenstax/FreeV2G #68: SOC changes not being reported for Whitebeet EI (v2.0.1)

ColinCGC, 2022-06-06:
Whitebeet EI, firmware v2.0.1.

It seems that the DC Charge Parameters Changed EVSE status message (0x85) is not being sent to the host when the SOC reported by the EV changes (while in main charge loop of course).
The params changed message only seems to get sent when the voltage or current changes (by 1V or 1A), but not when the SOC changes.

This is with our own SPI EVSE implementation, but the same occurs with the reference python FreeV2G app. I have not tried earlier firmware versions.

The Current Demand Request CCS message is being transferred regularly (looks to be around every 100ms) and contains the correct, updating SOC, but the Whitebeet EI does not forward this to the host.

Most likely there is some config thing I've done which Whitebeet doesn't like, as I'm sure someone else would have noticed this issue.
Is there a way to prompt the Whitebeet to send msg 0x63 that could be used as a workaround until fixed?

jpo-stx, 2022-06-07:
Hi @ColinCGC ,
when the voltage or current changes and you get the status message, is the SOC correct, or is it never getting updated?

jpo-stx, 2022-06-07:
Can you add a wireshark log and a log of the UART port of the whitebeet? Please enable the port mirror for the wireshark log.

ColinCGC, 2022-06-09:
With wireshark port mirror all I saw was SLAC messages, then nothing.

UART output below, but not much to see. After "Start charging" there are some update as VI change, then nothing.

[   95.306] [WARN  ] V2GCPS: stxV2GCPS_Start() - Failed
[  100.525] [NOTICE] V2GCPS: Change State from A to B
[  100.525] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  100.526] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[  100.526] [NOTICE] V2GCPS: STATE B
[  100.557] [NOTICE] SLAC: stxSLAC_Start()
[  102.559] [NOTICE] PLCV-HL: PIB File was updated => Download PIB without reset
[  103.057] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[  104.995] [WARN  ] SLAC:   Unexpected message!
[  105.015] [WARN  ] SLAC:   Unexpected message!
[  106.577] [NOTICE] SLAC: SLAC matching was successful!
[  106.577] [NOTICE] SLACSERV: # NC_SLAC_SUCCESS
[  106.639] [NOTICE] V2GSVC: Activate external payment option
[  106.639] [NOTICE] V2GSVC: Deactivate contract payment option
[  106.640] [NOTICE] V2GAPP: Successfully set configuration
[  106.641] [NOTICE] V2GAPP:   Protocols: DIN, ISO
[  106.641] [NOTICE] V2GAPP:   Energy modes: DC_Core, DC_Extended, DC_Combo
[  106.642] [NOTICE] V2GAPP:   Payment options: External
[  106.643] [NOTICE] V2GAPP:   Certificate installation: not supported
[  106.644] [NOTICE] V2GAPP:   Certificate update: not supported
[  106.698] [NOTICE] V2GSVC: Starting in EVSE mode.
[  106.698] [NOTICE] V2GAPP: Listen was started
[  106.699] [NOTICE] V2GSVC: TCP: Listen on port 53103 for non-TLS
[  106.700] [NOTICE] V2GSVC: TCP: Listen on port 53104 for TLS
[  108.314] [NOTICE] PLCV-HL: PIB File was updated => Download PIB without reset
[  110.542] [NOTICE] V2GSVC: TCP: A client connected on the TCP port
[  110.545] [NOTICE] V2GISO13: EV requested to start session.
[  110.546] [WARN  ] V2GISO13: V2GSVC_NC_SENT notification in wrong state (V2GMH_ISO13_STATE_RECV_EXI).
[  110.974] [NOTICE] V2GAPP: Received event 'Session started'
[  110.974] [NOTICE] V2GAPP: Received event 'Payment selected'
[  111.084] [NOTICE] V2GISO13: EV requested to start authorization.
[  111.085] [NOTICE] V2GAPP: Received event 'Authorization Status requested'
[  111.119] [NOTICE] V2GISO13: EVSE finished authorization
[  111.225] [NOTICE] V2GISO13: EV requested to start charge parameter discovery.
[  111.226] [NOTICE] V2GAPP: Received event 'Schedules requested'
[  111.227] [NOTICE] V2GAPP: Received event 'Energy transfer mode selected'
[  111.228] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  111.229] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  111.230] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  111.230] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  111.231] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  111.279] [NOTICE] V2GISO13: EVSE finished charge parameter discovery
[  112.041] [NOTICE] V2GISO13: EV requested to start cable check.
[  112.042] [NOTICE] V2GAPP: Received event 'Cable check requested'
[  112.099] [NOTICE] V2GISO13: EVSE finished cable check
[  112.153] [NOTICE] V2GCPS: Change State from B to C
[  112.153] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[  112.154] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[  112.155] [NOTICE] V2GCPS: STATE C
[  112.205] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  112.206] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  112.207] [NOTICE] V2GISO13: EV requested to start pre charging.
[  112.208] [NOTICE] V2GAPP: Received event 'Pre charging started'
[  112.209] [NOTICE] V2GAPP: stxXBAR_V2G_SendXBARStatus_EmptyMsg() - SubAdd = 0x88
[  112.316] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  112.317] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  113.816] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  113.817] [NOTICE] V2GISO13: EV requested to start charging.
[  113.818] [NOTICE] V2GAPP: Received event 'Start charging requested'
[  114.112] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  118.288] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  122.144] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  126.320] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  130.175] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  134.137] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  138.313] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[  142.168] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'

jpo-stx, 2022-06-10:
@ColinCGC I just did a test where the EV only changed the state of charge and the change in the value was sent correctly in the status message:
```
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 300V
EV maximum power: 12000W
EV ready: True
Error code: 0
SOC: 98%
EV target voltage: 200V
EV target current: 40A
Charging complete: 0
"Start Charging Requested" received
Schedule tuple ID: 1
Charging profiles: []
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 300V
EV maximum power: 12000W
EV ready: True
Error code: 0
SOC: 99%
EV target voltage: 200V
EV target current: 40A
Charging complete: 0
```
You can see that the only parameter that changes is the SOC of the EV.

Regarding the port mirror, just send the command as mentioned in **13.3.10 Set Port Mirror State**. You should see TCP traffic in the log when you are connected directly to the whitebeet. If you have a switch in between you won't see the V2G messages.

jpo-stx, 2022-07-07:
Hi @ColinCGC , I assume that you resolved the issue, since it is now 4 weeks since your last comment. If the issue still persists you are free to open the issue again.

closed 2022-07-07 by jpo-stx

ColinCGC, 2022-07-07:
Well no, no suggestions were given, nothing has changed, still don't get updates on SOC change.
I can only assume there is some difference in your setup/session config which makes a difference - although your test doesn't actually show it working since it is a single point and seems to have a "Start Charging Requested" received so it would appear to not be during normal operation, but at the start.

As an aside you really do need some documentation about the port mirror feature - what it is, how to set it up, prerequisites, troubleshooting etc. Sounds like it would be a really useful tool, but as I said all I saw was SLAC messages (we have a separate CCS monitor tool so doesn't really matter for us)

ColinCGC, 2022-08-05:
Ok, mystery solved - while investigating yet another problem I'd stumbled into (no charge param updates when using DIN), I came across this line in the fw release note: "#13596 Fix missing notification when EV parameters changed.", which jumped out at me like a big red flag, so I checked and sure enough I was actually running fw version 2.0.0, not 2.0.1 as I'd mistakenly indicated.

I managed to update (see below) and then of course it was fine, including the DIN issue (after un-reversing whatever setup params were swapped).
You'd think such major bug fixes affecting core functionality would warrant a more substantial release than an 0.0.1 increment, and known users would be notified and have the update pushed to them...

I recall trying to update a while ago, only for the update to fail (using Sevenstax_WHITEBEET_tools__2021-11-11 python updater), and was told something about needing to set the time first or something for cert validation. I tried the newer tools version (Sevenstax_WHITEBEET_tools__2022-05-04), only to find the python updater file missing, at which point I had no more energy to chase whitebeet issues so gave up until it was updated (which it still isn't btw).
This time I decided to give the windows updater version a go (having just thrown errors for me before) and this time it worked, after several attempts and noticing the difference between 7.2 and. 22.4.2.

An unrelated question, anybody know how to set the FreeService flag in the service discovery response?