Sevenstax/FreeV2G #71: Code 0x05 - Unexpected message after Set Charge Loop Parameters

Ivangel1972, 2022-06-26:
FW 01_01_07, EVSE, SPI

As in the title, I have a notorious problem with the transition of the module from Start Charge to Charge Loop. I receive Request Charge Loop Parameters and when I respond to it, the module responds with the code 0x05 - Unexpected message; module is not in a state to execute the command. What can be seen on the log, posted below, these status is always sent back after the time exceeding the Response Timeout - that is, 75ms.
I tried to delay the response, that is, after receiving the Request Charge Loop Parameters Request send to Set Charge Loop Parameters module Set in the range from 10 to 55ms - but no setting stabilized the situation. Tested on BWM I3 and Tesla. the module can run a full loading session 5 times with a given delay, e.g. 30 ms, and then send back the status 0x05 5 times. At the end of each log line there is a timer in ms - which allows you to easily check time dependencies.
I - interrupt - asynchronous data from module
S - set comands
A - answer from module to the set commands
R - request from module - asynchronous too

Log:

> CCS CP state changed to 1 2013-1-1 6:58:13 73518
> I C0 29 81 FF 00 01 01 D0 C1 2013-1-1 6:58:13 73533
> S C0 28 42 BB 00 01 01 00 C1 2013-1-1 6:58:13 73716
> wb_SLAC_start_subID OK 2013-1-1 6:58:13 73731
> A C0 28 42 BB 00 01 00 56 C1 2013-1-1 6:58:13 73842
> S C0 28 44 BB 00 00 00 C1 2013-1-1 6:58:14 73931
> wb_SLAC_start_matching_process_subID OK 2013-1-1 6:58:14 73945
> A C0 28 44 BB 00 01 00 54 C1 2013-1-1 6:58:14 73960
> I C0 05 80 FF 00 01 00 F6 2013-1-1 6:58:15 75786
> I C0 05 81 FF 00 01 00 F5 2013-1-1 6:58:15 75878
> S C0 29 44 BB 00 02 00 2F 00 C1 2013-1-1 6:58:15 75946
> CCS CP_Set_PWM_duty OK 2013-1-1 6:58:15 75961
> A C0 29 44 BB 00 01 00 53 C1 C1 2013-1-1 6:58:15 75976
> I C0 05 80 FF 00 01 00 F6 2013-1-1 6:58:20 80853
> I C0 05 81 FF 00 01 00 F5 2013-1-1 6:58:20 81002
> CCS SLAC_process_successful OK , CCS=SLAC_matched 2013-1-1 6:58:28 89434
> I C0 28 80 FF 00 00 D4 C1 2013-1-1 6:58:28 89449
> S C0 27 40 BB 00 01 01 00 2013-1-1 6:58:28 89484
> wb_V2G_Set_mode_subID OK , CCS=supported_protocols 2013-1-1 6:58:28 89499
> A C0 27 40 BB 00 01 00 59 C1 2013-1-1 6:58:28 89625
> S C0 27 60 BB 00 03 02 00 02 00 C1 2013-1-1 6:58:28 89699
> wb_V2G_Set_supported_protocols_subID OK , CCS=energy_payment 2013-1-1 6:58:28 89714
> A C0 27 60 BB 00 01 00 39 C1 2013-1-1 6:58:28 89729
> S C0 27 64 BB 00 01 00 00 C1 2013-1-1 6:58:28 89744
> wb_V2G_Set_payment_options_subID OK , CCS=energy_transfer 2013-1-1 6:58:29 89834
> A C0 27 64 BB 00 01 00 35 C1 2013-1-1 6:58:29 89849
> S C0 27 66 BB 00 02 01 01 00 C1 2013-1-1 6:58:29 89864
> wb_V2G_Set_energy_transfer_mode_subID OK , CCS=start_V2G 2013-1-1 6:58:29 89880
> A C0 27 66 BB 00 01 00 33 C1 2013-1-1 6:58:29 89895
> S C0 27 42 BB 00 00 00 C1 2013-1-1 6:58:29 90016
> wb_V2G_start_subID OK , CCS=V2G_started 2013-1-1 6:58:29 90030
> A C0 27 42 BB 00 01 00 57 C1 2013-1-1 6:58:29 90045
> wb_V2G_session_started_subID OK , CCS=V2G_started, Protocol= 1 SessionID= 20065249874209183247 EV_ID= F07F0C028B23 2013-1-1 6:58:32 93160
> R C0 27 82 FF 00 05 00 00 06 08 00 C0 C1 2013-1-1 6:58:32 93176
> wb_V2G_request_EVSE_ID_subID OK, Format 0, EVSE ID= +48*123*0000102 2013-1-1 6:58:32 93279
> S C0 27 68 BB 00 11 00 0F 2B 34 38 2A 31 32 33 2A 30 30 30 30 31 30 32 00 C1 2013-1-1 6:58:32 93297
> wb_V2G_Set_EVSE_ID_subID OK , CCS=V2G_started1 2013-1-1 6:58:32 93312
> A C0 27 68 BB 00 01 00 31 C1 2013-1-1 6:58:32 93327
> wb_V2G_request_schedules_subID OK, Entries= 12 2013-1-1 6:58:32 93458
> R C0 27 85 FF 00 06 00 00 EA 60 00 0C 73 C1 F7 06 F0 7F 0C 02 8B 23 2013-1-1 6:58:32 93549
> wb_V2G_request_discovery_charge_parameters_subID OK, max_current=125A, min_current=0A, max_voltage=405V, min_voltage=0V, max_power=0W 2013-1-1 6:58:32 93595
> R C0 27 84 FF 00 14 00 00 06 01 00 00 7D 00 00 00 00 0F D2 FF 00 01 64 01 50 2013-1-1 6:58:32 93612
> S C0 27 6A BB 00 1A 00 00 01 00 7D 00 00 01 00 03 E8 00 00 C8 00 2A F8 00 01 00 02 00 00 02 00 00 00 C1 2013-1-1 6:58:32 93629
> wb_V2G_Set_discovery_charge_parameters_subID OK , CCS=V2G_started3 2013-1-1 6:58:32 93644
> A C0 27 6A BB 00 01 00 2F C1 2013-1-1 6:58:32 93734
> S C0 27 6B BB 00 11 00 01 00 01 00 01 00 00 00 00 00 01 00 7F 00 0B 00 00 C1 00 2013-1-1 6:58:32 93753
> wb_V2G_Set_schedules_subID OK 2013-1-1 6:58:32 93769
> A C0 27 6B BB 00 01 00 2E C1 2013-1-1 6:58:32 93784
> CCS CP state changed to 2 2013-1-1 6:58:32 94012
> I C0 29 81 FF 00 01 02 CF C1 2013-1-1 6:58:33 94102
> wb_V2G_request_cable_check_status_subID OK 2013-1-1 6:58:33 94144
> R C0 27 86 FF 00 04 00 00 EA 2013-1-1 6:58:33 94159
> wb_V2G_request_cable_check_parameters_subID OK 2013-1-1 6:58:33 94175
> R C0 27 87 FF 00 06 00 00 06 2013-1-1 6:58:33 94192
> S C0 27 6D BB 00 03 00 00 01 00 C1 2013-1-1 6:58:33 94283
> wb_V2G_Set_cable_check_parameters_subID OK , CCS=V2G_started5 2013-1-1 6:58:33 94298
> C0 27 6C BB 00 01 00 00 C1 2013-1-1 6:58:33 94328
> wb_V2G_Set_cable_check_status_subID OK , CCS=V2G_started4 2013-1-1 6:58:33 94343
> A C0 27 6C BB 00 01 00 2D C1 2013-1-1 6:58:33 94358
[... 33 lines cut ...]
> S C0 27 6E BB 00 06 00 00 01 00 A5 00 00 C1 / 165V 2013-1-1 6:58:36 98275
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:36 98291
> S C0 27 6E BB 00 06 00 00 01 00 E4 00 00 C1 / 228V 2013-1-1 6:58:37 98508
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:37 98524
> S C0 27 6E BB 00 06 00 00 01 00 E4 00 00 C1 / 228V 2013-1-1 6:58:37 98759
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:37 98775
> S C0 27 6E BB 00 06 00 00 01 01 28 00 00 C1 / 296V 2013-1-1 6:58:37 98990
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:37 99081
> S C0 27 6E BB 00 06 00 00 01 01 3F 00 00 C1 / 319V 2013-1-1 6:58:37 99240
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:37 99256
> S C0 27 6E BB 00 06 00 00 01 01 3F 00 00 C1 / 319V 2013-1-1 6:58:38 99472
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:38 99488
> S C0 27 6E BB 00 06 00 00 01 01 5E 00 00 C1 / 350V 2013-1-1 6:58:38 99751
> A C0 27 6E BB 00 01 00 2B C1 2013-1-1 6:58:38 99767
> wb_V2G_request_start_charging_subID OK 2013-1-1 6:58:38 100082
> R C0 27 89 FF 00 0D 00 00 06 03 00 00 00 00 01 29 01 00 00 8B 2013-1-1 6:58:38 100097
> S C0 27 6F BB 00 03 00 00 01 00 C1 2013-1-1 6:58:38 100116
> wb_V2G_Set_start_charging_status_subID OK , CCS=start_charge 2013-1-1 6:58:38 100206
> A C0 27 6F BB 00 01 00 2A C1 2013-1-1 6:58:38 100221
> wb_V2G_request_charge_loop_parameters_subID OK, req_volt=405V, req_curr=0A, 2013-1-1 6:58:38 100237
> R C0 27 8A FF 00 1D 00 00 00 45 00 01 00 7D 00 01 0F D2 FF 00 0F D2 FF 00 00 00 29 00 01 00 01 00 00 01 00 F9 C1 2013-1-1 6:58:38 100253
> S C0 27 70 BB 00 18 00 00 01 01 5E 00 00 00 00 01 00 1B 00 01 01 95 00 01 2A F8 00 00 00 00 00 C1 2013-1-1 6:58:38 100298
**> A C0 27 70 BB 00 01 05 24 C1 2013-1-1 6:58:38 100391**
> CCS CP state changed to 1 2013-1-1 6:58:38 100407
> I C0 29 81 FF 00 01 01 D0 C1 2013-1-1 6:58:38 100421

jpo-stx, 2022-06-27:
Hi @Ivangel1972 ,

1. can you guarantee that your system is handling the TX_PENDING interrupt in time?
2. In your log what is the last number? Is this milliseconds since the system started?

Ivangel1972, 2022-06-27:
1. Yes, i quarantee
2.  Yes, is milisecons timer since system started.
As you see, in this log delay is 45ms. i try many daleys, from10 to 55ms.

jpo-stx, 2022-06-28:
Hi @Ivangel1972 ,

I just calculated the response times for your "S" messages and they are not really consistent. Most of the time it is ~15ms, but somtimes they are ~100ms:
[response_times.csv](https://github.com/Sevenstax/FreeV2G/files/9000574/response_times.csv)

When I check the timings of the python reference application I get consistent response time of ~30ms.
Any clue what can cause these delays? Are there any other tasks running on your system?

lho-stx, 2022-10-04:
Hey @Ivangel1972,
as there is no progress in this issue it will be closed. If you still encounter this issue, please reopen!
Best regards,
lho

closed 2022-10-04 by lho-stx