Sevenstax/FreeV2G #74: The code isnt working for each vehicle

codderembedded, 2022-07-07:
Hi,

We are testing with different vehicles. We adjested "energy_transfer_mode": [1] in the _handleNetworkEstablished function for evse_config. Zoe vehicle wasn't charging while energy transfer mod is [0,1,2,3]. it didnt get frame after authorize. 

But, with the code like this, we tested. But Porsche Taycan vehicle wasnt charging. we looked debug and we saw that wrote "Not in state for receiving SLAC parameter request!" 

And we changed energy transfer mod : [0, 1, 2, 3]. And it was charging. 

How will we run same code for each vehicle?

jpo-stx, 2022-07-07:
Hi @codderembedded ,
can you please add the wireshark with port mirror and a debug log for the four cases to this issue?
1. Zoe not charging
2. Zoe charging
3. Taycan not charging
4. Taycan charging

Really strange that the Zoe wasn't charging with only mode 1 activated but did with 0, 1, 2 and 3 activated.

codderembedded, 2022-07-07:
No. Zoe was charging with only mode 1 activated, but didnt with 0, 1, 2, 3

codderembedded, 2022-07-07:
Taycan was charging with mode 0, 1, 2, 3

jpo-stx, 2022-07-07:
Can you add the logs, especially for the two Zoe cases?

codderdreamer, 2022-07-18:
I wonder did you understand why?

codderdreamer, 2022-07-26:
Hi, 
The issue is somewhat urgent sorry. Station zoe and Porsche taycan can't charge in the same code. Could you take a look at the logs?

tng-stx, 2022-07-26:
Hi codderdreamer,

unfortunately some parts of  dev team is in vacation and the backup team is out of office due to illness. My actual information is that lho-stx will be back in the office tommorrow. I will push him to check this urgently.

codderdreamer, 2022-07-26:
Thank you very much

lho-stx, 2022-07-27:
hey @codderdreamer,
I appologize for your inconvenience, due to sickness I just came back to office.

At the moment I am investigating the logs. I kindly ask you to provide me a log of the FreeV2G python output for both cases...

Best regards,
lho

codderdreamer, 2022-07-27:
Case  self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [0, 1], 
            "payment_method": [0],
            "energy_transfer_mode": [0,1,2,3],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }

WHITEBEET MODULE MAC ADDRESS:C4:93:00:21:4E:15 INTERFACE:enp4s0
Initiating framing interface
iface: enp4s0, mac: C4:93:00:21:4E:15
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
[0, 1, 2, 3]
Start V2G
"Session started" received
Protocol: 0
Session ID: 87ab29db7b0c5f7c
EVCC ID: 7cbc84416d97
"Payment selcted" received
Selected payment method: 0
"Request Authorization" received
Authorization timeout: 53
Vehicle was authorized by user!
We did not receive a frame before the timeout of 30s
EVSE loop finished
Whitebeet shutdown
Whitebeet shutdown if
SUT Adapter Stop
Stopping process 64916
Process_receive exiting...

Case  self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [0, 1], 
            "payment_method": [0],
            "energy_transfer_mode": [1],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }

WHITEBEET MODULE MAC ADDRESS:C4:93:00:21:4E:15 INTERFACE:enp4s0
Initiating framing interface
iface: enp4s0, mac: C4:93:00:21:4E:15
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
EV already connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
[1]
Start V2G
"Session started" received
Protocol: 0
Session ID: 350de299fd5552de
EVCC ID: 7cbc84416d97
"Payment selcted" received
Selected payment method: 0
"Request Authorization" received
Authorization timeout: 53
Vehicle was authorized by user!
"Energy transfer mode selected" received
Maximum voltage: 412.0
Maximum current: 165.0
Maximum power: 50000
Energy Capacity: 0
Ready: no
Error code: 0
SoC: 47
Selected energy transfer mode: 1
"Request Schedules" received
Max entries: 0
Set the schedule: {'code': 0, 'schedule_tuples': [{'schedule_tuple_id': 1, 'schedules': [{'start': 0, 'interval': 1800, 'power': 50000}, {'start': 1800, 'interval': 1800, 'power': 37500}, {'start': 3600, 'interval': 82800, 'power': 25000}]}]}
EV target voltage: 0V
EV target current: 0A
EV target voltage: 0V
EV target current: 0A
EV target voltage: 351.40000000000003V
EV target current: 1.0A
"Pre Charge Started" received
EV target voltage: 351.6V
EV target current: 1.0A
EV target voltage: 351.5V
EV target current: 1.0A
EV target voltage: 351.3V
EV target current: 1.0A
EV target voltage: 351.5V
EV target current: 1.0A
EV target voltage: 351.6V
EV target current: 1.0A
EV target voltage: 351.3V
EV target current: 1.0A
EV target voltage: 351.6V
EV target current: 1.0A
EV target voltage: 351.5V
EV target current: 1.0A
EV target voltage: 351.6V
EV target current: 1.0A
EV target voltage: 351.3V
EV target current: 1.0A
EV target voltage: 351.5V
EV target current: 1.0A
EV target voltage: 351.40000000000003V
EV target current: 1.0A
EV target voltage: 351.6V
EV target current: 1.0A
EV target voltage: 351.40000000000003V
EV target current: 1.0A
EV target voltage: 351.6V
EV target current: 1.0A
EV target voltage: 351.40000000000003V
EV target current: 1.0A
EV target voltage: 351.5V
EV target current: 1.0A
EV target voltage: 351.40000000000003V
EV target current: 1.0A
"Start Charging Requested" received
Schedule tuple ID: 0
Charging profiles: []
EV target voltage: 412.0V
EV target current: 1.2000000000000002A
EV target voltage: 412.0V
EV target current: 2.3000000000000003A
EV target voltage: 412.0V
EV target current: 4.1000000000000005A
EV target voltage: 412.0V
EV target current: 6.7A
EV target voltage: 412.0V
EV target current: 9.9A
EV target voltage: 412.0V
EV target current: 13.8A
EV target voltage: 412.0V
EV target current: 18.1A

lho-stx, 2022-07-28:
Hey @codderdreamer,

there is something strange in your logs:
When a [0,1,2,3] is selected, the FreeV2G and whitebeet debug output states, that a authorization request was made, but the corresponding wireshark log is cutoff after the startListen command. Is there really no more communication? Or is the log file wireshark1 truncated? 

Best regards,
lho

codderdreamer, 2022-07-28:
I editted logs.zip Can you download again?

lho-stx, 2022-07-29:
Yes, thank you!

Best regards,
lho

lho-stx, 2022-08-03:
Hey @codderdreamer,

Thank you for your patience. Unfortunately I can not reproduce your bug. In my setup the authorization is done and the charging starts, but i have an idea what may be wrong.

To further investigate the issue, please turn on the port mirror of the whitebeet (in Application.py put `evse.whitebeet.networkConfigSetPortMirrorState(1)`  right before `evse.loop() `) and do a wireshark log of the session with [0, 1, 2, 3]. Also please try a session with [0, 1, 2]. 

Thank you,
lho

codderdreamer, 2022-08-04:
[wireshark.zip](https://github.com/Sevenstax/FreeV2G/files/9257172/wireshark.zip)

I tested wireshark1 --> energy_transfer_mode = [0,1,2]
             wireshark2 --> energy_transfer_mode = [0,1,2,3]

I dont understand energy transfer mode how it works. We are sending to whitebeet card. And it is being send to vehicle. And does it choose the ones that suit its from 0, 1, 2, 3? if so option 0, 1, 2, 3 should always work. Why is it working when I am adjusting to only 1?

lho-stx, 2022-08-04:
Hey,

Yes this is how it works.

Normally you would setup the energy transfer modes that are supported by the EVSE hardware, e.g. [0,1,2,3] and the supported protocols, e.g. [0, 1]. Then the EV will choose an suitable configuration to start the charging process, e.g. energy transfer mode 1 (DC extended) and protocol 0 (DIN). 

Your EV is using DIN protocol and chooses DC extended, when you set energy transfer mode to [1], like expected. Not every combination is suitable, for example you can not use energy transfer mode DC unique (3) with DIN protocol (0). If you try to setup the whitebeet EVSE with DIN protocol and [0, 1, 2, 3] then the whitebeet will throw an error, because this is not a suitable configuration. But this is not the case here. I am also confused why your EV does nothing after you set the authorization. 

In your logs it is shown, that with selected DIN protocol and energy transfer mode [0, 1, 2] it is the same problem, the authorization is not finished.... I expect that the EV will automatically choose the right energy transfer mode and if there is no suitable energy transfer mode there should be a ChargeParameterDiscoveryRes with FAILED_WrongEnergyTransferMode in it.

I tested your configuration and set the EV counter part to only support DIN protocol. In my case the charging session starts after i confirm authorization. 

I also expect to see V2G communicationm between whitebeet and EV in you logs, but the communiation is missing. Is the whitebeet connected to a switch? You have to connect the whitebeet directly to your network interface in order to receive the V2G messages, because the switch filters out the mac adresses. I did not point out that this is necessary, please apologize my mistake.

Could you confirm that the whitebeet is directly connect to the network interface of your host PC and try to acquire a new wireshark log? Then i can see if there is any error message in the V2G communication.

Thank you and best regards,
lho

codderdreamer, 2022-08-04:
[wireshark3.zip](https://github.com/Sevenstax/FreeV2G/files/9259240/wireshark3.zip)

Here you go...

lho-stx, 2022-08-04:
Thank you!

In frame 121 the whitebeet offers external payment with DC_core as energy transfer mode. This is then requested by the EV and granted by whitebeet.

In frame 131 the whitebeet sends the authentication response with code 'OK' to the EV (in compliance with V2G2-563 of the ISO15118-2), but then nothing happens. In frame 132 the EV sends the TCP ACK to the whitebeet, so then message is received by the EV. The next allowed message is the ChargeParameterDiscoveryReq which shall be send by the EV, but there is no ChargeParameterDiscoveryReq. Therefore the whitebeet waits for the EV.

From the log i can not see why the EV is not sending the ChargeParameterDiscoveryReq. It may be a misconfiguration of the whitebeet or maybe a problem on the EV side.

Could you do me a favor and do a log of the session with [1] for energy transfer mode?

Best regards,
lho

lho-stx, 2022-08-11:
Hey @codderdreamer,

As I said the EVSE is waiting for a response of the EV. To further investigate this issue we like to have a closer look at the EV. There are multiple versions of the Renault Zoe with different connector types and charging capabilities. May I ask you which version of the Zoe do you have?

Thank you and best regards,
lho

codderdreamer, 2022-08-11:
[wireshark.zip](https://github.com/Sevenstax/FreeV2G/files/9309466/wireshark.zip)

zoe version ZE50

the wait time was 30 seconds in the code. "stop session" came when i upgraded it

lho-stx, 2022-08-18:
Hello @codderdreamer,

we found a bug in the EVSE whitebeet FW. The FreeV2G is offering DC_core as energy transfer mode, which is currently not supported by any EV. DC_core should be used with connector type CC and DD, which are reserved for future use right now.

If the new firmware is available it will be announced in the discussion section of this repository.

Thank you and best regards,
lho

codderdreamer, 2022-08-19:
Okey, thank you

closed 2023-02-01 by lho-stx