Sevenstax/FreeV2G #76: EVSE did not accept our ServiceDetailReq!

mvargasevans, 2022-07-26:
**Setup**
Platform: PEV 
Firmware Version: V01_00_05
Host Controller Interface: Ethernet 
Host: FreeV2G (EV_v1.0.5_0)

**Describe the bug**
WhiteBEET PEV 1.1 Carrier board connected to a Porsche Charge Station.

Firmware details:
APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_05
  Target:    WHITE-BEET
  CPU:       STM32F745
 
PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_02_01
 Hex:    0x000a0201
 Dec:    100201

While running the FreeV2G App in EV mode, the session is terminated after a couple of seconds of it starting.
The FreeV2G App receives a message that the sessions was terminated by the EVSE.
Over Serial I get the attached logs. In particular, I get:

[ 5464.680] [WARN  ] V2GISO13: EVSE did not accept our ServiceDetailReq!
...
[ 5465.246] [NOTICE] V2GSVC: TCP: We were disconnected from the server
[ 5465.246] [NOTICE] V2GAPP: Received event 'Session stopped'

...
Create new charging session
ControlPilot duty cycle: 5.5
"Session Stopped" received
Change State to State B
EV loop finished
Goodbye!

I have not been able to find how to control the ServiceDetailReq, if it is even possible.

- Since this seems to be a situtation in between the White BEET and the EVSE, is there a way to see what was sent in this request to the EVSE?
- Is there a way to control the service request?
- I see people attaching Wireshark captures. That would work in between Host and White BEET. Would port mirroring transfer what is sent in between the EVSE and White BEET to the Ethernet port?

**To Reproduce**
Requires Porsche Wallbox. Unfortunately I do not have another Wallbox to test at the moment.

**Expected behavior**
EVSE should start charging instead of terminating the session.

**Logs**
- Serial
- Python
[logs.zip](https://github.com/Sevenstax/FreeV2G/files/9191602/logs.zip)

mvargasevans, 2022-07-27:
Hi
Our supplier just replied that they do not support non-TLS mode, which would explain our issue.
I am missing in the documentation how TLS is activated and configured. 
Could someone indicate if that is possible?

mvargasevans, 2022-07-27:
Hello again,

After hours of checking, I have not found a way to enable TLS communication.
I read in other closed tickets that a while ago it was not supported. Is this the case still for the PEV module?

I tested that the module works correctly and it is all configured in default settings.

lho-stx, 2022-07-28:
Hello @mvargasevans,

thank you for using the issue template! 
Unfortunately the PEV module does not support TLS at the moment.

Best regards,
lho

mvargasevans, 2022-07-28:
Hi @lho-stx 
When is it planned to support it? 
Is it a PEV module software stack issue or FreeV2G issue?

Regards

lho-stx, 2022-07-28:
Hey @mvargasevans,

as TLS ist not a mandatory feature for PEV EIM it is not supported by the PEV module (yet). The TLS functionality will come with a PEV PnC firmware update. The release date is not finaly scheduled yet. As soon as we have a fixed release date it will be announced through your local distributor.

Best regards,
lho

mvargasevans, 2022-07-28:
Hi @lho-stx 

Great to know that a firmware update is in the works to include this.
My problem is that now I hit a wall in development since I acquired the White Beet specifically to achieve communication to commercial wallboxes. At least the marketing documentation did not show indication that this was not supported.

What options do I have now? Is there a pre-release version that I could use to at least move forward?
Whom do I need to talk to get further info on this. Otherwise, I must walk away from this product and find any other solution.

lho-stx, 2022-07-28:
Hey @mvargasevans,

unfortunately I can not offer you a practical solution. If you can not wait for the PEV release, it would be best for you to move on.

Best regards,
lho

mvargasevans, 2022-07-28:
@lho-stx 
Any indication if this firmware release is planned for this week, month, year or decade?

lho-stx, 2022-07-28:
Hello @mvargasevans ,

I have to talk to the project lead about this.

Thank you for your patience,
lho