Sevenstax/FreeV2G #77: Control Pilot state not reported for PI

ColinCGC, 2022-08-02:
**Setup**
Platform: PEV
Firmware Version: V01_00_04/05
Host Controller Interface: SPI
Host: Own Implementation

I assume this is a known issue, but I have not seen it mentioned before, and it still persists in latest firmware so...

When reading the control pilot state (0x48), only the value '5' (State F, which should never be read for EV) is returned, it never changes. (The request is acknowledged).
Also no state change notifications are ever sent (0x81).

- CP mode is set to EV.
- PWM duty is reported correctly, as documented.
- For Whitebeet-EI the control pilot reporting is correct, using the same driver code as for PI.
- When looking at the freev2g ref EV sw, I notice it never reads the CP state, only the duty, which I assume is because it is known to not work

example response data frame:
[image]

lho-stx, 2022-08-03:
Hello @ColinCGC,

Thank you for pointing out the problem.

Due to illness and vacation season we are short staffed. I will look into this issue and will come back to you next week.

Best regards and thank you for your patience,
lho

lho-stx, 2022-08-18:
Hey @ColinCGC,

we assume that the EV should read state E when the EVSE is not connect. @jpo-stx is currently investigating this issue further.

Best regards,
lho

jpo-stx, 2022-08-23:
Hi @ColinCGC ,
there is an updated version available (1.0.6) which fixes this issue.

closed 2022-08-23 by jpo-stx