Sevenstax/FreeV2G #78: ModuleNotFoundError: No module named 'pylibpcap'

zgb526642524, 2022-08-15:
pi@raspberrypi:~/FreeV2G $ python Application.py "eth0" -m 70:20:22:07:22:01 -r PEV
Traceback (most recent call last):
  File "/home/pi/FreeV2G/Application.py", line 5, in <module>
    from Evse import *
  File "/home/pi/FreeV2G/Evse.py", line 2, in <module>
    from Whitebeet import *
  File "/home/pi/FreeV2G/Whitebeet.py", line 5, in <module>
    from FramingInterface import *
  File "/home/pi/FreeV2G/FramingInterface.py", line 6, in <module>
    import EthernetAdapter
  File "/home/pi/FreeV2G/EthernetAdapter.py", line 12, in <module>
    from pylibpcap.base import Sniff
ModuleNotFoundError: No module named 'pylibpcap'

lho-stx, 2022-08-18:
Hey @zgb526642524,

please install the required packages from the attached requierments.txt with pip:

`pip install -r requierments.txt`

[requirements.txt](https://github.com/Sevenstax/FreeV2G/files/9371695/requirements.txt)

Also please be aware that valid -r parameter of Application.py are `EV` or `EVSE`.

Please let me know if this solves your issue!

Best regards,
lho

zgb526642524, 2022-08-21:
@lho-stx 
thanks for the reply：
According to the steps you provided, in the Raspberry Pi environment, the following problems occur。
-----------------------------------------------------------------------------------------------
pi@raspberrypi:~/FreeV2G $ python3 Application.py "eth1" -m 70:20:22:08:18:02 -r EVSE
Traceback (most recent call last):
  File "/home/pi/FreeV2G/Application.py", line 5, in <module>
    from Evse import *
  File "/home/pi/FreeV2G/Evse.py", line 2, in <module>
    from Whitebeet import *
  File "/home/pi/FreeV2G/Whitebeet.py", line 5, in <module>
    from FramingInterface import *
  File "/home/pi/FreeV2G/FramingInterface.py", line 6, in <module>
    import EthernetAdapter
  File "/home/pi/FreeV2G/EthernetAdapter.py", line 12, in <module>
    from pylibpcap.base import Sniff
ImportError: libpcap.so.0.8: cannot open shared object file: No such file or directory
--------------------------------------------------------------------------------
thanks！！！

lho-stx, 2022-08-23:
Hey @zgb526642524 ,

i tried to repdroduce your bug on my raspberry pi 4, but in may case I had no such issue. This were my steps to setup FreeV2G:

Download the repository:

<pre>$ git clone https://github.com/Sevenstax/FreeV2G.git
$ cd FreeV2G
$ git checkout EVSE_v2.0.1_3</pre>

Create the virtual environment:

<pre>$ python3 -m venv .venv
$ source .venv/bin/activate</pre>

optional:
Install Python development packages for building pylibpcap:
<pre>(.venv)$ sudo apt install python3-dev</pre>

Install requirements
<pre>(.venv)$ pip install Cython
(.venv)$ pip install -r requierements</pre>

Start Application (NOTE: you have to give the path to the interpreter in your virtual environment, because of the sudo command):
<pre>(.venv)$ sudo .venv/bin/python Application.py "eth0" -r EVSE</pre>

Best regards,
lho

zgb526642524, 2022-08-23:
hi~@lho-stx 
Installation libpcap package：
sudo apt install libpcap-dev
perfectly solved my problem.

thank you very much！

closed 2022-08-23 by zgb526642524

lho-stx, 2022-08-23:
Hey @zgb526642524,

thank you for your feedback. This was already installed in my setup. I will update the documentation accordingly!

Best regards,
lho