Sevenstax/FreeV2G #80: MG ZS Vehicle Test Problem

codderdreamer, 2022-09-07:
We are testing different vehicles and the mg is having problems. Why does the car charge for a while and send a session stop? 
I am sending wireshark, terminal and whitebeet debug.
[mg.zip](https://github.com/Sevenstax/FreeV2G/files/9505637/mg.zip)

jpo-stx, 2022-09-26:
Hello @codderdreamer ,
there is no PLC communication present in the wireshark log. Did you enable the port mirror and are you connected directly to the whitebeet?

codderdreamer, 2022-09-26:
yes for example;
Port mirror is enable. I connected directly to the whitebeet
[image]
[image]

jpo-stx, 2022-09-27:
If you are connected directly to the whitebeet (no switch in between) and the port mirror is on (which I can see in the logs) the high level communication on the TCP protocol should be seen in the log, which is not the case here.

lho-stx, 2022-12-23:
This issue is closed due to inactivity. If the issue is still present, please re-open the issue!

closed 2022-12-23 by lho-stx