Sevenstax/FreeV2G #81: SPI multiple slave problem

ColinCGC, 2022-09-07:
Seems to be an issue when more than one whitebeet (EI or PI ) are connected to the SPI bus (or any other SPI slave device for that matter).

Being an SPI slave, the whitebeet SPI interface should switch MISO to high impedance when chip select is not asserted, but I don't think this is happening, instead the MISO looks to be driven low, pulling down the data that the active device with the asserted CS is trying to send on MISO. End result is the host master sees no SPI response from active slave. Surely this cannot be the case??

(In addition, there doesn't seem to be any pullup for CS, so if CS is not connected, or if the host micro resets for example, then the whitebeet sees an asserted CS, even though it's not being driven, causing SPI synch issues and who knows what else...)

jpo-stx, 2022-09-26:
Hello @ColinCGC ,
I checked the SPI configuration and the CS has indeed no pull up configured.
We will look into this for future versions.
I added the issue to our internal tracker.

lho-stx, 2022-12-23:
Closing this issue because it is now tracked with our internal tracker.

closed 2022-12-23 by lho-stx