Sevenstax/FreeV2G #90: No 'CP State Changed' Status Message in EVSE FW V02_00_01?

st4hli, 2022-09-28:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface: Ethernet
Host: Own Implementation

**Describe the bug**
With the V01_XX_XX firmware we were able to receive CP state updates from the codico whitebeet EVSE (mod_id=0x29, sub_id=0x81) when the CP state changes. After our recent update to V02_00_01 we don't receive these messages anymore, although it is still included in the User Manual. Is this a configuration problem on our side or a bug in your firmware?

**To Reproduce**
We modified both the self.v2gEvseReceiveRequestSilent() as well as the self.v2gEvseReceiveRequest() methods to add the respective mod_id and sub_id to the filters for _receive() and _receiveSilent(). As soon as a change in the CP signal happens, the methods should receive a CP State Changed message, but they don't.

**Expected behavior**
With the Firmware version V01_00_07 I can confirm that this worked like a charm. Our host received the request as soon as any EV connected or changed the CP state. With V02_00_01, I get no update although the debug output of the whitebeet EVSE shows detected CP state changes.
Everything else, such as performing a standard charging sequence, seems to work well.

**Logs**
see attached debug output file. 

[whitebeet_evse_debug_log.txt](https://github.com/Sevenstax/FreeV2G/files/9664232/whitebeet_evse_debug_log.txt)

st4hli, 2022-09-28:
Added a wireshark log. There it seems that something is sent via the eth0 interface when the CP state changes. (the last few packets).
What am I doing wrong?
[wireshark_log.txt](https://github.com/Sevenstax/FreeV2G/files/9664456/wireshark_log.txt)

st4hli, 2022-10-03:
I tried to exchange the receive_next_frame() method in the FramingInterface with the version from the master branch (commit hash 12eb9621) and now everything works again. I am able to receive and parse the CP state updates now.
It seems that your changes in the receive_next_frame() method in the PnC branch have a bug which causes this behaviour. Please fix this.

jpo-stx, 2022-10-04:
Hello @st4hli ,

I just did a clean checkout of the tag EVSE_v2.0.1_3 started the python application and connected an EV:

Output of python script:
```
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: Realtek PCIe GbE Family Controller #2, mac: 00:01:01:63:77:33
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
```

You can also see the status messages for the state changes on the ethernet interface (144, 147):
[cp_state_change.zip](https://github.com/Sevenstax/FreeV2G/files/9704557/cp_state_change.zip)

Can you please verfiy that you are using the latest tag?

st4hli, 2022-10-04:
Hi @jpo-stx ,

thanks for taking a look at this issue. Your python reference implementation has no code which tries to react on the cp state update message (sub_id 0x81) sent asynchronously via the ethernet interface. Instead of using your reference implementation, we're using our own modified python implementation where we added this sub_id to the filter list. In our wireshark log the CP state changes are also visible, so apparently it's not a firmware problem.
However, there were changes in your receive_next_frame() method which we have been using all the time. As described above, the new version of this method from the PnC branch (extracted from commit hash e95882ed) seems to break something because we can't receive this cp update anymore. When extracting the method-version from commit hash 12eb962, the receiving works again.

Can you reproduce this?

jpo-stx, 2022-10-04:
@st4hli ,

I see. the FreeV2G application is polling the state.

I was able to reproduce your issue. Please change:
```
response = self._receive(mod_id_list, sub_id_list, 0x00, timeout_s)
```
to

```
response = self._receive(mod_id_list, sub_id_list, [0x00, 0xff] , timeout_s)
```
since the control pilot status messages are sent with 0xff as request ID.

st4hli, 2022-10-04:
Oh, I completely missed the request ID of 0xff. Sorry, my fault, now it works of course. Thank you very much for clarifying, the issue can be closed. It's still funny that it worked with the old version from the master branch because it skipped the request_id check there, if 0x00 was passed. 😆 

PS.: According to my experience, the PnC branch together with the V02_00_01 firmware works much more reliable than the master branch with the V1_00_07 firmware on weak computers such as a raspberry pi. With the V1 firmware we had many timing issues due to our host during the charge loop.. Maybe this should be mentioned somewhere in the docs.

closed 2022-10-25 by st4hli