Sevenstax/FreeV2G #91: Fail to initiate white beet EI with command line - $ python3 Application.py "eth0" -m c4:93:00:22:22:22 -r EVSE

MalkaGal, 2022-10-03:
Platform: EVSE
Firmware Version: i.e. V01_01
Host Controller Interface: Ethernet 
Host: FreeV2G (EV_v1.0.4_0) / Own Implementation

The bug:
Fail to initiate white beet EI throw the Python terminal.

Expected behavior
as i see in the Readme file, after typing ****$ python3 Application.py "eth0" -m c4:93:00:22:22:22 -r EVSE**** i supposed to get that message:

Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: eth0, mac: c4:93:00:22:22:22
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects

and i'm getting some error messages as you can see at the picture attached.
[image]

i will be glad to know whats wrong in the configuration. thank you !

schmitzcj, 2022-10-03:
The ethernet interface "eth0" is incorrect and needs to reference the appropriate interface.

From the looks of it, you're running on Windows. "eth0" is Linux notation for a network interface. I do not believe this code is meant to run on Windows machines, but I could be mistaken.

lho-stx, 2022-10-04:
Hey @MalkaGal,

like @schmitzcj mentioned, you have to replace "eth0" with the interface name of your windows machine. You can query your interface names by running the command "ipconfig". Then the application should just run fine.

Best regards,
lho

jpo-stx, 2022-10-24:
Since you haven't responded I expect that the issue was resolved with the information given. If not fell free to reopen it.

closed 2022-10-24 by jpo-stx