Sevenstax/FreeV2G #92: No Response to ServiceDetailRequest Results in Session Stopped

schmitzcj, 2022-10-03:
**Setup details**
EV
Hardware: CODICO WB-CARRIER-BOARD-PI 1.1 
Firmware: V01_00_04

EVSE
Hardware: Vector vSECC Controller
Firmware: 2.5.1

**Problem**
When using FreeV2G, the EVSE does not know how to parse the WHITE-Beet's (WB) "ServiceDetailRequest for Service ID 1". As a result, the session times out and the program terminates. The EVSE manufacturer, Vector, states that they did not include any processing for this type of request because EVs typically do not request this information. 

There is no clear mention of a command in the WB manual to instruct the device to not issue this request. Is there a way to work around this with FreeV2G, or is this strictly a firmware issue with the WB?

Please find attached log files relating to this issue.
[UART_Log_27SEP22.txt](https://github.com/Sevenstax/FreeV2G/files/9699913/UART_Log_27SEP22.txt)
[Terminal_Output_27SEP22.txt](https://github.com/Sevenstax/FreeV2G/files/9699914/Terminal_Output_27SEP22.txt)

Thank you for your help!

jpo-stx, 2022-10-04:
Hello @schmitzcj 

The Whitebeet will always try to get details about the services available at the EVSE.

It's a helpful information that some EVSEs do not implement the processing of these messages. We can change this and make it possible to disable the discovery of the service details.

In the meantime you can try charging with DIN70121 protocol. There the Whitebeet will not send a service detail request because the protocol does not allow it.

anzaf67, 2022-11-22:
Hello @jpo-stx,

 I am facing similar problem. Same EV set up tried with two other EVSE controllers and getting the same issue.

tomc797, 2022-12-08:
Hi @jpo-stx,

The messages ServiceDetailReq and  ServiceDetailRes are optional for AC and DC charging (ISO 15118-2 (2014), Table 104). It doesn't appear that WhiteBeet recovers when it encounters EVSE that don't support the message.

lho-stx, 2023-02-01:
Hey @tomc797, @anzaf67, @schmitzcj 

there will be a fix in the next Whitebeet firmware scheduled for late Q1/early Q2.

Thank you,
lho

closed 2023-02-24 by lho-stx